raspberrypi/linux #2797: TUN/TAP device fails with a kernel oops on 4.19.13+ (BRANCH=next)

DougieLawson, 2019-01-07:
OpenVPN fails to start and gets kernel oops.

```[25967.595525] tun: Universal TUN/TAP device driver, 1.6
[25968.212587] Unable to handle kernel paging request at virtual address ffffffe8
[25968.218864] pgd = af2972cd
[25968.225055] [ffffffe8] *pgd=1effa861, *pte=00000000, *ppte=00000000
[25968.231567] Internal error: Oops: 37 [#23] ARM
[25968.237571] Modules linked in: rfkill tun ipt_MASQUERADE iptable_nat nf_nat_ipv4 rpcsec_gss_krb5 snd_bcm2835(C+) snd_pcm raspberrypi_hwmon snd_timer snd bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_dma_contig i2c_bcm2835 videobuf2_memops videobuf2_v4l2 videobuf2_common spi_bcm2835 videodev media vc_sm_cma(C) fixed uio_pdrv_genirq uio ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 xt_hl ip6t_rt ipt_REJECT nf_reject_ipv4 xt_comment nf_log_ipv4 nf_log_common xt_LOG xt_limit xt_tcpudp xt_addrtype xt_conntrack ip6table_filter ip6_tables nf_conntrack_netbios_ns nf_conntrack_broadcast rtc_ds1307 nf_nat_ftp w1_therm nf_nat wire cn hwmon nf_conntrack_ftp i2c_dev nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_filter ip_tables x_tables ipv6
[25968.270366] CPU: 0 PID: 2418 Comm: modprobe Tainted: G      D  C        4.19.13+ #1186
[25968.277036] Hardware name: BCM2835
[25968.283621] PC is at alloc_vmap_area+0x16c/0x3c4
[25968.290065] LR is at 0x4c000
[25968.296460] pc : [<c014b750>]    lr : [<0004c000>]    psr: 00000013
[25968.302840] sp : dac0bd40  ip : 00093000  fp : dac0bd94
[25968.309138] r10: ffffffff  r9 : c0000000  r8 : c09f5014
[25968.315388] r7 : 00000001  r6 : c0ae48fc  r5 : 00000001  r4 : 00093000
[25968.321558] r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : bf34c000
[25968.327723] Flags: nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[25968.333897] Control: 00c5387d  Table: 1c784008  DAC: 00000055
[25968.340038] Process modprobe (pid: 2418, stack limit = 0x34440326)
[25968.346204] Stack: (0xdac0bd40 to 0xdac0c000)
[25968.352287] bd40: dbb06b00 00000000 bf000000 00000001 bf093000 00000000 00000001 bf000000
[25968.358453] bd60: dc7fd620 4b3aa1b7 dfa95054 00093000 00000022 006002c0 00000001 bf000000
[25968.364660] bd80: dc7fd620 dfa71b5c dac0bdc4 dac0bd98 c014ba54 c014b5f0 ffffffff 006002c0
[25968.370854] bda0: 000918a3 ffffffff ffffffff 000918a3 006002c0 dfaa7904 dac0be04 dac0bdc8
[25968.377096] bdc0: c014caf8 c014b9b4 c0000000 ffffffff 006002c0 c008f4c0 dfa9951c c008f4c0
[25968.383387] bde0: ffffffff c0a6e478 000918a3 0007d278 dfaa7904 dfa71b5c dac0be3c dac0be08
[25968.389674] be00: c0015e6c c014ca9c 006002c0 0000004f 00000000 ffffffff c008f4c0 dac0be28
[25968.396032] be20: dac0bf30 dfa5b2a0 dfa5b2a0 c09d3028 dac0bf0c dac0be40 c008f4c0 c0015e20
[25968.402413] be40: 00000000 dac0be50 c0171e28 c01b79d0 00000000 df9e5000 dfa5b2a0 df9e5000
[25968.408796] be60: dfa5b3b4 000015ed 0002cd30 0001462b dac0be94 c08a0754 c0171e70 c0171ce4
[25968.415257] be80: dac0bea0 000c292c dac0bedc dac0be98 c01782f8 c0171e4c dac0bec4 dac0bf28
[25968.421694] bea0: 000c292c 00000000 00000001 4b3aa1b7 00000071 dbe610a0 dbe610a0 dac0bf28
[25968.428169] bec0: dac0bf24 7fffffff 00000000 00000000 dac0bf0c dac0bee0 c017846c 4b3aa1b7
[25968.434726] bee0: 7fffffff c09d3028 00000000 00000001 0002cd30 7fffffff 00000000 00000000
[25968.441206] bf00: dac0bfa4 dac0bf10 c0090ee0 c008ebd8 7fffffff 00000000 00000003 00000000
[25968.447736] bf20: 00000000 df9e5000 000c292c 00000000 dfa471e9 dfa5b2a0 df9e5000 000c292c
[25968.454223] bf40: dfaa6dc4 dfa5b9ab dfa71b5c 0006a398 0007d278 00000000 00000000 00000000
[25968.460719] bf60: 00035ab8 00000047 00000048 0000003e 0000002d 0000001f 00000000 4b3aa1b7
[25968.467287] bf80: c9f60800 00000000 00000000 0000017b c00091a4 dac0a000 00000000 dac0bfa8
[25968.473783] bfa0: c0009000 c0090e34 c9f60800 00000000 00000001 0002cd30 00000000 0002d71c
[25968.480310] bfc0: c9f60800 00000000 00000000 0000017b 00040000 00000000 00000000 019d70e0
[25968.486784] bfe0: bee40a60 bee40a50 0002212c b6e634f0 60000010 00000001 00000000 00000000
[25968.493266] [<c014b750>] (alloc_vmap_area) from [<c014ba54>] (__get_vm_area_node+0xac/0x144)
[25968.499859] [<c014ba54>] (__get_vm_area_node) from [<c014caf8>] (__vmalloc_node_range+0x68/0x220)
[25968.506201] [<c014caf8>] (__vmalloc_node_range) from [<c0015e6c>] (module_alloc+0x58/0xc4)
[25968.512551] [<c0015e6c>] (module_alloc) from [<c008f4c0>] (load_module+0x8f4/0x20fc)
[25968.518826] [<c008f4c0>] (load_module) from [<c0090ee0>] (sys_finit_module+0xb8/0xcc)
[25968.524976] [<c0090ee0>] (sys_finit_module) from [<c0009000>] (ret_fast_syscall+0x0/0x28)
[25968.530989] Exception stack(0xdac0bfa8 to 0xdac0bff0)
[25968.536801] bfa0:                   c9f60800 00000000 00000001 0002cd30 00000000 0002d71c
[25968.542673] bfc0: c9f60800 00000000 00000000 0000017b 00040000 00000000 00000000 019d70e0
[25968.548670] bfe0: bee40a60 bee40a50 0002212c b6e634f0
[25968.554650] Code: ea00000b e5922018 e1520008 0a000045 (e5121018)
[25968.561230] ---[ end trace 6f9ff4b6d492bc6c ]---```

Was OK on 4.19.7+

I've reverted that system to 4.14.90+ for the time being.

popcornmix, 2019-01-07:
Has this always been broken on BRANCH=next or did it work initially (e.g. 4.19.1?)

popcornmix, 2019-01-07:
Can you confirm if 4.19.8 or 4.19.12 worked?

DougieLawson, 2019-01-07:
I never had 4.19.8+ or 4.19.12+ on that machine, it's twenty six days since I've replaced the kernel (been away for New Year). So I'll have to go looking for the Hexxeh commit ids for 4.19.8 and/or 4.19.12.

DougieLawson, 2019-01-07:
4.19.8+ works OK.

DougieLawson, 2019-01-07:
4.19.12+ works OK.

popcornmix, 2019-01-08:
Looks like an upstream regression.
Looking at commits between 4.19.12 and 4.19.13 I don't see any network related ones, but there are a number of "mm:" ones that may be possibilities.

I'll see if I can spot any discussion upstream about this. Hopefully it's been spotted on other platforms and a fix will be forthcoming. If not we could try test builds with commits reverted.

DougieLawson, 2019-01-08:
I found a commit in ./drivers/net/tun.c from a month ago which changes
```
static int tun_validate(struct nlattr *tb[], struct nlattr *data[],
			struct netlink_ext_ack *extack)
{
	if (!data)
		return 0;
	return -EINVAL;
}
```
to

```
static int tun_validate(struct nlattr *tb[], struct nlattr *data[],
			struct netlink_ext_ack *extack)
{
	NL_SET_ERR_MSG(extack,
		       "tun/tap creation via rtnetlink is not supported.");
	return -EOPNOTSUPP;
}
```

I'll have a look at getting OpenVPN installed on another RPi (that isn't used for my normal tunnel), I'll pull the 4.19.13 source and see if that patch is the cause of the issue.

popcornmix, 2019-01-24:
Any update on this?
We are now up to 4.19.17 so there have been a few upstream bumps - any improvement?

DougieLawson, 2019-01-24:
Sorry @popcornmix I've not had any spare time. I'll try and get my test system running 4.19.17 tonight.

I'm taking a 3B with a touch screen away with me on a business trip next week. If I get bored enough in a hotel in Copenhagen I may try OpenVPN on 4.19.17 on that Raspberry.

DougieLawson, 2019-02-18:
Just tested this on BRANCH=master 4.19.23+ and the problem has vanished.

closed 2019-02-18 by DougieLawson