raspberrypi/linux #2799: crng.init blocks boot process without a mouse.

rin67630, 2019-01-08:
**Is this the right place for my bug report?**
I believe that crng.init belongs to the kernel.
It is supposed to provide some entropy to the random generator from the mouse events.
The Raspberry Pi is frequently used headless e.g. over VNC.

**Describe the bug**
VNC does not start **until a mouse is plugged in**.

**To reproduce**
The behaviour does not happen with a pristine install of  2018-11-13-raspbian-stretch, but has been frequently reported, also with previous versions of Raspbian, on all Raspberry Pi models.

**Expected behaviour**
The boot process should perform smoothly **also without a mouse plugged in**.

**Actual behaviour**
During boot the system hangs after the initialisation of fuse: (etc/logs/messages)
Jan  8 16:12:50 Stretch11 kernel: [  316.155022] random: crng init done
Jan  8 16:12:50 Stretch11 kernel: [  316.155036] random: 7 urandom warning(s) missed due to ratelimiting

It can stay at that stage several minutes or forever. **As soon as you plug a mouse, it continues**.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Bug found on many current models incuding Pi3B+, Pi3A+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2018-11-13, but also on previous versions.
* Which firmware version (`vcgencmd version`)?
Nov  4 2018 16:31:07, but also on previous versions.
* Which kernel version (`uname -a`)?
`Linux Stretch11.2018-3APlus 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux`
**Logs**
If applicable, add the relevant output from `dmesg` or similar.
See Additional context.
**Additional context**
Does not happen on very pristine installs, needs additional software installed (did not yet find which one), but has been frequently reported.
Does NEVER happen with a mouse connected. Boot process resumes as soon, as a mouse get connected.

System hangs at: (etc/logs/messages)
```
Jan  8 15:17:14 Stretch11 kernel: [   16.795565] fuse init (API version 7.26)
_[system hangs here, after plugging a mouse it continues]_
Jan  8 16:12:50 Stretch11 kernel: [  316.155022] random: crng init done
Jan  8 16:12:50 Stretch11 kernel: [  316.155036] random: 7 urandom warning(s) missed due to ratelimiting.
Jan  8 16:12:50 Stretch11 vncserver-x11[418]: ServerManager: Server started
Jan  8 16:12:50 Stretch11 vncserver-x11[418]: ConsoleDisplay: Found running X server (pid=458, binary=/usr/lib/xorg/Xorg)
Jan  8 16:12:51 Stretch11 vncserver-x11[418]: Connections: connected: 192.168.188.21::60722
Jan  8 16:12:52 Stretch11 vncserver-x11[418]: Connections: authenticated: 192.168.188.21::60722, as pi (f permissions)
```
(found out through plugging the mouse exactly at 16:12:50)

Workaround solution: install haveged.
Then it boots like a charm! Also without a mouse.

Please consider revising that step of the boot process: 
The Raspberry Pi is predestined to work headless. Seeking for enthalpy  from a mouse is not purposedly.
I have lost days, until I found the workaround and I am not the only one. 
The fora have many similar issues and a lot of users are clueless, fiddling on the wrong gears...

Thank you for considering my issue.

popcornmix, 2019-01-08:
I run multiple Pi's headless without a mouse (as do hundreds of thousands of users).
This is not an issue with raspbian lite or raspbian full images.

I think you'll need to work out what you did that caused it to stop booting without a mouse.

Also not that the Pi has a hardware random number generator which should be able to provide entropy.

rin67630, 2019-01-08:
Ok, I know that most Pi' installs are not showing that behaviour, some are also fine with some minutes until the boot completes, but I am not the only one, who has got massive problems. You will find dozens of threads where people fiddle weeks long at the wrong gears hoping to get it running.
Haveged solves the problem, now my RPi 3A+ reboots in less than a minute without a mouse.

andrum99, 2019-01-09:
This is reported to be an issue when the RealVNC server is used on the Pi 3A+ - see https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=230779 and https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=230026

JamesH65, 2019-01-10:
You say that it does not happen with a pristine install, so what are you doing after install that seems to provoke the issue. Without knowing that it will be difficult to fix anything.  What did you do to fix the issue?

rin67630, 2019-01-13:
I use to make many changes to get my measuring stations online..., so looking back to what harms is a exhausting task, since you need to reboot each time to check.
Anyhow installing **haveged** solves **my** problem instantly. I am fine with that, but i see many others reporting trouble as well, and most have got no clue and will be fiddlling with e.g wpa_supplicant unless their system is completely screwed...
Having a mouse connected solves the probelm as well, but that ist not the purpose of a headless unit.

It looks like if the crng.init and urandom process can just recover without a mouse if not too much other processes are in the way, or what else means "urandom warning(s) missed due to ratelimiting" ?

jktomer, 2019-02-21:
I started having a closely related problem (networking services unavailable until mouse plugged in; my router did not even believe it was connected, though the ethernet lights on the RPi and the switch it was plugged into were on and blinking) when upgrading my RPi to buster tonight. Not sure if this is a coincidence, but I suspect not, as before the upgrade my RPi rebooted due to a power outage in the house and came up headless without issues.

The problem did not manifest in the same way as the OP's report; the system did boot successfully, but dhcpcd.service failed to start after a 93s timeout, leaving networking unavailable. (Including a static configuration stanza in dhcpcd.conf did NOT help, surprisingly: all networking remained unavailable until a keyboard and mouse were connected).

Here is a [syslog snippet](https://gist.github.com/jktomer/0270c69e17b0ba714e0257f5cc899ed5) from one such boot.

I solved the problem by installing rng-tools and the device works perfectly headless now. Perhaps rng-tools should be installed by default on raspbian: it's only 160KiB installed and headless operation is extremely common on these systems.

I suspect that either some part of networking initialization has sprouted an inappropriate dependency on /dev/random or stretch used to have an alternative /dev/random entropy source that has disappeared in buster.

rin67630, 2019-03-28:
> The problem did not manifest in the same way as the OP's report; the system did boot successfully, but dhcpcd.service failed to start after a 93s timeout, leaving networking unavailable. ...
> 

Just to mention: in the cases I reported, _the system always booted up to the GUI_, but VNC were unable to start.

JamesH65, 2019-03-29:
Can you try `sudo apt install haveged` to see if that helps?

andrum99, 2019-03-29:
> I think you'll need to work out what you did that caused it to stop booting without a mouse.
> 
> Also not that the Pi has a hardware random number generator which should be able to provide entropy.

It does, but it looks like at the moment on Stretch this is only appearing as /dev/hwrng and isn't getting hooked up as an entropy source for the kernel: installing rng-tools then either rebooting or manually starting the rng-tools service is enough to get the kernel using it, so reads from /dev/random don't block.

popcornmix, 2019-03-29:
@XECDesign is rng-tools enabled in raspbian by default?

XECDesign, 2019-03-29:
No, and I'm a bit surprised that /dev/random doesn't combine data from hwrng. I'll need to read up on it a bit before adding it to the default install, I think.

popcornmix, 2019-03-29:
I believe rng-tools or similar is needed to feed /dev/hwrng to /dev/random.
The kernel driver alone isn't enough.

XECDesign, 2019-03-29:
> While Linux itself uses the result from TRNG in /dev/random, if available, they are only used as a XOR after the entropy is collected by kernel. So /dev/random, by default, is slow even if you do have a TRNG. rngd feeds /dev/random itself, increasing the available entropy by far. 

https://wiki.archlinux.org/index.php/Rng-tools

From what I gather, the standard practice is not to include rng-tools in distros, but I couldn't find why. It could be that if you end up relying too much on the hwrng, which people can be suspicious of.

Either way, I don't see why we shouldn't have it in Raspbian.

popcornmix, 2019-03-29:
Yes, I have it installed on my setup and it seems to work fine.
I can't think of a reason not to (the hwrng driver is not much use without it).

XECDesign, 2019-03-29:
Yup, added to pi-gen.

Ferroin, 2019-04-01:
At least part of the argument against `rng-tools` is that it does NIST 'randomness' tests on the data before injecting it, without any option to turn off this behavior.  These tests actually check for cryptographic randomness, not mathematical randomness, and therefore can strip a non-negligible amount of good entropy form the data being added.  They also take a non-negligible amount of time to run.

popcornmix, 2019-04-01:
The hardware produces about 50K 32-bit random words a second, so losing entropy probably isn't an issue. It might be worth checking if it slows down boot.

schlamar, 2019-05-08:
I have seen a similar issue. My Python service did hang at `os.urandom` for a long time. Could this be related to this issue? Should installing `rng-tools` solve this issue?

popcornmix, 2019-05-10:
@schlamar it's certainly possible. Why not try it and report back?

andrum99, 2019-05-10:
Suggest this bug can be closed once the next official build of Raspbian comes out, since it will include rng-tools.

XECDesign, 2019-05-10:
The last release already has it

andrum99, 2019-05-10:
Time to close this one then 😀

JamesH65, 2019-09-19:
Closing this issue as questions answered/issue resolved.

closed 2019-09-19 by JamesH65