raspberrypi/linux #2800: CONFIG_VIDEO_GO7007

ShorTie8, 2019-01-09:
This is not a issue, but more of a question.
Is CONFIG_VIDEO_GO7007 really needed for the pi ??
Searching linux, I see it only being built by the bcm defconfig's
`grep -r 'GO7007' sources/linux/linux/arch`

I only ask this because they have dropped all firmware from the kernel.
So what we do is search linux for all needed firmware,
`find ${MOD_DIR} -name \*.ko | while read module; do`
and only include those that are needed out of the 200megs worth of linux-firmware.

That is actually how I found it being built, by looking thru buildroot/linux-firmware to see what all twas being included.

popcornmix, 2019-01-09:
The help text says "This is a video4linux driver for the WIS GO7007 MPEG encoder chip"
It appears to be something you can connect through USB, so is potentially being used by Pi users.

Are you suggesting it is no longer support by upstream linux? Any links for that?

ShorTie8, 2019-01-09:
Ya read that,
Looks like it is a some kind of analog to digital video converter.
I'd say it is still supported as it compiles, no way to test if it works. 

Just found it interesting that no other defconfig has it as 'not set' or something.
It's firmware is only 186k, So no big'y.

popcornmix, 2019-01-10:
It's enabled on my Ubnuntu machine.

closed 2019-01-11 by ShorTie8