raspberrypi/linux #2807: How to download the driver to raspberry pi

janzhen96, 2019-01-14:
Hi everyone. I'm very new to Linux and raspberry pi.

I have an inquiry regarding installing the driver to my raspberry pi. The link below is the driver that I am interested in using. I am interested in using usbtmc driver to connect it to my instrument. 

https://github.com/raspberrypi/linux/tree/rpi-4.14.y/drivers/usb/class  

So, how do i download this and install it to my raspberry pi?  I look through a lof of website but I'm still lost. 

Thanks in advance.
Jan Zhen

pelwell, 2019-01-14:
Done via https://github.com/raspberrypi/firmware/issues/929 - see https://github.com/raspberrypi/linux/commit/b3a4c69c56f4db99b091305a33aa7c9b090aaad1. Now you just have to wait for the next firmware release, which should be in the next few days, and run `sudo rpi-update`.

pelwell, 2019-01-14:
rpi-update should be sufficient - it will replace your firmware and kernel (and modules) with the most recent version.

I have no experience with the USB_TMC module or with devices that require it, but it appears to be a class driver so it ought to be loaded automatically when an appropriate device is plugged in.

popcornmix, 2019-01-14:
Available in latest rpi-update kernel.

JamesH65, 2019-09-19:
Closing this issue as questions answered/issue resolved.
Please update to the latest kernel which may contain a fix for this issue.

closed 2019-09-19 by JamesH65