raspberrypi/linux #2809: Wrong string descriptors in OTG mode (g_mass_storage)

mrcg, 2019-01-14:
**Describe the bug**
When checking several different USB devices which are compatible with my 3d printer the string descriptors are always configured like this.
```
iManufacturer .... 0x01 (String Descriptor 1)
iProduct .... 0x02 (String Descriptor 2)
iSerialNumber ... 0x03 (String Descriptor 3)
```
This can be verified with "dmesg" command on linux or usbtreeview gui on windows.
When emulating the usb share on Raspberry Pi Zero W device the string descriptors are not in the same (correct) order.

iManufacturer is 0x03 insted of 0x01
iProduct is 0x04 instead of 0x02
iSerialNumber is 0x05 instead of 0x03

**To reproduce**
The usb device emulation is configured like this:
`sudo modprobe g_mass_storage file=/usbdisk.img stall=0 removable=1 idProduct=0x1000 bcdDevice=0x1000 iManufacturer="PiShare" iProduct="3dPrinterShare" iSerialNumber=12345678912345`

**Expected behaviour**
usbtreeview of emulated usb share should look like this:
```
iManufacturer            : 0x01 (String Descriptor 1)
 Language 0x0409         : "PiShare"
iProduct                 : 0x02 (String Descriptor 2)
 Language 0x0409         : "3dPrinterShare"
iSerialNumber            : 0x03 (String Descriptor 3)
 Language 0x0409         : "12345678912345"
```

**Actual behaviour**
usbtreeview of emulated usb share looks like this:
```
iManufacturer            : 0x03 (String Descriptor 3)
 Language 0x0409         : "PiShare"
iProduct                 : 0x04 (String Descriptor 4)
 Language 0x0409         : "3dPrinterShare"
iSerialNumber            : 0x05 (String Descriptor 5)
 Language 0x0409         : "12345678912345"
```

**System**
Model: Raspberry Pi Zero W
OS: 2017-07-05-raspbian-jessie-lite.img

**Logs**
See actual behaviour

Is this issue related to g_mass_storage implementation (bug?) or is this also user configurable?
Thanks for support.

pelwell, 2019-01-14:
What makes you think that string descriptor assignments have to follow a fixed pattern?

mrcg, 2019-01-14:
I have checked more than ten HW shares which are working on the 3d printer and the string descriptor is as described. The emulated device is properly working on windows machine (with the exception that windows is always asking to repair the device when mounted) put it's not recognized from the 3d printer. I have checked really everything but the string descriptors are the only difference which i can find compared to the HW devices. Not sure if this is the real root cause but i wanted to sort this out to be sure.

ED6E0F17, 2019-01-17:
In testing I found an "ordinary" USB stick that would not be accepted on my PS3. I had to use "fdisk" to change the partition "type" from "83" (linux) to "c" (vfat-LBA) before it was recognised . This may not help, but it is worth mentioning.

It seems like an odd design choice in the kernel that the lower string numbers appear not to be used, but it does not look like a mistake.

mrcg, 2019-01-20:
The partition is not the issue. I have already tried various formats without success. I am using the same format as used on the sticks which are properly working.

Is there any possibility to change the string descriptor ID manually to match with the sequence as used on all of the tested USB storage devices?

ED6E0F17, 2019-01-22:
You cannot change the way that the composite mass storage gadget works. The legacy file storage gadget that was removed after Linux 3.7 may have worked in the way that you want.

I cannot suggest anything better than if you could clone your smallest USB stick to a file, and put that on your Pi Zero storage gadget.

JamesH65, 2019-09-19:
@mrcg Did you find any solution to this? AFAICT, it's not actually a fault as described.

mrcg, 2019-09-19:
Unfortunately i did not find any solution for this.