raspberrypi/linux #2811: f2fs issue ... 5.0.0-rc2 on rpi3b+

jeffersonchua, 2019-01-16:
rpi3b+. Running latest git version. Filesystem is mounted read-only.

Hardware name: BCM2835
Workqueue: writeback wb_workfn (flush-179:0)
PC is at f2fs_balance_fs_bg+0x26c/0x2a8
LR is at f2fs_build_free_nids+0x394/0x510
pc : [<803187e4>]    lr : [<803120b8>]    psr: 60030013
sp : aea6bd50  ip : aeaa3980  fp : aea6bee0
r10: 80a03d00  r9 : 80a09c88  r8 : ae5740ec
r7 : aea6bdf0  r6 : 80a09c88  r5 : 80a09c88  r4 : ae2e8000
r3 : 00000000  r2 : ae03df00  r1 : 00000000  r0 : 00000000
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Control: 10c5383d  Table: 2ae8806a  DAC: 00000055
Process kworker/u8:3 (pid: 788, stack limit = 0xf3aa5805)
Stack: (0xaea6bd50 to 0xaea6c000)
bd40:                                     af79d960 802f222c af79d960 ae598828
bd60: ae2e8000 148e697c 00000000 ae2e8000 00000000 80311ae0 aea6bee0 802f2774
bd80: ae969cc0 aea6bdf0 ae598914 148e697c aea8ca4c aea6bdf0 ae5740ec aea6bdf0
bda0: aea8ca4c 801b7f34 ae574000 aea8ca38 aea6bdf0 80222e94 00000000 addb9188
bdc0: ae5740a8 aea8ca38 ae574000 aea8ca4c aea25800 802231b8 00000001 aea6a000
bde0: ffff97c8 ae574068 00000000 00001000 00001000 00000000 00000000 00000000
be00: ffffffff 7fffffff 00000000 00000011 80a09c88 148e697c ffbdacc8 aea8ca38
be20: aea25800 ae5988d0 00000000 aea6bee0 aea8ca4c 80a03d00 ffff97d2 802234d0
be40: 00000007 aea8ca38 aea6bee0 ffff97c8 ffff97c8 0000000a 80a03d00 80a14950
be60: aea8ca64 802236b0 ffff0a00 aea8cb14 00000e14 80a09c88 ae8e2180 ffff8c10
be80: aea6be80 aea6be80 aea6be88 aea6be88 ae000000 148e697c aea8ca3c aea8cb1c
bea0: aea8cb10 aea8ca38 aea8cb14 aea8ca3c 80a1494c 80a03d00 00000e14 80224274
bec0: 00000000 80a09c88 2e879000 aeaa3980 ad4ea018 aeaa3980 af1bac00 ae969cc0
bee0: 00000e14 00000000 aea6be7c 00000000 00000006 00000003 00000000 00000000
bf00: 00000000 148e697c aea2551d aea8cb1c ae8e2180 ae806200 ae805a00 00000000
bf20: aea8cb20 00000000 ae806200 80135984 00000088 80a03d00 ae8e2180 ae806200
bf40: ae8e2194 00000088 80a03d00 ae806218 aea6a000 80136720 aea6a000 ae8e4380
bf60: 00000000 ae8e43c0 aea6a000 ae8e4380 00000000 ae8e2180 801366f0 ae975eb4
bf80: ae8e43dc 8013b370 00000001 ae8e4380 8013b26c 00000000 00000000 00000000
bfa0: 00000000 00000000 00000000 801010d8 00000000 00000000 00000000 00000000
bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
bfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[<803187e4>] (f2fs_balance_fs_bg) from [<80311ae0>] (f2fs_write_node_pages+0x40/0x148)
[<80311ae0>] (f2fs_write_node_pages) from [<801b7f34>] (do_writepages+0x20/0x7c)
[<801b7f34>] (do_writepages) from [<80222e94>] (__writeback_single_inode+0x2c/0x1b0)
[<80222e94>] (__writeback_single_inode) from [<802231b8>] (writeback_sb_inodes+0x1a0/0x444)
[<802231b8>] (writeback_sb_inodes) from [<802234d0>] (__writeback_inodes_wb+0x74/0xac)
[<802234d0>] (__writeback_inodes_wb) from [<802236b0>] (wb_writeback+0x1a8/0x1fc)
[<802236b0>] (wb_writeback) from [<80224274>] (wb_workfn+0x178/0x33c)
[<80224274>] (wb_workfn) from [<80135984>] (process_one_work+0x204/0x418)
[<80135984>] (process_one_work) from [<80136720>] (worker_thread+0x30/0x59c)
[<80136720>] (worker_thread) from [<8013b370>] (kthread+0x104/0x134)
[<8013b370>] (kthread) from [<801010d8>] (ret_from_fork+0x14/0x3c)
Exception stack(0xaea6bfb0 to 0xaea6bff8)
bfa0:                                     00000000 00000000 00000000 00000000

My best,
Jeff

pelwell, 2019-01-16:
You've trimmed the start of the kernel exception log - what were the first few lines?

lategoodbye, 2019-01-16:
I think this should be reported to upstream:
Jaegeuk Kim <jaegeuk@kernel.org>
Chao Yu <yuchao0@huawei.com>
linux-f2fs-devel@lists.sourceforge.net

jeffersonchua, 2019-01-16:
> I think this should be reported to upstream:

Ok, I shall do that!

Thanks,
Jeff

jeffersonchua, 2020-01-27:
Please close. With Linux 5.5.0-rc7, it's no longer an issue.

Thanks,
Jeff

closed 2020-01-27 by jeffersonchua