raspberrypi/linux #2815: -rt kernel crash 4.14.91-rt49+

adambarta, 2019-01-17:
cmdline contains

dwc_otg.lpm_enable=0 dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0 dwc_otg.nak_holdoff=0 console=serial0,115200 elevator=deadline

[    2.535232] irq 53: nobody cared (try booting with the "irqpoll" option)
[    2.535250] CPU: 0 PID: 82 Comm: irq/53-bcm2835- Not tainted 4.14.91-rt49+ #2
[    2.535254] Hardware name: BCM2835
[    2.535259] Backtrace: 
[    2.535312] [<c00131b4>] (dump_backtrace) from [<c0013480>] (show_stack+0x18/0x1c)
[    2.535326]  r7:00000035 r6:d700c600 r5:d700c600 r4:d700c600
[    2.535359] [<c0013468>] (show_stack) from [<c06345dc>] (dump_stack+0x20/0x28)
[    2.535391] [<c06345bc>] (dump_stack) from [<c005dc44>] (__report_bad_irq+0x2c/0x118)
[    2.535410] [<c005dc18>] (__report_bad_irq) from [<c005dfec>] (note_interrupt+0x23c/0x294)
[    2.535426]  r9:d700a000 r8:d6415eb0 r7:00000035 r6:d700c600 r5:00000000 r4:d700c600
[    2.535443] [<c005ddb0>] (note_interrupt) from [<c005b56c>] (handle_irq_event_percpu+0x6c/0x84)
[    2.535459]  r10:ffffe000 r9:d700a000 r8:d6415eb0 r7:00000002 r6:d700c600 r5:00000000
[    2.535465]  r4:c005c57c r3:00000000
[    2.535481] [<c005b500>] (handle_irq_event_percpu) from [<c005b5d8>] (handle_irq_event+0x54/0x98)
[    2.535493]  r7:00000001 r6:00000000 r5:c09b6114 r4:d700c600
[    2.535510] [<c005b584>] (handle_irq_event) from [<c005ea20>] (handle_level_irq+0xb0/0x17c)
[    2.535518]  r5:c09b6114 r4:d700c600
[    2.535535] [<c005e970>] (handle_level_irq) from [<c005a5dc>] (generic_handle_irq+0x28/0x3c)
[    2.535542]  r5:c09b6114 r4:00000035
[    2.535557] [<c005a5b4>] (generic_handle_irq) from [<c005acb8>] (__handle_domain_irq+0x64/0xdc)
[    2.535574] [<c005ac54>] (__handle_domain_irq) from [<c0009420>] (bcm2835_handle_irq+0x34/0x3c)
[    2.535589]  r9:d6414000 r8:c005c674 r7:d6415ee4 r6:ffffffff r5:c0987b84 r4:d6415eb0
[    2.535615] [<c00093ec>] (bcm2835_handle_irq) from [<c064de18>] (__irq_svc+0x58/0xa0)
[    2.535623] Exception stack(0xd6415eb0 to 0xd6415ef8)
[    2.535635] 5ea0:                                     c0987b84 d8800210 d700c600 00400000
[    2.535649] 5ec0: d700c600 d6bdba00 d700c610 ffffe000 c005c674 c09b6ab9 ffffe000 d6415f1c
[    2.535660] 5ee0: d6415ed0 d6415f00 c005e95c c005c57c 80000113 ffffffff
[    2.535668]  r5:80000113 r4:c005c57c
[    2.535686] [<c005c50c>] (irq_finalize_oneshot.part.0) from [<c005c6ec>] (irq_forced_thread_fn+0x78/0xa0)
[    2.535698]  r7:d700c600 r6:00000000 r5:d6bdba00 r4:d700c600
[    2.535714] [<c005c674>] (irq_forced_thread_fn) from [<c005c9d0>] (irq_thread+0x104/0x1d4)
[    2.535725]  r7:d700c600 r6:00000001 r5:ffffe000 r4:d6bdba00
[    2.535749] [<c005c8cc>] (irq_thread) from [<c00409c8>] (kthread+0x144/0x15c)
[    2.535765]  r10:d7111c64 r9:c005c8cc r8:d6bdba00 r7:d6414000 r6:00000000 r5:d6bdd2e0
[    2.535769]  r4:d6bdba40
[    2.535788] [<c0040884>] (kthread) from [<c000faa0>] (ret_from_fork+0x14/0x34)
[    2.535803]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0040884
[    2.535808]  r4:d6bdd2e0
[    2.535810] handlers:
[    2.535820] [<c005b61c>] irq_default_primary_handler threaded [<c02c789c>] bcm2835_auxirq_handler
[    2.535849] Disabling IRQ #53

TiejunChina, 2019-01-18:
I have not seen this on my pi3 b. But you might try this https://patchwork.kernel.org/cover/10741389/

pelwell, 2019-01-18:
We can backport when that series (or just that commit) is accepted upstream.

adambarta, 2019-01-18:
Yes this is on PI Zero W on PI 3 and PI Zero non W everything works.

JamesH65, 2019-09-19:
It appears that the patches referred to have not yet been accepted upstream, so not sure how we proceed on this.