raspberrypi/linux #2817: drivers/mt76: mt7610u usb wireless dongle works abnormally on raspberrypi 3b aarch64

isjerryxiao, 2019-01-19:
**Describe the bug**
The new mt76 kernel driver is very unstable on my raspberrypi 3b.   
There are 3 symptoms:  
1. random kernel panic (see logs below)
2. network interface not showing up
3. interface shows up, but scan returns nothing  (see logs below)  

**Steps to reproduce**
- plug in the dongle, it should work.
- reboot several times and you may get one of the three symptoms above.
- multiple reboots can't fix the problem
- when you unplug and replug the dongle, it'll work.
- after several days, the dongle stops working and I get symptom No.3

My dongle:
`Bus 001 Device 006: ID 148f:761a Ralink Technology, Corp. MT7610U ("Archer T2U" 2.4G+5G WLAN Adapter`
However, mt76 works **fine** on my x86 machines, and another aarch64 dev board(amlogic s905d, running kernel 4.19).

**System**

* Pi 3B
* archlinux arm aarch64 user space
* self-compiled kernel, using [raspberrypi/linux](https://github.com/raspberrypi/linux) as source.
* linux-firmware of archlinux (from kernel.org)
* kernel version `4.19.14-1-ARCH+` at this [commit](https://github.com/raspberrypi/linux/commit/7e312d57b01683ee93699fdac1121d5cc62fb211)

**Logs**
- symptom No.1
[kmesg](https://ptpb.pw/v9lW) 
- symptom No.3
[dmesg](https://ptpb.pw/yxg4)  [`ip a`](https://ptpb.pw/qtGQ)  [`iw dev`](https://ptpb.pw/kOm0)  [`iw phy`](https://ptpb.pw/t5to)
After replugging
[dmesg](https://ptpb.pw/cdbf)  [`ip a`](https://ptpb.pw/WX40)  [`iw dev`](https://ptpb.pw/buqv)  [`iw phy`](https://ptpb.pw/Evno)
Also here's a [diff](https://ptpb.pw/UHi8.html) of the kernel debugfs.

Is it an upstream issue or it's related to rpi? Thanks.

lategoodbye, 2019-01-26:
AFAIK arm64 isn't actively support here. So i think there are two options for you:
1. Try to reproduce it with rpi-4.19 (arm32) and add your results here
2. Try to reproduce it with a recent kernel.org 4.20/5.0 (arm64) an report it to linux-netdev/linux-usb mailing list

isjerryxiao, 2019-01-26:
@lategoodbye Thanks. I've got a new rtl8812au dongle. Closing this issue.

lategoodbye, 2019-01-29:
@isjerryxiao Do you still have the Archer T2U? Could you please provide `lsusb -v` on a working system (amlogic s905d, x86)?

lategoodbye, 2019-01-29:
Thanks. It looks like the mt76 driver is using 8 USB endpoints which is propably more than the BCM2835 can offer.

isjerryxiao, 2019-01-29:
Thanks for your information.

isjerryxiao, 2019-01-31:
I also have two usb sound cards attached to the raspberrypi, unable to record audio when fiq_fsm is enabled.

isjerryxiao, 2019-01-31:
according to this [commit](https://github.com/raspberrypi/linux/commit/1c3155cbaed244faeaa01a544103c44fdb620586), It seems that fiq is not ported to arm64, but adding `dwc_otg.fiq_fsm_enable=0` does help.

isjerryxiao, 2019-01-31:
@P33M no, one at a time.

P33M, 2019-01-31:
On aarch64 without FIQ support, the realtime guarantees necessary for USB split transactions to be correctly performed will not be met. Unless Linux gets FIQ support on AARCH64, the recommendation is to stick with our supported (shipped) kernel.

closed 2019-01-31 by P33M

lategoodbye, 2019-02-09:
I was reporting the probing issue to [linux-wireless](https://marc.info/?l=linux-wireless&m=154971413730974) and was pointed to the following series:
https://patchwork.kernel.org/cover/10764453/

The author mentioned that the series also improve the situation with the foundation kernel.