raspberrypi/linux #2818: Please add MCP23008 I2C I/O extender overlay

soosp, 2019-01-19:
**Describe the bug**
There is an overlay for MCP23017 but there is no one for MCP23008
The kernel driver (pinctrl_mcp23s08) is present and properly loaded by the following command:
echo mcp23008 0x20 > /sys/class/i2c-adapter/i2c-1/new_device

**To reproduce**
ls /boot/overlays/mcp23*
/boot/overlays/mcp23017.dtbo  /boot/overlays/mcp23s17.dtbo

**Expected behaviour**
The prsence of the overlay.

**Actual behaviour**
Missing overlay

**System**
* Which model of Raspberry Pi?
Raspberry Pi 3 model B

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2

* Which firmware version (`vcgencmd version`)?
Nov  4 2018 16:31:07
Copyright (c) 2012 Broadcom
version ed5baf9520a3c4ca82ba38594b898f0c0446da66 (clean) (release)

* Which kernel version (`uname -a`)?
Linux pi3 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

pelwell, 2019-01-21:
[This modified mcp23017 overlay](https://drive.google.com/file/d/1OgIN7ysx7kxNqS08eS2Ej4Z3MFUt655a/view?usp=sharing) accepts an extra parameter - `mcp23008` - that should make it compatible with the MCP23008. Please download it, copy it to /boot/overlays (you'll need to `sudo cp mcp23017.dtbo /boot/overlays`...), then edit your config.txt to include the new parameter, e.g.:
```
dtoverlay=mcp23017,mcp23008,addr=0x20,gpiopin=4
```
That can become the new mcp23017 overlay if it works for you.

soosp, 2019-01-21:
Thank you for quick response!
I made a quick test, and it works.

My /boot/config.txt entry:
```
# Enable MCP23008 I/O expander
dtoverlay=mcp23017,mcp23008,addr=0x20
```
The result:
```
lsmod | grep mcp23
pinctrl_mcp23s08       24576  8
```
```
ls -1 /sys/class/gpio/
export
gpio504
gpio505
gpio506
gpio507
gpio508
gpio509
gpio510
gpio511
gpiochip0
gpiochip100
gpiochip128
gpiochip504
unexport
```

I tested some gpios to toggle relays and it worked for me.

pelwell, 2019-01-22:
Thanks - that's now merged, and will be in future releases.

closed 2019-05-08 by pelwell