raspberrypi/linux #2819: Please add overlay for MCP3422/3/4 ADCs

soosp, 2019-01-19:
**Describe the bug**
There is no overlay for these chips but they are supported by linux kernel.
The command
echo mcp3422 0x68 > /sys/class/i2c-adapter/i2c-1/new_device
loads the proper kernel module (mcp3422)

**To reproduce**
ls /boot/overlays/mcp34*
ls: cannot access '/boot/overlays/mcp34*': No such file or directory

**Expected behaviour**
Overlay available.

**Actual behaviour**
Overlay is missing.

**System**
* Which model of Raspberry Pi?
Raspberry Pi 3 model B

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2

* Which firmware version (`vcgencmd version`)?
Nov  4 2018 16:31:07
Copyright (c) 2012 Broadcom
version ed5baf9520a3c4ca82ba38594b898f0c0446da66 (clean) (release)

* Which kernel version (`uname -a`)?
Linux pi3 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

pelwell, 2019-01-21:
There's a trial mcp342x overlay [here](https://drive.google.com/file/d/1WSreBcu-8UZhKb0WFlvQxOwSaIEVzpX0/view?usp=sharing). The README will say:
```
Name:   mcp342x
Info:   Overlay for activation of Microchip MCP3421-3428 ADCs over I2C
Load:   dtoverlay=mcp342x,<param>=<val>
Params: addr                    I2C bus address of device, for devices with
                                addresses that are configurable, e.g. by
                                hardware links (default=0x68)
        mcp3421                 The device is an MCP3421
        mcp3422                 The device is an MCP3422
        mcp3423                 The device is an MCP3423
        mcp3424                 The device is an MCP3424
        mcp3425                 The device is an MCP3425
        mcp3426                 The device is an MCP3426
        mcp3427                 The device is an MCP3427
        mcp3428                 The device is an MCP3428
```

pelwell, 2019-01-22:
Thanks - that's now merged, and will be in future releases.

closed 2019-05-08 by pelwell