raspberrypi/linux #2821: CM3 Freeze on reboot mmc0: timeout waiting for hardware interrupt

kkoopa, 2019-01-22:
I have a CM3 on a custom IO board with a USB2513B hub. I have built an image with Yocto and the meta-raspberrypi layer. The system locks up on each attempt to reboot and only a hard power cycle will get it running again.

If I kill power to the USB2513B prior to attempting a reboot, the success rate increases, but similar problems with timeouts waiting for hardware interrupts then occur more randomly in other contexts.

This issue occurs regardless of kernel versions, config or VC4 firmware, tested with 4.14, 4.18, 4.19 and 4.20 kernels. I have however not noticed this behavior with a stock Raspbian image.

Here is the output with a 4.19.14 kernel and matching firmware:
```
root@raspberrypi-cm3:~# reboot

INIT: Sending processes the TERM signal
root@raspberrypi-cm3:~# Stopping Dropbear SSH server: stopped /usr/sbin/dropbear (pid 374)
dropbear.
Stopping system message bus: dbus.
Stopping syslogd/klogd: stopped syslogd (pid 380)
stopped klogd (pid 383)
done
ALSA: Storing mixer settings...
Deconfiguring network interfaces... ifdown: interface eth0 not configured
done.
Stopping watchdog daemon...Sending all processes the TERM signal...
logout
Sending all processes the KILL signal...
Unmounting remote filesystems...
Deactivating swap...
Unmounting local filesystems...
[   64.438779] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Rebooting... [   66.858107] sdhost_busy_irq: intmask 440, data   (null)
[   77.271647] mmc0: timeout waiting for hardware interrupt.
[   77.277131] [04975608] TCM> bc2d3798 0
[   77.280934] [04975609] FDA> bc2d36b8 0
[   77.284736] [0497560a] IRQ> 0 0
[   77.287922] [0497560f] TSK< bc2d36b8 0
[   77.291724] [0497561e] TSK> bc2d36b8 0
[   77.295527] [8497d4a3] REQ< bc2d98b8 10801
[   77.299682] [8497d4a5] CMD< 17 2
[   77.302956] [8497d4ad] FCM< bc2d98b8 bc2d98fc
[   77.307375] [8497d4af] RSP  900 0
[   77.310736] [8497d4b0] CMD< 12 4943c
[   77.314362] [8497d4b2] CMDD 2 200
[   77.317723] [8497d4b3] FCM< bc2d98b8 bc2d9930
[   77.322141] [8497d4b6] RSP  900 0
[   77.325502] [8497d4b6] FCM> bc2d98b8 0
[   77.329304] [8497d4b7] FCM> bc2d98b8 0
[   77.333106] [8497d4b8] CMD  12 200
[   77.336555] [8497d4b8] REQ> bc2d98b8 0
[   77.340357] [0497d5a6] IRQ< 1 0
[   77.343542] [0497d5a6] IRQD bc2d9998 1
[   77.347344] [0497d5a8] XFP< bc2d9998 2
[   77.351146] [0497d5c2] XFP> bc2d9998 2
[   77.354949] [0497d5c3] IRQ> 1 0
[   77.358133] [0497d5c9] IRQ< 1 0
[   77.361317] [0497d5c9] IRQD bc2d9998 1
[   77.365119] [0497d5ca] XFP< bc2d9998 1
[   77.368921] [0497d5df] XFP> bc2d9998 1
[   77.372722] [0497d5e0] FDA< bc2d98b8 0
[   77.376524] [0497d5e1] TCM< bc2d9998 0
[   77.380326] [0497d5e2] WTC< 10804 0
[   77.383864] [0497d5e3] WTC> 10804 0
[   77.387402] [0497d5e4] TCM> bc2d9998 0
[   77.391203] [0497d5e5] FDA> bc2d98b8 0
[   77.395006] [0497d5e5] IRQ> 0 0
[   77.398190] [0497d5e9] TSK< bc2d98b8 0
[   77.401992] [0497d5f6] TSK> bc2d98b8 0
[   77.405794] [0498430d] REQ< bc2d10b8 10801
[   77.409948] [0498430f] CMD< 17 2
[   77.413220] [04984317] FCM< bc2d10b8 bc2d10fc
[   77.417638] [04984319] RSP  900 0
[   77.421000] [0498431a] CMD< 12 42af2
[   77.424624] [0498431c] CMDD 2 200
[... 1037 lines cut ...]
[  112.770885] [071977c9] IRQB bb1dbcd8 440
[  112.774863] [07198c82] IRQ> 40 0
[  112.778136] [87b9af45] TIM< 0 0
[  112.781326] mmc0:>cmd op 6 arg 0x3200101 flags 0x49d - resp 00000000 00000000 00000000 00000000, err -110
[  112.791031] mmc0: =========== REGISTER DUMP ===========
[  112.796330] mmc0: SDCMD  0x00004806
[  112.799865] mmc0: SDARG  0x03200101
[  112.803400] mmc0: SDTOUT 0x017d7840
[  112.806935] mmc0: SDCDIV 0x00000003
[  112.810470] mmc0: SDRSP0 0xffffffff
[  112.814005] mmc0: SDRSP1 0x0000ff7f
[  112.817540] mmc0: SDRSP2 0xffffffff
[  112.821075] mmc0: SDRSP3 0xffffffff
[  112.824611] mmc0: SDHSTS 0x00000040
[  112.828145] mmc0: SDVDD  0x00000001
[  112.831681] mmc0: SDEDM  0x00010801
[  112.835216] mmc0: SDHCFG 0x0000040e
[  112.838752] mmc0: SDHBCT 0x00000006
[  112.842288] mmc0: SDHBLC 0x00000001
[  112.845822] mmc0: ===========================================
[  112.851703] mmc0: cache flush error -110
[  112.855688] mmc0: error -110 during shutdown
[  112.855908] mmc0: card 0001 removed
[  112.860510] reboot: Restarting system
```

pelwell, 2019-01-22:
With custom hardware and multiple levels of custom software you shouldn't expect much support, but this doesn't look right:
```
Unmounting remote filesystems...
Deactivating swap...
Unmounting local filesystems...
[   64.438779] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Rebooting... [   66.858107] sdhost_busy_irq: intmask 440, data   (null)
```
Why is the OS remounting the SD card when it's supposed to be shutting down?

kkoopa, 2019-01-31:
The problem stemmed from the eMMC getting disabled for some reason if USB boot was enabled.

closed 2019-01-31 by kkoopa