raspberrypi/linux #2831: Kernel oops on `cat /sys/kernel/debug/bcm2835_thermal/regset`

yumkam, 2019-01-27:
(After Oops, system had problems with "normal" reboot (had to `SysRq+EUSB` from keyboard))

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi3B+

* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Nov  4 2018 16:35:17 
Copyright (c) 2012 Broadcom
version ed5baf9520a3c4ca82ba38594b898f0c0446da66 (clean) (release)
```
* Which kernel version (`uname -a`)?
```
Linux raspberrypi 4.14.92-v7+ #1 SMP Sun Jan 13 00:10:55 MSK 2019 armv7l GNU/Linux

$ cat /proc/version
Linux version 4.14.92-v7+ ($user@$host) (gcc version 6.3.0 20170516 (Debian 6.3.0-18)) #1 SMP Sun Jan 13 00:10:55 MSK 2019
```
(self-cross-compiled from commit 4de3f6305841c2121257b8fdbda5c3c0bf77e669 at debian/stretch with debian's `gcc-6-arm-linux-gnueabihf_6.3.0-18cross1`, with minor `.config` tweaks)

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
```
[  379.082304] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[  379.087712] pgd = b41c0000
[  379.093049] [00000000] *pgd=3415e835, *pte=00000000, *ppte=00000000
[  379.098322] Internal error: Oops: 17 [#1] SMP ARM
[  379.103572] Modules linked in: lan78xx2(O) bnep hci_uart btbcm serdev bluetooth ecdh_generic nf_log_ipv6 ip6t_REJECT nf_reject_ipv6 nf_conntrack_ipv6 nf_defrag_ipv6 ip6table_filter ip6_tables xt_tcpudp nf_log_ipv4 nf_log_common xt_LOG ipt_REJECT nf_reject_ipv4 nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack nf_conntrack iptable_filter joydev evdev brcmfmac(O) crc32_arm_ce sha512_arm brcmutil sha256_arm aes_arm snd_bcm2835(C) cfg80211 snd_pcm rfkill snd_timer snd fixed uio_pdrv_genirq uio zbud lz4 lz4_compress ip_tables x_tables ipv6
[  379.122225] CPU: 1 PID: 1107 Comm: cat Tainted: G         C O    4.14.92-v7+ #1
[  379.128661] Hardware name: BCM2835
[  379.134945] task: b3229e00 task.stack: b436a000
[  379.141343] PC is at debugfs_print_regs32+0x78/0xac
[  379.147651] LR is at 0x0
[  379.154019] pc : [<804200d0>]    lr : [<00000000>]    psr: 80000013
[  379.160485] sp : b436bdf8  ip : 00001000  fp : b436be2c
[  379.166888] r10: 00000000  r9 : 8097d328  r8 : 80989d1c
[  379.173394] r7 : 809763c4  r6 : 00000000  r5 : b4305580  r4 : 80848c6c
[  379.179919] r3 : 00000000  r2 : 809a8d94  r1 : 00000000  r0 : 00000000
[  379.186491] Flags: Nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  379.193187] Control: 10c5383d  Table: 341c006a  DAC: 00000055
[  379.199791] Process cat (pid: 1107, stack limit = 0xb436a210)
[  379.206478] Stack: (0xb436bdf8 to 0xb436c000)
[  379.213092] bde0:                                                       b76e6510 00000002
[  379.219804] be00: 00001000 b4305580 b429bc00 b429bc00 b436bf78 00000000 00000001 014000c0
[  379.226629] be20: b436be44 b436be30 80420134 80420064 809763c4 802cadd8 b436be9c b436be48
[  379.233410] be40: 802cae04 80420110 b4378900 00000000 b43055b0 76b2e000 b436be60 00020000
[  379.240351] be60: 00000000 00000000 00000000 00000000 00000000 802cac28 b429bc00 00000000
[  379.247248] be80: 76b2e000 00020000 80821d50 b436bf78 b436becc b436bea0 8041eb88 802cac34
[  379.254233] bea0: b436bf78 b429bc00 8041eb28 b429bc00 76b2e000 b436bf78 00020000 00000003
[  379.261293] bec0: b436bf44 b436bed0 802a37c4 8041eb34 b436bf1c 80c09384 00000817 807cf448
[  379.268298] bee0: 76b2d000 b436bfb0 00001000 000001ff b436bfac b436bf00 80101200 807cf454
[  379.275364] bf00: b436bf40 b62bf880 76b2d000 00000000 b436bf74 802a3930 80258330 ffffe000
[  379.282337] bf20: 00020000 b429bc00 76b2e000 b436bf78 00000000 00000003 b436bf74 b436bf48
[  379.289404] bf40: 802a3958 802a3794 00000000 802c4388 b429bc00 b429bc00 00000000 00000000
[  379.296483] bf60: 76b2e000 00020000 b436bfa4 b436bf78 802a3f00 802a38c8 00000000 00000000
[  379.303581] bf80: 00020000 7fffe000 00020000 00000003 80108184 b436a000 00000000 b436bfa8
[  379.310751] bfa0: 80107fc0 802a3ec0 00020000 7fffe000 00000003 76b2e000 00020000 00027184
[  379.317847] bfc0: 00020000 7fffe000 00020000 00000003 00000003 00020000 ffffffff 00000000
[  379.325093] bfe0: 00000000 7e847ac4 000141cc 76f410cc 60000010 00000003 00000000 00000000
[  379.332292] [<804200d0>] (debugfs_print_regs32) from [<80420134>] (debugfs_show_regset32+0x30/0x3c)
[  379.339678] [<80420134>] (debugfs_show_regset32) from [<802cae04>] (seq_read+0x1dc/0x504)
[  379.347043] [<802cae04>] (seq_read) from [<8041eb88>] (full_proxy_read+0x60/0x90)
[  379.354527] [<8041eb88>] (full_proxy_read) from [<802a37c4>] (__vfs_read+0x3c/0x134)
[  379.362121] [<802a37c4>] (__vfs_read) from [<802a3958>] (vfs_read+0x9c/0x164)
[  379.369709] [<802a3958>] (vfs_read) from [<802a3f00>] (SyS_read+0x4c/0xa0)
[  379.377422] [<802a3f00>] (SyS_read) from [<80107fc0>] (ret_fast_syscall+0x0/0x28)
[  379.385049] Code: 0a000000 ebfaacfc e914000c e08a3003 (e5933000) 
[  379.392900] ---[ end trace bdfa3c6c437dd1d4 ]---
```
**Additional context**
`objdump -dSr /path/to/debugfs/file.o`
```
00001944 <debugfs_print_regs32>:
    1944:       e1a0c00d        mov     ip, sp
    1948:       e92ddff0        push    {r4, r5, r6, r7, r8, r9, sl, fp, ip, lr, pc}
    194c:       e24cb004        sub     fp, ip, #4
    1950:       e24dd00c        sub     sp, sp, #12
    1954:       e52de004        push    {lr}            ; (str lr, [sp, #-4]!)
    1958:       ebfffffe        bl      0 <__gnu_mcount_nc>
                        1958: R_ARM_CALL        __gnu_mcount_nc
    195c:       e3520000        cmp     r2, #0
    1960:       e1a05000        mov     r5, r0
    1964:       e50b2030        str     r2, [fp, #-48]  ; 0xffffffd0
    1968:       e1a0a003        mov     sl, r3
    196c:       e59b7004        ldr     r7, [fp, #4]
    1970:       da00001c        ble     19e8 <debugfs_print_regs32+0xa4>
    1974:       e3009000        movw    r9, #0
                        1974: R_ARM_MOVW_ABS_NC .LC10
    1978:       e3008000        movw    r8, #0
                        1978: R_ARM_MOVW_ABS_NC .LC11
    197c:       e2814008        add     r4, r1, #8
    1980:       e3409000        movt    r9, #0
                        1980: R_ARM_MOVT_ABS    .LC10
    1984:       e3408000        movt    r8, #0
                        1984: R_ARM_MOVT_ABS    .LC11
    1988:       e3a06000        mov     r6, #0
    198c:       ea000002        b       199c <debugfs_print_regs32+0x58>
    1990:       e51b3030        ldr     r3, [fp, #-48]  ; 0xffffffd0
    1994:       e1530006        cmp     r3, r6
    1998:       0a000012        beq     19e8 <debugfs_print_regs32+0xa4>
    199c:       e3570000        cmp     r7, #0
    19a0:       e1a02007        mov     r2, r7
    19a4:       e1a01009        mov     r1, r9
    19a8:       e1a00005        mov     r0, r5
    19ac:       0a000000        beq     19b4 <debugfs_print_regs32+0x70>
    19b0:       ebfffffe        bl      0 <seq_printf>
                        19b0: R_ARM_CALL        seq_printf
    19b4:       e914000c        ldmdb   r4, {r2, r3}
    19b8:       e08a3003        add     r3, sl, r3
>>> 19bc:       e5933000        ldr     r3, [r3]
    19c0:       f57ff04f        dsb     sy
    19c4:       e1a01008        mov     r1, r8
    19c8:       e1a00005        mov     r0, r5
    19cc:       ebfffffe        bl      0 <seq_printf>
...
```
My arm assembler reading skills are limited, and I don't completely get what is going here. It looks like NULL dereference happens in `readl(base + regs->offset)`; `base` is passed in from `regset->base`, which in turn assigned in `bcm2835_thermal_debugfs` from `data->regs`, which in turn assigned in `bcm2835_thermal_probe` from `devm_ioremap_resource()`. If it was `NULL`, there would be a lot of problems before; so, it's something strange (some `*dev*->*privs*` mismatch?).

I have not found any relevant changes upstream in either `drivers/thermal/broadcom/*.c` or `fs/debugfs/file.c`. When I triggered recompilation of `bcm2835_thermal.c`, there were no compiler warnings (about type mismatch/etc), so nothing obvious.

(Fortunately, this looks like debugging knob, not likely to be triggered by normal use, and debugfs access is limited to root)

yumkam, 2019-01-27:
Ah-ha. Seems I found it. Indeed, *dev*->*priv* mismatch. `bcm2835_thermal_probe` calls `thermal_zone_of_sensor_register(...,data,&...ops)`, `thermal_zone_of_sensor_register` calls `thermal_zone_of_add_sensor(..., data, ops)`, and `thermal_zone_of_add_sensor` sets `tz->sensor_data = data`.

Then, `bcm2835_thermal_probe` calls
```
	platform_set_drvdata(pdev, tz)
	bcm2835_thermal_debugfs(pdev);
```
and
```
static void bcm2835_thermal_debugfs(struct platform_device *pdev)
{
	struct thermal_zone_device *tz = platform_get_drvdata(pdev);
	struct bcm2835_thermal_data *data = tz->devdata;
```
... it dereferences `tz->devdata` instead of `tz->sensor_data`. Whoops.
There are same problem in `bcm2835_thermal_remove` too:
```
static int bcm2835_thermal_remove(struct platform_device *pdev)
{
	struct thermal_zone_device *tz = platform_get_drvdata(pdev);
	struct bcm2835_thermal_data *data = tz->devdata;
```
Probably, it should set `platform_*_drvdata` to `data` instead of `tz` anyway? Something like
```
diff --git a/drivers/thermal/broadcom/bcm2835_thermal.c b/drivers/thermal/broadcom/bcm2835_thermal.c
index 23ad4f9f2..a2baafe72 100644
--- a/drivers/thermal/broadcom/bcm2835_thermal.c
+++ b/drivers/thermal/broadcom/bcm2835_thermal.c
@@ -126,8 +126,7 @@ static const struct debugfs_reg32 bcm2835_thermal_regs[] = {
 
 static void bcm2835_thermal_debugfs(struct platform_device *pdev)
 {
-	struct thermal_zone_device *tz = platform_get_drvdata(pdev);
-	struct bcm2835_thermal_data *data = tz->devdata;
+	struct bcm2835_thermal_data *data = platform_get_drvdata(pdev);
 	struct debugfs_regset32 *regset;
 
 	data->debugfsdir = debugfs_create_dir("bcm2835_thermal", NULL);
@@ -273,7 +272,7 @@ static int bcm2835_thermal_probe(struct platform_device *pdev)
 
 	data->tz = tz;
 
-	platform_set_drvdata(pdev, tz);
+	platform_set_drvdata(pdev, data);
 
 	bcm2835_thermal_debugfs(pdev);
 
@@ -288,8 +287,8 @@ static int bcm2835_thermal_probe(struct platform_device *pdev)
 
 static int bcm2835_thermal_remove(struct platform_device *pdev)
 {
-	struct thermal_zone_device *tz = platform_get_drvdata(pdev);
-	struct bcm2835_thermal_data *data = tz->devdata;
+	struct bcm2835_thermal_data *data = platform_get_drvdata(pdev);
+	struct thermal_zone_device *tz = data->tz;
 
 	debugfs_remove_recursive(data->debugfsdir);
 	thermal_zone_of_sensor_unregister(&pdev->dev, tz);
```
Please verify it and send upstream, I consider upstream submitting procedure too complicated for me :-)

pelwell, 2019-01-28:
The current code relies on thermal_zone_of_sensor_register storing data in a place it can retrieve it - internal implementation detail - and then looks in the wrong place. I think using `data` as the drvdata is the right fix.

pelwell, 2019-01-29:
I agree - having skimmed the original issue I ended up with an identical patch. It's on its way upstream.

yumkam, 2019-01-29:
FWIW, I've rebuilt kernel ( rpi-4.14.y at 4de3f6305841c2121257b8fdbda5c3c0bf77e669 ) with above patch applied, and it fixed `cat .../regset`:
```
$ cat /sys/class/thermal/thermal_zone0/temp
47774
$ sudo cat /sys/kernel/debug/bcm2835_thermal/regset
ctl = 0x04380006
stat = 0x000006a5
```
(and `stat` changes when temperature changes, as expected)

JamesH65, 2019-09-19:
@pelwell Did upstream accept the patch?

lategoodbye, 2019-09-19:
Yes, it is fixed in Linux 5.1 ( 35122495a8c6683e863acf7b05a7036b2be64c7a ).

pelwell, 2019-09-19:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/thermal/broadcom/bcm2835_thermal.c?h=v5.3&id=35122495a8c6683e863acf7b05a7036b2be64c7a

JamesH65, 2019-09-19:
Happy to back port this, just need to know the correct procedure.

pelwell, 2019-09-19:
It's already done:
4.19: c5161c689e816e1d134801c771ae304aa842c620
5.0: f1a315ca06be80dddab380c7af1ddace3ee5f110

JamesH65, 2019-09-19:
Splendid.

Please update to the latest kernel which contains a fix for this issue.

closed 2019-09-19 by JamesH65