raspberrypi/linux #2835: Question about rpi-4.9.y-stable branch

shib, 2019-01-29:
What is the difference between rpi-4.9.y and rpi-4.9.y-stable?

popcornmix, 2019-01-29:
See: https://github.com/raspberrypi/linux/issues/2465 and
https://github.com/raspberrypi/linux/compare/rpi-4.9.y-stable...rpi-4.9.y

The rpi-4.9y-stable was created to find the source used in building the kernel for the raspbian that supported the Pi3 B+. We didn't want to push that to the public repo before the product was released, but needed to have images available on launch day.

Both trees are end-of-life now.

shib, 2019-01-30:
Ok. I am planning to upgrade from 4.4.y (yes, I know...) branch now for our project. Thats why I was looking at 4.9.y... I guess 4.14.y should be way forward... 

Cheers

6by9, 2019-01-30:
4.19.y will be the standard kernel within a month or so.

closed 2019-02-05 by popcornmix