raspberrypi/linux #2836: Ghostscript version missing from repository

DrAl-HFS, 2019-01-29:
Ghostscript install (triggered by xscreensaver install) fails; you can test with:

wget http://raspbian.raspberrypi.org/raspbian/pool/main/g/ghostscript/ghostscript_9.26~dfsg-0+deb9u2_armhf.deb

pelwell, 2019-01-29:
This is the Raspberry Pi Linux kernel repo, but your issue is with Raspbian, not the kernel.

Raspbian issues should be reported here: https://github.com/RPi-Distro/repo/issues

closed 2019-01-29 by pelwell

XECDesign, 2019-01-29:
https://bugs.launchpad.net/raspbian/ , even.

Either way, there doesn't seem to be an issue.

That should be `wget http://raspbian.raspberrypi.org/raspbian/pool/main/g/ghostscript/ghostscript_9.26a~dfsg-0+deb9u1_armhf.deb`, which works fine.