raspberrypi/linux #2842: Raspberrypi 0 wifi wifi driver not probed at all

nandra, 2019-01-30:
Build image for raspberrypi0w from thud branch (using meta-raspberrypi - kernel 4.14.79). After startup check ifconfig -a and no wlan0 interface was present. Anyway FW files are present:
```
ls /lib/firmware/brcm/
BCM43430A1.hcd          brcmfmac43430-sdio.bin  brcmfmac43430-sdio.txt
```
Steps to reproduce the issue:

    build image from thud branch for rpi0w (linux kernel version 4.14.79)

Describe the results you received:
No wlan0 interface available

Describe the results you expected:
wlan0 interface should be working

dmesg: https://pastebin.com/AG7YVdVD

lsmod:
```
root@raspberrypi0-wifi:~# lsmod                                                 
Module                  Size  Used by                                           
nfc                    67650  0                                                 
bnep                   12269  2                                                 
bluetooth             368017  9 bnep                                            
ecdh_generic           17544  1 bluetooth                                       
rfkill                 22731  3 bluetooth,nfc                                   
ipv6                  404061  22                                                
fixed                   3001  0                                                 
uio_pdrv_genirq         3686  0                                                 
uio                    10016  1 uio_pdrv_genirq 
```

In config:
```
CONFIG_MMC_BCM2835_MMC=y
CONFIG_MMC_BCM2835_DMA=y
CONFIG_MMC_BCM2835_PIO_DMA_BARRIER=2
CONFIG_MMC_BCM2835_SDHOST=y
```

Any ideas what can cause this?

pelwell, 2019-01-30:
You can start by dumping the state of the GPIOs. On Raspbian we'd use `raspi-gpio get` - I don't know what distribution you are using, but it's easily built from source: https://github.com/RPi-Distro/raspi-gpio

nandra, 2019-01-31:
@pelwell looks like pins are configured properly (also checked by dts):
```
GPIO 34: level=0 fsel=7 alt=3 func=SD1_CLK
GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD
GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0
GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1
GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2
GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3
```

pelwell, 2019-01-31:
Indeed. What do you get from the following commands?:
```
$ sudo cat /sys/kernel/debug/mmc1/ios
$ ls /sys/class/mmc_host/mmc1
$ grep . /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/*
```

pelwell, 2019-01-31:
And does the Bluetooth modem appear?

nandra, 2019-01-31:
cat /sys/kernel/debug/mmc1/ios
```
clock:          0 Hz
vdd:            0 (invalid)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     0 (off)
bus width:      0 (1 bits)
timing spec:    0 (legacy)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)
```

ls /sys/class/mmc_host/mmc1
```
device     power      subsystem  uevent
```
grep . /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/*
```
grep: /sys/class/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/*: No such file or directory
```

pelwell, 2019-01-31:
[ Full disclosure - I'm debugging on a 3B+ because it's convenient, but the results should still apply. ]

Odd. Deleting the brcmfmac modules still leaves the SDIO card being detected, so it is failing before then. Disabling the WL_ON and BT_ON signals in a custom dt-blob.bin leads to the results you are seeing, but that would show up in raspi-gpio.

Please try a Raspbian image to confirm that it isn't a hardware problem.

nandra, 2019-01-31:
I've tried with previous yocto build and it was working perfectly fine so it cannot be HW issue.

pelwell, 2019-01-31:
If you can see it working now on another build - whatever it is - then I'm satisfied it isn't hardware. I think you're going to have to bisect the yocto changes yourself - I'm out of ideas and time.

nandra, 2019-01-31:
@pelwell ok thanks for help anyway. I'll try to find what is going on and if find an issue I'll post update. Thx. :+1:

nandra, 2019-01-31:
@pelwell flashed raspbian (2018-11-13-raspbian-stretch-lite.img) and wifi interface also doesn't show up so conclusion is that probably wifi chip is dead somehow. Strange. Thanks

pelwell, 2019-01-31:
That's a shame, but it makes sense.

closed 2019-01-31 by pelwell

nandra, 2019-02-04:
@pelwell one more info with raspbian bluetooth works (wifi not). Does is make any sense?

pelwell, 2019-02-04:
The Bluetooth modem and the WiFi interface are effectively two different devices in one package, except with a co-existence channel between them. I can well believe that some kind of fault on the WiFi SDIO interface would not affect Bluetooth's UART interface.