raspberrypi/linux #2844: UDP packet gets checksummed incorrectly by the network interface

flux77, 2019-01-31:
**Describe the bug**
When running a DHCP server on the Raspberry Pi, the output from `sudo tcpdump -n udp port 67 -e -vv` shows that there are packets coming from the Raspberry Pi that have bad checksums. Specifically, the DHCP clients never receive the `DHCPOFFER` packet.

Once the offloading of checksums has been disabled (using `sudo ethtool --offload eth0 tx off`) there are no more packets with bad checksums (i.e. there is no longer any problem). That is why I suspect that there is an issue with some drivers in the kernel.

Because I am a beginner at networking, I have not reached the stage where I know how to produce a minimal example of packets. I hope you'll find it permissible for me to refer you to [my question on Stack Exchange](https://raspberrypi.stackexchange.com/questions/93731/raspberry-pi-dhcp-server-clients-not-getting-ip-addresses/93736#93736).

**Expected behaviour**
The UDP packet has the correct checksum.

**Actual behaviour**
The UDP packet has a bad checksum.

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 3
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2018-06-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 0d34fb94398a4bd79711f21d6a0b5d35fe5f4a76, stage5
* Which firmware version (`vcgencmd version`)?
Jun  7 2018 15:31:38 
Copyright (c) 2012 Broadcom
version 4800f08a139d6ca1c5ecbee345ea6682e2160881 (clean) (release)
* Which kernel version (`uname -a`)?
Linux rpi 4.14.50-v7+ #1122 SMP Tue Jun 19 12:26:26 BST 2018 armv7l GNU/Linux

pelwell, 2019-01-31:
Some checksumming issues have been found and fixed since the kernel version you have now. Please run `sudo rpi-update` to install the latest kernel, then reboot and retest.

6by9, 2019-01-31:
Please confirm that this is on a 3B and not a 3B+ - they have different network chips and therefore this is critical to debugging the issue.

4.14.50 is 7 months old and there have been a far number of fixes to both LAN drivers since then, although not apparently directly in this area. Can you update your kernel to the latest 4.14 kernel please? (The latest release from the Raspbian repos should be from mid December IIRC)

After updating, please provide the complete hex dump from tcpdump for these nominally corrupt packets.

There's also something strange about the MAC address that your computer is using. The first 3 octets of a MAC address should be registered to the manufacturer. f4:8e:38 is not a registered manufacturer, and I'd want to check whether it is actually valid (there are rules to dictate unicast vs multicast vs broadcast based on the MAC address, and I can't recall them off the top of my head).

flux77, 2019-01-31:
@6by9 I can confirm that this is on a 3B (according to the PCB markings: "Raspberry Pi 3 Model B V1.2").

The output from `lshw` says that the network interface is a "RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller" by "Realtek Semiconductor Co., Ltd.".