raspberrypi/linux #2846: no wifi internal or external works

madah123, 2019-02-01:
here is my dmesg
https://pastebin.com/KDEE7fPe

pelwell, 2019-02-01:
I suspect your Pi is underpowered. What does `vcgencmd get_throttled` report?

madah123, 2019-02-01:
throttled = 0x0

madah123, 2019-02-01:
if it's relevant iwconfig and ifconfig only show eth0 and lo no wlan0 i can't attempt to change wifi or wifi code in raspi-config

pelwell, 2019-02-01:
Can you try booting with no USB peripherals attached (you can add them after booting has completed), then upload a new dmesg log?

madah123, 2019-02-01:
how can i dmesg if i don't use mouse + keyboard?

madah123, 2019-02-01:
https://pastebin.com/TpWc8R4D

pinguinpfleger, 2019-02-03:
I had the same issue maybe.
Wi-Fi suddenly stopped working with my Raspberry Pi.
Something must have been changed after an apt-get upgrade these days.
After adding country=XX (XX = your ISO country code) to wpa_supplicant.conf file Wi-Fi went up again.
https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md

pelwell, 2019-02-04:
The current Raspbian images do require that the County Code is set, otherwise the wlan interface is blocked using rfkill.

JamesH65, 2019-09-19:
@madah123 Are you still seeing this issue? Did setting the country code fix it?

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2019-12-20 by JamesH65