raspberrypi/linux #2849: Add DVBSky S960 patch to fix broken driver

gurabli, 2019-02-05:
There is a long standing bug in kernel 4.1x+ versions for more than a year that broke a perfectly working DVB adapter on all kernels. Finally, there is a patch that fixes this annoying issue.
Is it possible to add this to the Raspbian kernel?

It is already included in [LibreElec](https://github.com/LibreELEC/LibreELEC.tv/pull/3232) mediabuild, and [Armbian](https://github.com/armbian/build/commit/cbfcd8571515ba84be922a32d308a8a306fde0f1) as well. 

If the patch is applied, when will it be available in Raspbian Stretch Lite? Or a how can I upgrade to the patched kernel?

pelwell, 2019-02-05:
[ The patch is on its way upstream - see https://patchwork.kernel.org/patch/10761423/ ]

We are usually happy to take bug-fix backports - just let us know when it has been accepted in one of the kernel branches and we can cherry-pick it. From then, it would likely be only a few days until a firmware release including the patch - `sudo rpi-update` would be the quickest route to install the new build.

6by9, 2019-02-05:
It would have helped if he had added a "Fixes" tag, as then it would have been picked up and backported to the stable branches automagically (although after it has been merged into a mainline release). That may happen anyway once it has been merged.

(I do have an S960 at home, so can test the patch if needed).

pelwell, 2019-02-05:
We can hope for a helpful reviewer. Or contribute to the thread...

6by9, 2019-02-05:
Rereading the thread I think two issues have become conflated, with yet a third issue being referenced.

- 7d95fb7 broke things completely and was reverted.
- There was still an underlying problem causing lockups after some time that this patch fixes. I suspect that issue has been around forever so doesn't fix any particular commit. 9d659ae is referenced, but causes the failure due to the driver bug.
- https://bugzilla.kernel.org/show_bug.cgi?id=199323#c38 then mentions a race condition that is a separate issue.

I would contribute to the thread if linux-media would deliver email to my account these days - it stopped back in December.

gurabli, 2019-02-05:
Thank you guys for looking into this.
What you write is beyond my knowledge, don't get it how this actually work (will look into it in more details). 

Only thing I know that DVBSky S960 and similar devices are very popular and they work rock solid for many years (on my Ubuntu Server 16.04 LTS and before on 14.04 LTS I never had a single problem and the tuner worked for months without a single restart. Then came a commit that broke everything, and many of us suffered for more than a year on newer kernels (especially on Libreelec where kernel change is way harder than on an Ubuntu Server). 

I hope this patch (that is confirmed to render the S960 fully working again) will be in mainline soon, also on Raspbian.

Thanks for your efforts! I can contribute by testing, if needed. Have an rpi2 and tbe S960 waiting :)

lategoodbye, 2019-02-07:
Since i couldn't find this patch in linux-next someone need to ping the maintainers about this patch. Also it's still possible to add the stable tag for the maintainers.

gurabli, 2019-02-07:
I would love to, but as I said, don't know much about this part (mainteners, linux-next, stable tag). I just know that finally it is out (and already fully working on Libreelec and Armbian).

6by9, 2019-02-08:
The general way I've seen the linux-media maintainers (mainly Mauro) working is that they batch merge stuff as it comes up to the merge window.

I believe https://patchwork.linuxtv.org/project/linux-media/list/ is the patchwork system that they use, and https://patchwork.linuxtv.org/patch/53885/ is this patch on there. It's still in state "New" and no delegation, so none of the linux-media maintainers have traiged it, despite Antti Palosaari as the file maintainer having added an R-B tag.

Armbian and Libreelec both maintain a set of patches over and above the mainline kernel (as do we). Their requirements on the patches are obviously slightly lower than ours in not requiring the patch to have been merged someone upstream before cherry-picking.

snowstormfred, 2019-04-21:
The patch looks to be accepted now.  How easy is it to find out when it has been/will be updated in the kernel?

pelwell, 2019-04-21:
It's in the 'next' tree (https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/media/dvb-frontends/m88ds3103.c?h=next-20190418&id=981fbe3da20a6f35f17977453bce7dfc1664d74f), so it can now be cherry-picked into rpi-4.19.y. See 88ef00d.

snowstormfred, 2019-04-22:
Sorry about my confusion, does this mean it is in the rpi kernel, or that it is ready to be in the kernel at some point?  What do I need to check to see when it's in the kernel and what version it is??

pelwell, 2019-04-22:
The patch is in our active kernel source branches. At some point in the next week we'll probably release a new kernel build which will be available first via rpi-update. Raspbian kernel package updates are less frequent.

popcornmix, 2019-04-23:
@snowstormfred rpi-update kernel now contains this patch.

snowstormfred, 2019-04-27:
Works great now.  Thanks for that.  I've been waiting (im)patiently for months for this.

closed 2019-04-27 by pelwell