raspberrypi/linux #2850: Please can you add the AD193x (AD1938) driver please

danriches, 2019-02-07:
I need to use the AD1938 driver and the source files are there but I can't select the driver when I do make menuconfig for some reason. Can you add the drivers in please, it would be greatly appreciated. I was asked to do this by someone on the RasPi forums when I asked for assistance there.

Kind Regards, 
Dan

pelwell, 2019-02-07:
I think the problem is that, like lots of other codec drivers, its Kconfig symbol doesn't have a string associated with it:
```
config SND_SOC_AC97_CODEC
	tristate "Build generic ASoC AC97 CODEC driver"
	select SND_AC97_CODEC
	select SND_SOC_AC97_BUS

config SND_SOC_AD1836
	tristate

config SND_SOC_AD193X
	tristate
```
This seems to prevent it being enabled except by another Kconfig item "select"ing it, and the only such selection comes from SND_SOC_ALL_CODECS. As a result, in order to enable it we'd need to either hack `sound/soc/codecs/Kconfig` or set SND_SOC_ALL_CODECS.

There are currently about 30 enabled codecs, ~140 with no string and ~260 altogether, so flicking the big switch is not an attractive proposition. @popcornmix - shall I make it user-selectable?

popcornmix, 2019-02-07:
@pelwell I guess that's the least bad option.

danriches, 2019-02-07:
That would be great if you could and obviously I don't want this to cause any issues upstream which I'm fairly sure it won't, however I'll leave it to you guys :)

pelwell, 2019-02-07:
We can manage this as a downstream patch, at least until it proves to be a maintenance burden.

pelwell, 2019-02-07:
Do you want it enabled for SPI or I2C?

danriches, 2019-02-07:
It's for the AD1938 so spi please,  thanks!!

pelwell, 2019-02-07:
I've enabled it for both. It's in rpi-4.14.y now, and will arrive in the newer branches soon.

danriches, 2019-02-07:
Superb, thanks ever so much. I'll give it a go and report back and close this issue.

popcornmix, 2019-02-13:
Should be in current rpi-update kernel.

JamesH65, 2019-09-19:
Closing this issue as questions answered/issue resolved.

closed 2019-09-19 by JamesH65