raspberrypi/linux #2851: sockets with PCI  USB card hardlocks system

SairesArt, 2019-02-08:
I'm having a truly bizarre problem. I'm on Raspbian Desktop, on a 970 chipset AMD motherboard.
Due to a [firmware bug](https://github.com/raspberrypi/gpioexpander/issues/2) with the raspberry pis I have to use a PCI (not PCI-E, those slots are taken) to USB Card to connect to the PiZero as a gpio expander.
The thing is: if I start reading GPIOs in a loop after a certain time the whole system hardlocks. Full on freeze. No kernel crash, no nothing. Just screen freeze and only a hardreset makes a recovery.

It does not matter, whether I do it my C code with sockets or if make a bash loop of (pigs r 23 w 5 1) for instance. After a certain time everything dies. So it's not my code's fault.

If I use the normal USB ports, as presented by the internal USB controllers all works fine. But due to the [firmware bug](https://github.com/raspberrypi/gpioexpander/issues/2) I cannot use the normal USB ports.
The moment of the freeze is not dependent on a specific amount of socket packets.
I have tested multiple PCI cards.
The three PCI to USB cards with VIA controllers all fail between 4-12MB of reads/writes (as read by ifconfig).
A card with an "ALI" controller survives certainly longer, with ~40MB of reads / writes before locking up the system, multiple confirmed this. It almost behaves like an overheating problem, but I cannot see any signs of that. The Southbridge doesn't feel hotter than normal, neither do the USB controllers.

I'm truly at a loss. Can someone help?

JamesH65, 2019-02-08:
@ghollingworth @bennuttall Have you any idea who deals with gpioexpander stuff? TBH, I didn't even know that repo existed, or that the gpioexpander was a thing.

bennuttall, 2019-02-08:
Gordon contracted [name escapes me] to make it.

ghollingworth, 2019-02-10:
Is this with an SD card plugged into the Raspberry Pi or are you still trying to USB boot it as well...

SairesArt, 2019-02-11:
> Is this with an SD card plugged into the Raspberry Pi or are you still trying to USB boot it as well...

I'm USB booting.
Although I don't know if you could somehow do virtual USB network connection with a SD card to test whether this is a strictly USBboot phenomenon, or whether it is simply a PCI / USB Chipset thing.

I rewrote all of my code to only fetch information that I need and limited these information fetches to 5hz. It seemed to have solved the problem, but it just looms in the background and I wonder whether after a couple of days of run time my machine won't just die and leave the client clueless, since I can't turn on an error lamp, because everything is dead. I thought of a plane style blinker that blinks every 5 secs to check if the machine still lives :D