raspberrypi/linux #2853: VC4 and Hifiberry: Stable way to set ALSA card ordering

vanfanel, 2019-02-08:
Hi there!

I'be been trying for a year to fix this issue.
Thing is, I have an Hifiberry card on my Pi setup, where I use the open source VC4 driver stack. That gives me two ALSA cards: snd_rpi_hifiberry_dacplus and vc4-hdmi. It works almost fine, but sometimes alsa-restore.service fails on boot because card ordering changes between boots, so it tries to apply control states to the wrong card, thus failing.

So, I need a way to stabilize alsa card ordering among boots.
I tried creating /etc/modprobe.d/alsa-base.conf with these contents:

```
options snd_soc_hifiberry_dacplus index=0
options vc4 index=1
```

Alternatively I also tried: 

`options vc4 index=-2`

(It's supposed to tell modprobe to NEVER allow a card to be the first ALSA one).

However, none of those seem to work. From boot to boot, I get different cards as the first ALSA card...
Can you guys tell me of a working way to get the Hifiberry to ALWAYS be the first ALSA card?
Disabling the vc4-hdmi card entirely would be an option, but there seems to be no way either.

Thanks!

6by9, 2019-02-08:
Duplicating https://github.com/anholt/linux/issues/130

vanfanel, 2019-02-08:
@6by9 I believe these issues, belonging to different repositories, are read by different people, so if it's a bad idea to post the issue on both, then tell me and I will remove the least appropiate one.

JamesH65, 2019-09-19:
@spl237 Wasn't this what we were talking about yesterday?

spl237, 2019-09-19:
Yes; I'm not aware that there is any solution to this. ALSA is a bit suboptimal sometimes...