raspberrypi/linux #2854: Random crashes since raspian stretch

theiconian, 2019-02-12:
Experiencing some seemingly random crashes since raspbian stretch, sometimes it occurs a few minutes after booting, sometimes it happens after running for a day, its not happening every day, so giving some proper reproduction steps is sadly a problem. Im using the raspi mostly as a firewalling device between 2 subnets. I've no clue about debugging kernel issues really, but both of the 2 crashes that i observed within the last coupe days since i attached a debug cable had this PC is at hash_net4_test+0x1f0/0x54c [ip_set_hash_net] line in common, so i suspect its something network related.

- Model: Pi3b

- OS version:
$cat /etc/rpi-issue
Raspberry Pi reference 2018-04-18
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, d6c238c1b2b1b070a574d3e25048ca442e3e221f, stage2

- Firmware:
$ vcgencmd version
Nov  4 2018 16:32:03
Copyright (c) 2012 Broadcom
version ed5baf9520a3c4ca82ba38594b898f0c0446da66 (clean) (release)

- Kernel:
 $ uname -a
Linux raspi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

- Connected hardware:
an usb network adapter
$lsusb
Bus 001 Device 004: ID 0bda:8153 Realtek Semiconductor Corp.
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

- Powersupply: original 
$vcgencmd get_throttled
throttled=0x0

```
[  104.988398] Unhandled fault: alignment exception (0x001) at 0xbc0bc7bc
[  104.999257] pgd = 80004000
[  105.006223] [bc0bc7bc] *pgd=3c01141e(bad)
[  105.014526] Internal error: : 1 [#1] SMP ARM
[  105.023008] Modules linked in: xt_nat xt_REDIRECT nf_nat_redirect ipt_MASQUERADE nf_nat_masquerade_ipv4 ip6table_filter ip6_tables ip_set_hash_net xt_recent xt_iprange xt_comment iptable_nat nf_nat_ipv4 ipt_REJECT nf_reject_ipv4 xt_addrtype xt_set ip_set_hash_ip ip_se
t xt_mark iptable_mangle xt_tcpudp xt_CT iptable_raw xt_multiport nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack xt_NFLOG nfnetlink_log xt_LOG nf_log_ipv4 nf_log_common nf_nat_tftp nf_nat_snmp_basic nf_conntrack_snmp nf_nat_sip nf_nat_pptp nf_nat_proto_gre nf_nat_irc nf_n
at_h323 nf_nat_ftp nf_nat_amanda ts_kmp nf_conntrack_amanda nf_nat nf_conntrack_sane nf_conntrack_tftp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_netbios_ns nf_conntrack_broadcast nf_conntrack_irc nf_conntrack_h323 nf_conn
track_ftp
[  105.143555]  nf_conntrack nfnetlink iptable_filter cdc_ether r8152 snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq fixed uio ip_tables x_tables ipv6
[  105.166700] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C      4.14.79-v7+ #1159
[  105.183977] Hardware name: BCM2835
[  105.192098] task: 80c06cc0 task.stack: 80c00000
[  105.201383] PC is at hash_net4_test+0x1f0/0x54c [ip_set_hash_net]
[  105.212265] LR is at init_thread_union+0x19f0/0x2000
[  105.222015] pc : [<7f3743c8>]    lr : [<80c019f0>]    psr: 40000113
[  105.233117] sp : 80c01990  ip : 00000000  fp : 80c019dc
[  105.243223] r10: 80c02d00  r9 : 7f312ce0  r8 : 654e4a68
[  105.253320] r7 : bc0bc7b4  r6 : 00000020  r5 : ffffffff  r4 : ffffffff
[  105.264736] r3 : 00000000  r2 : 00000034  r1 : bc0bc7bc  r0 : b918d580
[  105.276140] Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  105.288189] Control: 10c5383d  Table: 3927406a  DAC: 00000055
[  105.298850] Process swapper/0 (pid: 0, stack limit = 0x80c00210)
[  105.309792] Stack: (0x80c01990 to 0x80c02000)
[  105.319079] 1980:                                     b903e900 bc1cd900 bc1cd800 bc63c000
[  105.337210] 19a0: 80c019f0 80c01a78 bc1cd800 00000003 80c019d4 7f3741d8 00000000 00000001
[  105.355493] 19c0: 00000000 00000034 bc0bce40 ba1cf000 80c01a3c 80c019e0 7f374810 7f3741e4
[  105.374018] 19e0: 00000000 00004000 654e4a68 20000000 00000000 00000000 00000000 00000000
[  105.392792] 1a00: 00000001 00000000 00000034 00000000 00000000 00000000 7f22e174 b918d580
[  105.411757] 1a20: b93e52b8 00000000 80c01a70 80c01b40 80c01a6c 80c01a40 7f30ee64 7f374730
[  105.430818] 1a40: 80c01a70 7f34b060 00000000 b93e52b8 00000000 bc5fc800 b901d6d0 bc0bce40
[  105.449984] 1a60: 80c01ab4 80c01a70 7f32aee0 7f30edc4 00000102 00000000 00000000 00000000
[  105.469351] 1a80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 ffffffff
[  105.488922] 1aa0: b93e5228 b93e5298 80c01b84 80c01ab8 7f0944f4 7f32ae40 807201bc 807247cc
[  105.508717] 1ac0: 00000000 654e4a68 0001a8c0 fd010018 00000000 bc125900 bc01f200 bce7221c
[  105.528747] 1ae0: 00000000 7f097204 7f0971d8 80c01bcc bcccc008 0006bb01 00000000 80c03e60
[  105.548992] 1b00: 80b8f780 7f097204 7f0971d8 80c01bcc b93c0008 bcccc040 00000001 80c03e60
[  105.569350] 1b20: 80b8f780 00000001 ba3533c0 00000002 bc3df400 b93c0040 b93c0000 00000000
[  105.589898] 1b40: 7f32d214 b93e52b8 80c01bcc 00000000 00000014 ba1cf000 80c01b94 00000001
[... 90 lines cut ...]
[  107.557130] Exception stack(0xbc083fb0 to 0xbc083ff8)
[  107.569364] 3fa0:                                     00844028 00000010 00000003 00847ce0
[  107.591814] 3fc0: 00844028 00000010 00000000 00000000 00000003 00847750 00000010 00000000
[  107.614249] 3fe0: 76fa3f3c 7ea0acc8 76ede0ac 76e97124 20000010 ffffffff
[  107.628034] CPU3: stopping
[  107.637784] CPU: 3 PID: 314 Comm: rs:main Q:Reg Tainted: G      D  C      4.14.79-v7+ #1159
[  107.660160] Hardware name: BCM2835
[  107.670522] [<8010ff30>] (unwind_backtrace) from [<8010c174>] (show_stack+0x20/0x24)
[  107.692161] [<8010c174>] (show_stack) from [<80788fa4>] (dump_stack+0xd4/0x118)
[  107.706471] [<80788fa4>] (dump_stack) from [<8010e3b4>] (handle_IPI+0x2bc/0x33c)
[  107.727673] [<8010e3b4>] (handle_IPI) from [<801014d8>] (bcm2836_arm_irqchip_handle_irq+0x7c/0xac)
[  107.750460] [<801014d8>] (bcm2836_arm_irqchip_handle_irq) from [<807a4b7c>] (__irq_svc+0x5c/0x7c)
[  107.773149] Exception stack(0xba18ddf0 to 0xba18de38)
[  107.785092] dde0:                                     be5d6eb0 00000000 bce720e0 000003f2
[  107.806966] de00: 00000000 bc3ad000 bc3ad000 00000007 80c24108 00000007 bc3ad0f0 ba18dea4
[  107.828800] de20: ba18de20 ba18de40 802e00e8 802e1268 60000013 ffffffff
[  107.842281] [<807a4b7c>] (__irq_svc) from [<802e1268>] (posix_lock_inode+0x2f4/0x79c)
[  107.863764] [<802e1268>] (posix_lock_inode) from [<802e1788>] (vfs_lock_file+0x54/0x58)
[  107.885506] [<802e1788>] (vfs_lock_file) from [<802e17c4>] (do_lock_file_wait+0x38/0xb8)
[  107.907379] [<802e17c4>] (do_lock_file_wait) from [<802e36f0>] (fcntl_setlk+0x13c/0x2a8)
[  107.929275] [<802e36f0>] (fcntl_setlk) from [<8029d650>] (do_fcntl+0x52c/0x740)
[  107.943543] [<8029d650>] (do_fcntl) from [<8029dad0>] (SyS_fcntl64+0x1dc/0x22c)
[  107.957787] [<8029dad0>] (SyS_fcntl64) from [<80108000>] (ret_fast_syscall+0x0/0x28)
[  107.979247] ---[ end Kernel panic - not syncing: Fatal exception in interrupt
```

theiconian, 2019-02-16:
Got another another one, anyone got an idea yet?
I hope i put this in the right place,.

```
[ 9253.467151] pgd = b9ad0000
[ 9253.474143] [ba637c3c] *pgd=3a61141e(bad)
[ 9253.482500] Internal error: : 1 [#1] SMP ARM
[ 9253.491074] Modules linked in: xt_nat xt_REDIRECT nf_nat_redirect ipt_MASQUERADE nf_nat_masquerade_ipv4 ip6table_filter ip6_tables ip_set_hash_net xt_recent xt_iprange xt_comment iptable_nat nf_nat_ipv4 ipt_REJECT nf_reject_ipv4 xt_addrtype xt_set ip_set_hash_ip ip_se
t xt_mark iptable_mangle xt_tcpudp xt_CT iptable_raw xt_multiport nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack xt_NFLOG nfnetlink_log xt_LOG nf_log_ipv4 nf_log_common nf_nat_tftp nf_nat_snmp_basic nf_conntrack_snmp nf_nat_sip nf_nat_pptp nf_nat_proto_gre nf_nat_irc nf_n
at_h323 nf_nat_ftp nf_nat_amanda ts_kmp nf_conntrack_amanda nf_nat nf_conntrack_sane nf_conntrack_tftp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_netbios_ns nf_conntrack_broadcast nf_conntrack_irc nf_conntrack_h323 nf_conn
track_ftp
[ 9253.612260]  nf_conntrack nfnetlink iptable_filter cdc_ether r8152 snd_bcm2835(C) snd_pcm snd_timer snd uio_pdrv_genirq uio fixed ip_tables x_tables ipv6
[ 9253.635527] CPU: 2 PID: 1165 Comm: squid Tainted: G         C      4.14.79-v7+ #1159
[ 9253.652816] Hardware name: BCM2835
[ 9253.660995] task: ba5c0f00 task.stack: b9bc4000
[ 9253.670353] PC is at hash_net4_test+0x1f0/0x54c [ip_set_hash_net]
[ 9253.681302] LR is at 0xb9bc5ad8
[ 9253.689287] pc : [<7f3643c8>]    lr : [<b9bc5ad8>]    psr: 40000013
[ 9253.700496] sp : b9bc5a78  ip : 00000000  fp : b9bc5ac4
[ 9253.710665] r10: 80c02d00  r9 : 7f302ce0  r8 : a5dadd58
[ 9253.720835] r7 : ba637c34  r6 : 00000020  r5 : ffffffff  r4 : ffffffff
[ 9253.732326] r3 : 00000000  r2 : 0000003c  r1 : ba637c3c  r0 : bc10d300
[ 9253.743802] Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 9253.755922] Control: 10c5383d  Table: 39ad006a  DAC: 00000055
[ 9253.766662] Process squid (pid: 1165, stack limit = 0xb9bc4210)
[ 9253.777591] Stack: (0xb9bc5a78 to 0xb9bc6000)
[ 9253.786958] 5a60:                                                       b9bc5aa4 bc14d900
[ 9253.805100] 5a80: bc14d800 ba7d8000 b9bc5ad8 b9bc5b60 bc14d800 00000003 b9bc5b14 7f3641d8
[ 9253.823334] 5aa0: 00000000 00000001 00000000 0000003c ba7940b8 7f098340 b9bc5b24 b9bc5ac8
[ 9253.841767] 5ac0: 7f364810 7f3641e4 00000000 00000000 a5dadd58 20000000 00000000 00000000
[ 9253.860419] 5ae0: 00000000 00000000 00000001 00000000 0000003c 00000000 00000000 00000000
[ 9253.879267] 5b00: 00000000 bc10d300 b6004c30 00000000 b9bc5b58 b9bc5c28 b9bc5b54 b9bc5b28
[ 9253.898317] 5b20: 7f2fee64 7f364730 b9bc5b58 00000000 00000000 b6004c30 00000000 bcc1e000
[ 9253.917424] 5b40: bce590f4 ba7940b8 b9bc5b9c b9bc5b58 7f31aee0 7f2fedc4 00000102 00000000
[ 9253.936712] 5b60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 9253.956153] 5b80: 00000000 ffffffff b6004ba0 b6004c10 b9bc5c6c b9bc5ba0 7f0944f4 7f31ae40
[ 9253.975831] 5ba0: 00000001 00000001 b9bc5c14 b9bc5bb8 7f21beb8 7f21bbdc a5dadd58 00000000
[ 9253.995747] 5bc0: 00000000 00000000 00000001 a5dadd58 00000000 7f33a620 00000000 a5dadd58
[ 9254.015845] 5be0: 00000000 00000000 00000000 7f097204 7f0971dc b9bc5cb4 b600000c b60dde00
[ 9254.036052] 5c00: 00000001 80c03e60 80b8f780 00000001 ba644e00 00000003 b9a91400 b6000040
[ 9254.056464] 5c20: b6000000 00000000 7f31d214 b6004c30 b9bc5cb4 00000000 00000014 b9bc5c00
[ 9254.077098] 5c40: b9bc5c6c 00000001 00000005 ba461728 ba461700 ba7940b8 b9bc5cb4 80c7a7c0
[ 9254.097954] 5c60: b9bc5c7c b9bc5c70 7f16d184 7f0941d4 b9bc5cac b9bc5c80 806d0b28 7f16d15c
[ 9254.118994] 5c80: b99f8900 0100007f 00000003 80c7a7c0 b2027080 00000000 ba65dd80 00000028
[ 9254.140188] 5ca0: b9bc5cec b9bc5cb0 806de3f0 806d0ae8 b9bc5ce4 00000003 80277602 00000000
[ 9254.161526] 5cc0: bcc1e000 b2027080 80c7a7c0 806dc3e8 ba7940b8 b2027080 80c7a7c0 00000000
[ 9254.183051] 5ce0: b9bc5d0c b9bc5cf0 806de46c 806de328 00000000 ba7940b8 b2027080 b20272f0
[ 9254.204736] 5d00: b9bc5d5c b9bc5d10 806de79c 806de454 00000000 ba7940b8 ba794000 8d4fdf3b
[ 9254.226650] 5d20: b9bc5d44 b9bc5d30 80677fb8 80701050 00000000 00000000 b2027080 ba7940b8
[ 9254.248742] 5d40: ba794000 278c8914 00000028 bce59108 b9bc5dbc b9bc5d60 806f8f00 806de668
[ 9254.270998] 5d60: b9bc5dbc 00bc5d70 8071953c b2027080 b20274e0 00000000 05b4000b 00000007
[ 9254.293353] 5d80: 00000000 96a21169 00000000 00000000 5dc54aa7 b2027080 ba794000 8d4fdf3b
[ 9254.315873] 5da0: 83126e97 00000000 00000002 00000000 b9bc5e2c b9bc5dc0 806fae34 806f8a00
[ 9254.338407] 5dc0: 00000000 b9bc5de7 00000000 03e6d1a6 00000000 00000000 00000000 00000000
[ 9254.360950] 5de0: aef363aa 079cc448 00000000 00000000 00000000 00000000 00000000 00000000
[ 9254.383505] 5e00: 00000000 b2027080 00000000 b9bc5f08 b20272c8 a5dadd58 80c7a7c0 ba65dd80
[ 9254.406068] 5e20: b9bc5e84 b9bc5e30 806fe728 806fa518 80182208 00000000 b2027284 00000000
[ 9254.428662] 5e40: 80c7ab80 00005000 00000000 00000000 be5c8440 00000000 00000000 b2027080
[ 9254.451256] 5e60: bb766540 00000010 b9bc5f08 801081c4 b9bc4000 00000802 b9bc5ed4 b9bc5e88
[ 9254.473876] 5e80: 80718010 806fe3d0 80673a5c 00000010 b9bc5f08 801081c4 b9bc5ebc b9bc5ea8
[ 9254.496649] 5ea0: 801238d4 801ee4e4 b2027080 bb766540 00000802 00000010 b9bc5f08 801081c4
[ 9254.519554] 5ec0: b9bc4000 0000011b b9bc5efc b9bc5ed8 8071828c 80717efc 00000000 b9bc5ee8
[... cut ...]
[ 9256.074936] [<80108a10>] (arch_cpu_idle) from [<807a42dc>] (default_idle_call+0x34/0x48)
[ 9256.096827] [<807a42dc>] (default_idle_call) from [<8016177c>] (do_idle+0xd8/0x150)
[ 9256.118355] [<8016177c>] (do_idle) from [<80161a90>] (cpu_startup_entry+0x28/0x2c)
[ 9256.139916] [<80161a90>] (cpu_startup_entry) from [<8010deac>] (secondary_start_kernel+0x134/0x140)
[ 9256.162981] [<8010deac>] (secondary_start_kernel) from [<0010196c>] (0x10196c)
[ 9256.177306] ---[ end Kernel panic - not syncing: Fatal exception in interrupt
```

dhireshan, 2019-06-25:
Hi. Did you ever get to the bottom of this? I am experiencing something similar but on a Pi Zero W.

JamesH65, 2019-09-19:
That function (hash_net4_test) is not in the latest kernel AFAICT, so I suspect moving to the latest kernel may well fix (or at least change) this error.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

dhireshan, 2019-09-19:
FYI. My problem went away after I changed the clock frequency of my Pi Zero W to 900MHZ. i.e. arm_freq=900 within config.txt.  So it's possible that it might be that the Pi's cpu is running at the very edge of stability - just a hunch. I don't mean to keep this issue open, however, I wanted to post my solution incase it helps others.

JamesH65, 2019-09-19:
Interesting, thanks for the report. I'll close this as we have an explanation/solution.

Closing this issue as questions answered/issue resolved.

closed 2019-09-19 by JamesH65