raspberrypi/linux #2855: Setup a custom MIPI DSI Display Driver workflow

llks, 2019-02-13:
I have a MIPI DSI display which uses the ILI9881c driver IC, i am planning to custom make a display driver for this IC. I think this project can greatly contribute to the Raspberry Pi communities whoever wanted to know how to create their own custom MIPI DSI display driver and i can conduct the test on my site. As far as my understanding, from all the information that i have gather from the following site,

https://github.com/anholt/linux/wiki/Raspberry-Pi-and-other-MIPI-DSI-display-panels  
https://github.com/raspberrypi/linux/issues/2521 
https://github.com/raspberrypi/linux/pull/2693

The necessary procedures to configure the DSI display (since the Official Raspberry Pi 7" Touchscreen Display are already supported in year 2015, i think kernel 4.0 and above should not have any issue in supporting the MIPI DSI display interface) can be understood below,

a. A working vc4_dsi (vc4_dsi.c) is needed to be able to support initialization of other display devices on the 15way DSI connector.

b. Use the kernel v4.20.y code reference, panel-ilitek-ili9881c.c (https://github.com/raspberrypi/linux/blob/rpi-4.20.y/drivers/gpu/drm/panel/panel-ilitek-ili9881c.c ) or panel-raydium-rm68200.c (https://github.com/raspberrypi/linux/blob/rpi-4.20.y/drivers/gpu/drm/panel/panel-raydium-rm68200.c ) as reference and put ili9881 initialization code to newly created driver “panel-tm050jdhg33.c“

c. build the kernel with this driver and update to the system.

d. Configure the firmware to load this driver using the MIPI-DSI interface in the “config.txt”,

dtoverlay=vc4-kms-v3d
ignore_lcd=1

From my understanding, we need the driver "vc4_dsi" to load the "driver panel-raydium-rm68200.c", but what is the minimum required kernel version in order to support this activity? What is the functional difference between "vc4-kms-v3d" and "vc4_dsi"?

The Official Raspberry Pi 7" Touchscreen Display is using the MIPI DSI interface also, how does the kernel know when the display is connected to the RPI board and we do not need to add any lines of code in the "config.txt" in order for the kernel to recognize it?

JamesH65, 2019-02-13:
The official display is handled by the firmware on the VC4, not through any Linux driver. The firmware is closed source, so this approach is not available to third parties, so you will need to use the Linux route. I have no experience with the Linux driver route, so cannot help there.

jamwu, 2019-02-14:
@llks 
i am doing the same thing like you. 
I found some information here https://github.com/anholt/linux/issues/133 ,it seem this guy has done it already.
but when i used anholt's repo -kernel 4.9.61,i can load the custom lcd driver,but can not fix the transfer interrupt wait timeout problem.
because i am a newbie in Linux driver,and there is too little imformation for custom a raspbery's dsi display. it's been weeks without progress.
hope the above information can help you.and please let me know if you make some progress.

aromanro, 2019-05-15:
First, you'll have to take one of the 'example' panel sources and write a similar one customized for your panel. You might have a different resolution, clock, timings, number of DSI lanes, etc. You probably need to reset your panel when initializing and that depends on your particular setup. Many panels require some dsi commands for initialization and the commands are panel dependent. I cannot give many details here.

After you got your panel kernel module, you'll have to have it loaded and connected to the vc4 driver. For that, you'll have to write an overlay. I had success for loading my panel driver with something like this (this is for the DSI1 port, for 0 modify accordingly):

```
/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

    fragment@0 {
        target=<&dsi1>;

        __overlay__ {
            status = "okay";

            #address-cells = <1>;
            #size-cells = <0>;

            port {
                dsi1_out_port: endpoint {
                    remote-endpoint = <&panel_dsi_in>;
                };
            };

            panel_test: panel@0 {
                compatible = "manufacturer,model";
                reg=<0>;
                label= "test_display";
                vc4_dsi = <&dsi1>;
                port {
                    panel_dsi_in: endpoint {
                        remote-endpoint = <&dsi1_out_port>;
                    };
                };
            };
        };
    };
};
```
where manufacturer,model must be changed to match your panel driver.
You'll have to add
dtoverlay=your_driver
in config.txt (where your_driver is your compiled overlay) as well as
dtoverlay=vc4-kms-v3d

This will get your driver bound to the driver and probed, but from here on I have no idea what comes next, I'm trying to send some commands to the panel and I'm getting 'transfer interrupt wait timeout'. 

Here is some info: https://github.com/anholt/linux/wiki/Raspberry-Pi-and-other-MIPI-DSI-display-panels

If anybody has more info on this, please share it!

aromanro, 2019-05-15:
Just a while after I wrote the above, I managed to send the commands to the panel.

I needed to add to the overlay this:
`power-domains = <&power 18>;`
18 is for dsi1, for dsi2 the value is 17

aromanro, 2019-05-16:
Does anybody know why such panel would work great with the kms driver but not with fkms?

popcornmix, 2019-05-16:
kms = display driven by kernel
fkms = display driven by gpu firmware

There is support for the display in the kernel but not the gpu firmware.

aromanro, 2019-05-16:
I understand that, but there are panels out there that work with both.

6by9, 2019-05-16:
Please don't ask the same question on the forum and here - you just waste our time as you may get duplicated answers (as you have).
https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=240389#p1468790

JamesH65, 2019-05-16:
Answered on forum

closed 2019-05-16 by JamesH65