raspberrypi/linux #2856: Recent changes to GPIO14/15 may have stopped Uart initialisting on DPI systems

willdean, 2019-02-13:
**Describe the bug**

We're running 4.19 on a CM3+, with UART0 mapped to GPIO36/37.

The relevant config.txt line is

`dtoverlay=uart0,txd0_pin=36,rxd0_pin=37,pin_func=6`

We're seeing that the UART doesn't get initialised, with the following errors in dmesg

```
[    0.737582] pinctrl-bcm2835 3f200000.gpio: pin gpio14 already requested by leds; cannot claim for 3f201000.serial
[    0.737883] pinctrl-bcm2835 3f200000.gpio: pin-14 (3f201000.serial) status -22
[    0.738082] pinctrl-bcm2835 3f200000.gpio: could not request pin 14 (gpio14) from group gpio14  on device pinctrl-bcm2835
[    0.738373] uart-pl011 3f201000.serial: Error applying setting, reverse things back
[    0.738598] uart-pl011: probe of 3f201000.serial failed with error -22
```
The serial port should not be trying to talk to GPIO14 - that's running a DPI LCD panel, and we are wondering if it is trying to talk to GPIO14 because of the following patch:

https://github.com/raspberrypi/linux/commit/68accec9c8b29df730ae099ee30e92e9948643f5#diff-2991a7f22e210883f9428277cb77ede7

I don't fully understand how this change works, or what's going on, but we'd like UART0 not to touch GPIO14/15 at all.  Or, at least, not fail if it finds it can't touch GPIO14/15.

We're not clear about the origin of the "pin gpio14 already requested by leds" message - there aren't any LEDs we're aware of, though the GPIO14 is in use elsewhere.

We feel as though this didn't happen with the latest 4.14, though we're in a very complex mix of moving parts at the moment, and we could be wrong about this.

I've attached a raspinfo output.

pelwell, 2019-02-14:
The uart0 overlay is using pinctrl to return the original UART pins 14 & 15, as initialised by the firmware, back to input. This has the side effect of leaving them claimed. The dpi overlays also use pinctrl, via the "gpio-leds" driver, to enable the DPI function on the same pins, also claiming them. One of them has to lose, and in this case it's the DPI overlay.

Without changing the firmware, the easiest way forward for you is to copy the old version of the overlay and give it a unique name that won't be used by anyone else, to prevent it being overwritten by future updates.

willdean, 2019-02-14:
Thanks @pelwell, in our case it's actually the serial port losing rather than the DPI but I get the point and we'll try the suggested fix, thanks.

It seems like this change (I do appreciate what the motivation for it was) will bite anyone whose boot process allocated GPIO14/15 to something else before loading the uart overlay.   Is it somehow possible for the failure to allocate these pins (solely for the purpose of setting them as input)  to be 'soft' - i.e. if they can't be allocated then things will just plough-on anyway?

Or could there be some optional overlay parameter which enabled/disabled this behaviour, perhaps?

pelwell, 2019-02-14:
I like the overlay parameter idea - it's easy enough to do. Give me a moment...

pelwell, 2019-02-14:
On second thoughts, this a problem created by the firmware (and its dt-blob), so it should be solved in the firmware. The fix is going to take a bit more thought, but the intention is to revert the offending commit and let the firmware undo any contention.

willdean, 2019-02-14:
@pelwell - So the plan is to make the firmware release/set-to-input these pins, at which point the uart overlay won't need to touch them at all if it's actually mapped to different pins?

That does makes sense if so - it does seem to avoid fix upon fix upon fix.

pelwell, 2019-02-14:
Yes, that's the plan. In an ideal world this would all be left to Device Tree, but the firmware configures (and can use) the UART, and other OS's may depend on the pins being configured before they are started.

JamesH65, 2019-09-19:
@pelwell Did this plan ever come to fruition?

pelwell, 2019-09-19:
No, but recent change to the order of file loading by the firmware should have made it a more tractable problem.

pelwell, 2019-09-28:
Another victim: https://www.raspberrypi.org/forums/viewtopic.php?t=252911
This issue is now top of the list.

arthurlavalabs, 2019-09-28:
Please note my report as well: https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=252389

pelwell, 2020-01-30:
The underlying cause of this issue has been fixed in the firmware since October. The required overlay change has been in the 5.4 tree since then, but wasn't merged to 4.19 to give users time to update their firmwares. Sufficient time as now passed, and the modified overlay has been merged to 4.19.

popcornmix, 2020-01-30:
rpi-update contains the potential fix. Can you test it?