raspberrypi/linux #2858: [PiZero] After a couple of reboots, WiFi doesn't come up anymore

agherzan, 2019-02-15:
**Describe the bug**
After a couple of reboots, usually 2-5, the wifi dongle doesn't expose a wifi interface anymore and `dmesg` reports USB errors as it follows:

```
pi@raspberrypi:~$ dmesg |grep -i usb
[    0.132776] usbcore: registered new interface driver usbfs
[    0.132922] usbcore: registered new interface driver hub
[    0.133154] usbcore: registered new device driver usb
[    0.356701] usbcore: registered new interface driver smsc95xx
[    0.390035] dwc_otg 20980000.usb: base=0xf0980000
[    0.842107] dwc_otg 20980000.usb: DWC OTG Controller
[    0.844754] dwc_otg 20980000.usb: new USB bus registered, assigned bus number 1
[    0.847462] dwc_otg 20980000.usb: irq 56, io mem 0x00000000
[    0.855589] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    0.858255] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.860806] usb usb1: Product: DWC OTG Controller
[    0.863394] usb usb1: Manufacturer: Linux 4.14.79+ dwc_otg_hcd
[    0.865977] usb usb1: SerialNumber: 20980000.usb
[    0.869616] hub 1-0:1.0: USB hub found
[    0.876245] usbcore: registered new interface driver usb-storage
[    0.927663] usbcore: registered new interface driver usbhid
[    0.930255] usbhid: USB HID core driver
[    2.246421] usb 1-1: new full-speed USB device number 2 using dwc_otg
[    2.492954] usb 1-1: device descriptor read/64, error -71
[    2.832977] usb 1-1: device descriptor read/64, error -71
[    3.172930] usb 1-1: new full-speed USB device number 3 using dwc_otg
[    3.412559] usb 1-1: device descriptor read/64, error -71
[    3.752983] usb 1-1: device descriptor read/64, error -71
[    3.873157] usb usb1-port1: attempt power cycle
```

```
pi@raspberrypi:~$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

When in this state, only unplug/plug fixes the issue.

Tested this with both updated Raspbian Stretch and Yocto based distro. 

**To reproduce**

Have a similar setup to:

-  Pi power supply
-  PiZero (regular)
-  ThePiHut USB-OTG adapter + PiHut wifi dongle
-  wifi config added
-  enable_uart=1 added to config.txt
-  connect to the device over the UART

.. and reboot the 2-5 times.

**Expected behaviour**
No USB errors and functional wifi interface.

**Actual behaviour**
After 2-5 reboots, USB errors appear and wifi interface disappears. 

**System**
* Which model of Raspberry Pi? PiZero
* Which OS and version (`cat /etc/rpi-issue`)? Raspberry Pi reference 2018-11-13
* Which firmware version (`vcgencmd version`)?
Nov  4 2018 16:31:07 
Copyright (c) 2012 Broadcom
version ed5baf9520a3c4ca82ba38594b898f0c0446da66 (clean) (release)
* Which kernel version (`uname -a`)? Linux raspberrypi 4.14.79+ #1159 Sun Nov 4 17:28:08 GMT 2018 armv6l GNU/Linux
[dmesg.txt](https://github.com/raspberrypi/linux/files/2869377/dmesg.txt)

**Logs**
If applicable, add the relevant output from `dmesg` or similar. (attached)

JamesH65, 2019-09-19:
@agherzan That's quite old kernel and firmware, do you know if the issue still happens on the latest kernel?

agherzan, 2019-09-19:
When the issue was created, I'm pretty sure it was not an old kernel. We will try to replicate on the current OS and see if we can close this.

JamesH65, 2019-09-19:
Sorry, I meant old relative to the current one, which has a considerable number of fixes in it.

agherzan, 2019-09-20:
We reproduced this on the latest Raspbian.

```
pi@raspberrypi:~$ uname -a Linux raspberrypi 4.19.57+ #1244 Thu Jul 4 18:42:50 BST 2019 armv6l GNU/Linux pi@raspberrypi:~$ cat /etc/rpi-issue Raspberry Pi reference 2019-07-10 Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage5 
```

agherzan, 2022-02-12:
It looks like an old issue - most probably outdated. Closing for now.

closed 2022-02-12 by agherzan