raspberrypi/linux #2859: bcm2835-v4l2 - overlay / viewfinder

zoff99, 2019-02-16:
```
# viewfinder
v4l2-ctl --overlay=1 # enable viewfinder
v4l2-ctl --overlay=0 # disable viewfinder
```
are there any more options to this overlay? or is it possible to implement some?
we are working on a video conferencing system, where this would be the best solution to see your own video.
but it is in the center of the screen, and also very large.

can we somehow put the viewfinder in a corner and make it a bit smaller?
any help welcome ...

6by9, 2019-02-16:
It's actually a feature that is being removed as we are looking towards switching from the legacy to the KMS driver where its not possible for random overlays to be added via other subsystems that fancy it.

You can alter it, but I'm not sure if it is exposed to userspace properly.
The relevant bits of the driver you're after are https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/staging/vc04_services/bcm2835-camera/bcm2835-camera.c#L1897 and remove the ```MMAL_DISPLAY_SET_FULLSCREEN``` flag from https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/staging/vc04_services/bcm2835-camera/bcm2835-camera.c#L675

It looks like in theory you can modify the destination rectangle as defined in https://www.linuxtv.org/downloads/v4l-dvb-apis-new/uapi/v4l/dev-overlay.html#overlay-window - vidioc_s_fmt_vid_overlay is defined. However I think that ```MMAL_DISPLAY_SET_FULLSCREEN``` flag is going to get in the way again.

zoff99, 2019-02-16:
is this the thing i need, but just not exposed to the v4l2 driver:

https://github.com/raspberrypi/userland/tree/master/host_applications/linux/apps/raspicam

--preview,	-p   	Preview window settings <'x,y,w,h'>

zoff99, 2019-02-16:
@6by9 are u sure those linenumbers are correct?

6by9, 2019-02-16:
Sorry, I'd copied from the 5.0 branch and only changed the branch in the URL.
https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/staging/vc04_services/bcm2835-camera/bcm2835-camera.c#L1934
https://github.com/raspberrypi/linux/blob/rpi-4.14.y/drivers/staging/vc04_services/bcm2835-camera/bcm2835-camera.c#L731
I'm wrong over the MMAL_DISPLAY_SET_FULLSCREEN flag - it is correct to set that as the actual parameter is then set to 0/false at line 734.

Raspicam -p function is setting exactly the same parameters on video_render.
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiPreview.c#L122
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/raspicam/RaspiPreview.c#L129

zoff99, 2019-02-16:
cool that seems easy.
is there a docu for dummies how to compile just this kernel module on a pi ?

zoff99, 2019-02-16:
so exposing these 4 params to v4l2 would be awesome:
```
		dev->overlay.w.left = 150;
		dev->overlay.w.top = 50;
		dev->overlay.w.width = 1024;
		dev->overlay.w.height = 768;
```

6by9, 2019-02-16:
You've got me wondering if it is or isn't exposed - it's not something I've looked at recently.

Try ```v4l2-ctl ----get-fmt-overlay``` which should return those numbers.
If so, then ```v4l2-ctl --set-fmt-overlay=top=0,left=0,width=640,height=480``` may reduce the size of the overlay and set it at top left. (```v4l2-ctl --help-overlay``` to get the full help on overlay commands).

zoff99, 2019-02-16:
@6by9 wow. thanks mate!! that actually works

closed 2019-02-16 by zoff99