raspberrypi/linux #2863: initramfs script output not sent to serial console

kpfleming, 2019-02-18:
**Describe the bug**
When both serial and virtual (HDMI) consoles are configured, kernel startup messages appear on both consoles but output generated by scripts in the iniramfs only appears on the virtual console.

**To reproduce**
1. Configure an initramfs using initramfs-tools and an 'initramfs' line in /boot/config.txt.
2. Enable serial console and virtual console in /boot/cmdline.txt: `console=serial0,115200 console-tty1`.
3. Reboot and observe both consoles.

**Expected behaviour**
Kernel messages (`printk`) and stdout/stderr from scripts in initramfs should appear on both consoles.

**Actual behaviour**
Only kernel messages appear on serial console.

**System**
* Which model of Raspberry Pi? Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)? Raspbian 9.8
* Which firmware version (`vcgencmd version`)? 1.20181112-1
* Which kernel version (`uname -a`)? Linux home18 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

config.txt
```
# Enable serial port
enable_uart=1

# Use initramfs for mounting root filesystem
initramfs initramfs.gz

# Configure the activity LED to show SD card activity
dtoverlay=act_led_trigger=mmc1

# Disable Bluetooth hardware
dtoverlay=pi3-disable-bt

# Disable WiFi hardware
dtoverlay=pi3-disable-wifi
```
Problem also occurs if cmdline specifies `ttyAMA0` for console.

JamesH65, 2019-09-19:
I'm not sure from the description if this is what is expected on a kernel, or whether it's what all kernels on all devices do. ie Is it specific to the Pi kernel. Not even sure how I would find out!

kpfleming, 2019-09-20:
That's a valid point! I don't have access to a system with a built-in serial port that is *not* a Pi in order to compare the behavior. I'll try to find someone who does.

JamesH65, 2020-02-13:
@kpfleming Did you have any luck with this?

burtyb, 2020-02-13:
I figured it was normal - on a Pi Zero to debug initramfs over serial to see the commands/output I've been appending " debug=y console=serial0" to cmdline.txt with " break=<phase>" for a BusyBox prompt.

kpfleming, 2020-02-16:
Unfortunately I have not had time to test this and my original project which required serial output is no longer needed.

closed 2020-11-20 by kpfleming

JeffPeters, 2023-10-26:
Try swaping your commandline
form ```console=serial0,115200 console-tty1```
to ```console-tty1 console=serial0,115200```

initramfs init script is only using the last console (as there is only one for that part of th boot)

This works for me to get initramfs working in qemu (which is using serial port in termal -nographic mode)

But it should also work on the pi zero

See: https://unix.stackexchange.com/a/736891