raspberrypi/linux #2868: No sound with VC4 on kernel 5.0.0-rc7

Dark-Sky, 2019-02-22:
I upgraded to the latest kernel 5.0.0-rc7 and the sound does not seem to work with VC4 via HDMI I have used it on prior kernels and it worked pretty good (a little scratchy). All of the sound modules seems to be loaded and the system see's everything I have seen in the past as to the hardware.When not booted into VC4 mode the sound works flawlessly with the new 5.0.0-rc7 kernel on my pi3b+.
 
Journal:
 
```
dbus-daemon[181]: [system] Activating via systemd: service name='org.freedesktop.RealtimeKit1' unit='rtkit-daemon.service' requested by ':1.30' (uid=1000 pid=304 comm="/usr/bin/pulseaudio --daemonize=no ")
rtkit-daemon[315]: Successfully made thread 304 of process 304 (/usr/bin/pulseaudio) owned by '1000' high priority at nice level -11.
rtkit-daemon[315]: Successfully made thread 325 of process 304 (/usr/bin/pulseaudio) owned by '1000' RT at priority 5.
pulseaudio[304]: XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
pulseaudio[304]:       after 16 requests (16 known processed) with 0 events remaining.
systemd[242]: pulseaudio.service: Main process exited, code=exited, status=1/FAILURE
systemd[242]: pulseaudio.service: Failed with result 'exit-code'.
systemd[242]: pulseaudio.service: Service RestartSec=100ms expired, scheduling restart.
systemd[242]: pulseaudio.service: Stop job pending for unit, delaying automatic restart.
systemd[242]: pulseaudio.socket: Succeeded.
```

Dark-Sky, 2019-02-22:
Forgot the ls mod:
 
```
[alarm@alarmpi ~]$ lsmod
Module                  Size  Used by
fuse                  114688  3
brcmfmac              307200  0
brcmutil               16384  1 brcmfmac
sha256_generic         20480  0
vc4                   159744  6
cfg80211              610304  1 brcmfmac
drm_kms_helper        180224  3 vc4
rfkill                 28672  4 cfg80211
drm                   425984  5 vc4,drm_kms_helper
drm_panel_orientation_quirks    16384  1 drm
snd_soc_core          180224  1 vc4
raspberrypi_hwmon      16384  0
bcm2835_v4l2           57344  0
hwmon                  28672  1 raspberrypi_hwmon
snd_compress           20480  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
v4l2_common            16384  1 bcm2835_v4l2
syscopyarea            16384  1 drm_kms_helper
videobuf2_vmalloc      16384  1 bcm2835_v4l2
sysfillrect            16384  1 drm_kms_helper
videobuf2_memops       16384  1 videobuf2_vmalloc
sysimgblt              16384  1 drm_kms_helper
videobuf2_v4l2         28672  1 bcm2835_v4l2
fb_sys_fops            16384  1 drm_kms_helper
videobuf2_common       53248  2 bcm2835_v4l2,videobuf2_v4l2
i2c_bcm2835            16384  0
videodev              200704  4 v4l2_common,videobuf2_common,bcm2835_v4l2,videobuf2_v4l2
media                  36864  3 videobuf2_common,videodev,videobuf2_v4l2
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
sch_fq_codel           20480  3
snd_bcm2835            24576  1
snd_pcm                94208  4 vc4,snd_pcm_dmaengine,snd_bcm2835,snd_soc_core
snd_timer              32768  1 snd_pcm
snd                    69632  9 snd_compress,snd_timer,snd_bcm2835,snd_soc_core,snd_pcm
ip_tables              28672  0
x_tables               32768  1 ip_tables
ipv6                  442368  52
```

lategoodbye, 2019-02-24:
@Dark-Sky Please provide output of `dmesg` and `aplay -l`

Dark-Sky, 2019-02-24:
Here is the out puts of aplay -l and dmesg. The last 5 bcm2835_audio entries in dmesg is what pops up when I tried to play a .mp3 file. I mentioned above there is no problem when I use the same image but disable the vc4 and reboot back into the regular fbturbo mode.

[aplay.txt](https://github.com/raspberrypi/linux/files/2898327/aplay.txt)

[dmesg.txt](https://github.com/raspberrypi/linux/files/2898328/dmesg.txt)

Dark-Sky, 2019-02-24:
Also just in case it is needed here is my .config. In case you are wondering in it I have modified modules for my dvb / atsc hobby and the capabilty for playing TBS devices.

[bcm2709-udl_defconfig.txt](https://github.com/raspberrypi/linux/files/2898343/bcm2709-udl_defconfig.txt)

Dark-Sky, 2019-02-24:
Sorry I posted from my AARCH64 sdcard. I know you do not support it. Trying to locate my 32 bit sdcard...... The same issis on my 32 bit install also.

Dark-Sky, 2019-02-24:
Here is aplay -l, dmesg and lsmod while booted with vc4 enabled and booted with out vc4 using the same image and my .config file. The.config file was created using your bcm2709_defconfig as a base and the only changes in it is the dvb and atsc modules as I am in to that has a hobby and there is more functionality. Some modules was added also so the TBS devices would work. Like I said above when booted into the same image without vc4 enabled the sound plays just fine.

[aplay_no-vc4.txt](https://github.com/raspberrypi/linux/files/2898456/aplay_no-vc4.txt)
[aplay_vc4-enabled.txt](https://github.com/raspberrypi/linux/files/2898457/aplay_vc4-enabled.txt)
[dmesg_no-vc4.txt](https://github.com/raspberrypi/linux/files/2898458/dmesg_no-vc4.txt)
[dmesg_vc4-enable.txt](https://github.com/raspberrypi/linux/files/2898459/dmesg_vc4-enable.txt)
[lsmod_no-vc4.txt](https://github.com/raspberrypi/linux/files/2898462/lsmod_no-vc4.txt)
[lsmod_vc4-enable.txt](https://github.com/raspberrypi/linux/files/2898460/lsmod_vc4-enable.txt)
[bcm2709-udl_defconfig.txt](https://github.com/raspberrypi/linux/files/2898468/bcm2709-udl_defconfig.txt)

Dark-Sky, 2019-02-25:
I disabled pulseaudio to day and the sound worked good with alsa playing a .mp3 and my dvb app so looking like a pulseaudio issue.

[image]
[image]

lategoodbye, 2019-03-01:
Looking at the screenshots show me that you are using Arch Linux?
Can you point me to the used rootfs?

Dark-Sky, 2019-03-01:
I will create an image and upload it to my googledrive for you.

lategoodbye, 2019-03-01:
Thanks. Since i never tested the combination pulseaudio + vc4_hdmi + Aarch64, could you please tell which kernel version(s) worked with vc4_hdmi for you?

Dark-Sky, 2019-03-01:
Just to be clear the kernel image I am uploading right now is 32 bit. I am not sure which kernel version I know where the sound worked but I can try to see and compile it tomorrow on my AARCH64 image. When I do I can also upload that image also for you. Here is a snapper of my AARCH64 with VC4 but the sound is not working either because it is kernel version 5.0.0-rc7:

[image]

Dark-Sky, 2019-03-02:
lategoodbye:

Here is the 32 bit arch linux image. There is @163M free space in the image. There is no such thing as raspi-config in arch but you need to expand the root file system to fill out your sdcard after burning the image either by gparted on your desktop or boot in to the image and follow the instructions on the pi's desktop to expand the root file system live while booted up. There is a README that may answer most of youe questions. It will boot into fbturbo mode so you can the sound works with a test file on the Desktop. Follow the instructions on the desktop to switch to VC4 mode and reboot and you can see what I am dealing with on the sound in VC4 mode.

This image will only work on a pi that has a v7 chip

There is a file also on the desktop to set your time zone.

Setting your locale:

https://wiki.archlinux.org/index.php/Locale

Keyboard:

https://wiki.archlinux.org/index.php/Linux_console/Keyboard_configuration
https://wiki.archlinux.org/index.php/Xorg/Keyboard_configuration

1.3G
451fd2eb9684426dcfaf3d4203f19224  rpi_2-3-3b+_5-0-rc7_3-1-19.zip
https://drive.google.com/file/d/1PdDlEKAaszVb7s5zFbATIUVlNxSURVKk/view?usp=sharing

lategoodbye, 2019-03-02:
Sorry, about the confusion i assumed that you use mainline or Arch Linux kernel. Please understand that i cannot support custom kernels. Can you retest it with recent mainline from kernel.org or from this repo?

Dark-Sky, 2019-03-03:
As I said above I use the Raspian kernel and the only things changed in the tree are the dvb/ATSC devices modules under drivers/media. I did a pull this mourning of the Raspian 5.0.0-rc8 branch and compiled it using make "bcm2709_defaultconfig" and changed nothing in the tree and it is doing the exact same thing with not being able to use pulseaudio but does alsa fine with pulse disabled. I looked back and found that the Raspian kernel 4.20-rc6 that I had saved in an image will play pulseaudio. It was not the best quality but tolerable.

Here is a screenshot showing Raspian's 5.0.0-rc8 doing the exact same thing and a clip from dmesg:

```
[   40.792398] bcm2835_audio bcm2835_audio: vchi message timeout, msg=5
[  166.234478] bcm2835_audio bcm2835_audio: vchi message timeout, msg=3
[  176.474488] bcm2835_audio bcm2835_audio: vchi message timeout, msg=3
[  186.714518] bcm2835_audio bcm2835_audio: vchi message timeout, msg=3
```
[image]

Dark-Sky, 2019-03-04:
lategoodbye:

Today I compiled kernel.org's mainline/stable kernel. As of yesterday they are the same version 5.0.0. And the pulseaudio audio worked just fine with vc4:

[image]

lategoodbye, 2019-03-04:
Thanks, please the used kernel defconfig (only the name not the all the options) to the mainline test results.

Dark-Sky, 2019-03-04:
Not sure what you are asking but I used "make multi_v7_defconfig" as suggested in the gentoo wiki here.

[](https://wiki.gentoo.org/wiki/Raspberry_Pi/Mainline_Kernel)

I basically followed the guide to a tee.

Dark-Sky, 2019-03-04:
I forgot I added some flags as I do not know if the issue has been fixed or not:

https://www.raspberrypi.org/forums/viewtopic.php?t=228012

export MAKEFLAGS="-j4"
export DTC_FLAGS="-@ -H epapr"

Dark-Sky, 2019-03-05:
latedoodnight:

I changed to fbturbo mode with the new upstream kernel and tried to play the music file with omxplayer and it gave an error about no vchiq instance so I changed one line in the .config and recompiled. It recompiled all of the .dtb's and the vchiq module and omxplayer worked:

From:
#CONFIG_BCM2835_VCHIQ is not set
To:
CONFIG_BCM2835_VCHIQ=m

Then:
make -j4
sudo make modules_install
sudo make dtbs_install
sudo reboot

lategoodbye, 2019-03-05:
The VCHIQ driver is disabled in multi_v7_defconfig because it's still in staging, but this driver has nothing to do with VC4 HDMI.

Dark-Sky, 2019-03-05:
The VCHIQ driver may not have anything to do with VC4 HDMI but it throws out an error when in VC4 mode with bcm2835_audio causing the bcm2835_audio to not work:

[   40.792398] bcm2835_audio bcm2835_audio: vchi message timeout, msg=5
[  166.234478] bcm2835_audio bcm2835_audio: vchi message timeout, msg=3

It seems that the latest in the Raspian tree has an issue with the vchiq / snd_bcm2835 enabled at the same time vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 is enabled. I enabled vchiq / snd_bcm2835 in Torvald's tree and ran into the exact same problem with the sound as the raspian tree.

Since you mentioned VCHIQ is in staging I revisited my Raspian kernel and got the sound to work very well in VC4 mode using pulseaudio by blacklisting vchiq & snd_bcm2835 and rebooting:

```
[alarm@alarmpi ~]$ cat /etc/modprobe.d/vchiq.conf 
blacklist vchiq
blacklist snd_bcm2835
```
Now the audio device shows up exactly as in using Torvald's kernel:

```
[alarm@alarmpi Desktop]$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
[image]

[image]

Now the audio is flawless with pulseaudio:

[image]

Here mpv is playing a 18Mbit video stream in VC4 mode:

[image]

lategoodbye, 2019-03-05:
Thank you for investigations

Dark-Sky, 2019-03-05:
I followed the above changes and my Raspian AARCH64 kernel now has pulseaudio sound back again and sounds excellent:

[image]

[image]

Dark-Sky, 2019-03-16:
lategoodnight:

I believe this issue is now fixed with yesterday's commit (revert). I pulled in the changes and compiled the new 5.0.2 kernel and all seems to be good with VC4 mode and pulseaudio and nothing seems to be broke using fbturbo mode.

[](https://github.com/raspberrypi/linux/commit/709198a03e40bbf035f75772e9162464dc7fc665)

There is one separate issue I forgot to mention is when in VC4 mode my Xorg log gets spammed heavily with:
```
[   901.979] (EE) modeset(0): Failed to get GBM bo for flip to new front.
[   901.979] (EE) modeset(0): present flip failed
```
I have to create a /etc/X11/xorg.conf.d/20-displaylink.conf file to disable the spam:

```
Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection
```
Screen shots with Raspian's new mods:

aplay -l fbturbo mode:

[image]

aplay -l VC4 mode:

[image]

If you do not change the default audio input / output device to "Audio Digital Stereo" in the system setting the audio will be scratchy:

[image]

[image]

I have not had time to compile the new Raspian kernel changes and test with my AARCH64 image yet but When I do I will report back.

Thanks, 
Ray

Dark-Sky, 2019-03-16:
Compiled the new Raspian kernel 5.0.2 for my AARCH64 and it also works now with VC4 and pulseaudio.

closed 2019-03-17 by Dark-Sky

lategoodbye, 2019-03-17:
@Dark-Sky Please report VC4 issue ( "present flip failed" ) to https://github.com/anholt/linux