raspberrypi/linux #2869: CVE-2019-8912 security bug - all kenrels through 4.20.11

Blub86, 2019-02-25:
Hallo,
I don’t really know if this belongs here but, a use-after-free issue was found in the networking subsystem's sockfs code of the Linux kernel:

https://security-tracker.debian.org/tracker/CVE-2019-8912

https://nvd.nist.gov/vuln/detail/CVE-2019-8912

This should be patched as soon as a patch is available.

6by9, 2019-02-25:
It already has been. https://github.com/raspberrypi/linux/commit/eb5e6869125f69dd28513f92992d97ec62bb9773#diff-bb4f94f3beef255f6acd61800ee30ca9

We deliberately adopt the LTS (Long Term Support) Linux kernel releases so that fixes such as this get backported by those upstream.
Looking at https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/?h=linux-4.19.y that has been done in 4.19.25 (patch "net: crypto set sk to NULL when af_alg_release").
The rpi-4.19.y branch was rebased to 4.19.25 yesterday (24/12/2019), and released via rpi-update.

Due to resources we only maintain one kernel version at a time as our current branch, and that changed from rpi-4.14.y to rpi-4.19.y last week. 

The Raspbian kernel gets bumped less frequently as stability is the key factor there. There may be a case to bump the 4.14 branch from 4.14.98 to 4.14.103 to get the corresponding fix, but I'll leave that for others to decide.
The CVE is lacking on how significant the bug is seeing as this is crypto userspace APIs, so on the face of it you need to be logged into the machine to be able to exploit it at all.

JamesH65, 2019-09-19:
Closing this issue as questions answered/issue resolved.

closed 2019-09-19 by JamesH65