raspberrypi/linux #2870: Infrared touch bug

Knisy, 2019-02-26:
When I use the infrared touch screen, the click will be selected, it will remember the location of the last click and the location of the current click to do the checkbox selection, but not in other operating systems, such as Ubuntu, windows
[image]

Knisy, 2019-02-26:
How to solve this problem

pelwell, 2019-02-26:
So what you are saying is that an infrared touchscreen works as you want it to in Raspbian but not on other platforms? That isn't a bug, and definitely not a Linux on Raspberry Pi bug. Please open a topic at the Raspberry Pi Forums.

closed 2019-02-26 by pelwell