raspberrypi/linux #2872: Missing linux-headers-4.19.23-v7

dkerr64, 2019-02-27:
I just updated to the new firmware for my 3+ which brings down kernel 4.19.23.  I need to rebuild some modules (wireguard) and it fails because headers are missing.

```
pi@raspberrypi:~ $ sudo apt install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-headers-4.19.23-v7
E: Couldn't find any package by glob 'linux-headers-4.19.23-v7'
E: Couldn't find any package by regex 'linux-headers-4.19.23-v7'
pi@raspberrypi:~ $
```
Help !

David

JamesH65, 2019-02-27:
4.19 is still in test, so apt will not have been updated to it. Did you rpi-update? We don't really recommend that in general usage. 

I'd suggest reverting to an earlier supported kernel.

dkerr64, 2019-02-27:
Yes I did an rpi-update.  What is the easiest way to revert to an earlier kernel?
Thanks
David

JamesH65, 2019-02-27:
I think an `apt update` ` apt dist-upgrade` will load the latest release kernel

XECDesign, 2019-02-27:
```
apt install raspberrypi-kernel{,-headers} 
```

dkerr64, 2019-02-27:
Thanks for the suggestions.  I found the answer here... https://www.raspberrypi.org/documentation/linux/kernel/updating.md

Given this was me getting ahead of things... should I close out this issue or leave it open?

JamesH65, 2019-02-27:
Please close if sorted out.

dkerr64, 2019-02-27:
Thanks, yes I am going again with the prior kernel.
Thanks for your help.
David

closed 2019-02-27 by dkerr64