raspberrypi/linux #2876: `fb_sh1106` driver not compiled in by default; consider enabling?

CRImier, 2019-02-27:
On default Raspbian Lite install, `fb_ssd1306` kernel module is installed but `fb_sh1106` (almost-identical display but cheaper and more popular) is not available. Has been happening for a couple of releases now. We could compile and add it by default in ZeroPhone SD card images, but it's a lot of hassle compared to what I'm proposing - setting it to `'y'` by default in upstream and allowing all the other users to use it, too.

```
pi@zerophone:~/ZPUI $ zgrep "SH1106" /proc/config.gz
# CONFIG_FB_TFT_SH1106 is not set
```

**System**

`raspinfo` command not installed by default on Raspbian Lite. Uname: `Linux zerophone 4.14.98+ #1200 Tue Feb 12 20:11:02 GMT 2019 armv6l GNU/Linux` 

```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

pelwell, 2019-02-27:
We are very unlikely to build this driver into the kernel - we aim to keep the core kernel as small as possible - but adding the module to the standard builds sounds reasonable. @popcornmix?

CRImier, 2019-02-27:
Ah sure, sorry, what I meant to propose is `"m"`, same as `fb_ssd1306`.

popcornmix, 2019-02-27:
No objection to adding this as a module.

pelwell, 2019-02-27:
It's added (https://github.com/raspberrypi/linux/commit/7f26b4456f70f9909c19936d550cf7c5dc47e1a5), and will be in the next firmware release.

CRImier, 2019-02-27:
Thank you! When will the next FW release be, if that's not a secret? How will it be distributed, `apt install` or `rpi-update`? (wasn't `rpi-update` outdated?)

pelwell, 2019-02-28:
Firmware releases are usually no more than a week apart, and they are available first via `rpi-update`.

popcornmix, 2019-03-11:
@CRImier this option is in latest rpi-update kernel. Can you test close if happy.

JamesH65, 2019-09-19:
Please update to the latest kernel which may contain a fix for this issue.

closed 2019-09-19 by JamesH65