raspberrypi/linux #2877: Please add Movian package

csgabe, 2019-02-27:
Please add Movian application package:
https://github.com/andoma/movian

pelwell, 2019-02-27:
Movian is an application, which by definition is not part of the Linux kernel, so your request does not belong here. We are also unlikely to add it to the default Raspbian images, when Movian offers standalone SD card images.

closed 2019-02-27 by pelwell

XECDesign, 2019-02-27:
If you would like it added to the repo, you can open an issue here - https://github.com/RPi-Distro/repo/issues/

It seems to have debian packaging already, so as long as it builds properly for the pi, passes lintian, adequate and piuparts, it could be added.

csgabe, 2019-02-27:
@pelwell 
Sorry for posting in wrong place.
@XECDesign 
OK, I'll open an issue there.

czz, 2019-07-11:
here is a deb package http://mp.czz78.com/rpi.html

XECDesign, 2019-07-11:
Looks good at a glance. Is there the .dsc and associated files to go along with it somewhere?

czz, 2019-07-11:
i have to modify some part of https://github.com/andoma/movian to compile it. most are fixes for gcc >= 8.  I might push up code on git, but it will be only for Raspberry so it will be easier to compile. I'll ask Andoma what he thinks about it before uploading code.