raspberrypi/linux #2879: arm7l distro in Raspbian missing module (usb_f_fs ) no write access to configFS /sys/kernel/config/

dernyn, 2019-02-28:
I know that if I compile my own and chose this module to be added when building this works.
I want this native support as it was prior to Raspbian 8

There is no access to module (usb_f_fs ) otherwise and no access to configFS /sys/kernel/config/ from userspace.

can we get this natively working on the Lite and the Full version of stretch?

pelwell, 2019-02-28:
You're going to have to unpack that a bit. Please explain what used to work, and in what version, that now doesn't work any more, and why it was useful.

dernyn, 2019-02-28:
So on older Raspbian distros era 2015 distros, there was an ability to custom your own USB OTG Gadget mode devices with your own strings of hardware IDs natively . In order for one to do this task, the root user needs to be given access from a kernel module (usb_f_fs) under user-space in order to have write access to /sys/kernel/config/usb_gadget/*
Said kernel module is not distributed anymore out of the box.

To confirm it, do modprobe usb_f_fs
It’s actually missing on arm7l and arm6l— (I have to change the title.)

The ability to do that natively from the distro was lost after 2015-ish

Meaning that the usb_f_fs module was included at one point now it’s no longer is.

Now you have to compile your own kernel and included the modules with; make menuconfig

ED6E0F17, 2019-03-01:
I think that commit https://github.com/raspberrypi/linux/commit/8443f2d2b7782fef35fe579bf1eb612c24951486
changed "select" to "depends on", so this is an unintentional loss in the "min-config"

pelwell, 2019-03-01:
I can't see any substantive change to the Kconfigs as a result of applying https://github.com/raspberrypi/linux/commit/8443f2d2b7782fef35fe579bf1eb612c24951486 :
```
$ git checkout 8443f2d
Note: checking out '8443f2d'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 8443f2d... usb: gadget: Gadget directory cleanup - group legacy gadgets
$ find . -name Kconfig -exec cat {} \; | sort > with
$ git checkout HEAD^
Previous HEAD position was 8443f2d... usb: gadget: Gadget directory cleanup - group legacy gadgets
HEAD is now at 8ecef00... usb: renesas_usbhs: add R-Car Gen. 2 init and power control
$ find . -name Kconfig -exec cat {} \; | sort > without
$ diff with without
23644,23646d23643
< 
< 
< 
24665,24669d24661
< #
< #
< #
< #
< #
24726a24719,24720
> 	  
> 	   
25184d25177
< #    (a) a peripheral controller, and
35294d35286
< # both kinds of controller can also support "USB On-the-Go" (CONFIG_USB_OTG).
35376d35367
< #    (b) the gadget driver using it.
84157d84147
< #  - Host systems (like PCs) need CONFIG_USB (with "A" jacks).
95425d95414
< # NOTE:  Gadget support ** DOES NOT ** depend on host-side CONFIG_USB !!
97620d97608
< #  - Peripherals (like PDAs) need CONFIG_USB_GADGET (with "B" jacks).
116492d116479
< #  - Some systems have both kinds of controllers.
117578d117564
< source "drivers/usb/gadget/legacy/Kconfig"
135502d135487
< # USB Gadget support on a system involves
137860d137844
< # With help from a special transceiver and a "Mini-AB" jack, systems with
```
Can you point me at any kernel version in which either of the Pi defconfigs resulted in CONFIG_USB_F_FS being enabled?

dernyn, 2019-03-01:
Raspbian(Debian Wheezy) February 2015

pelwell, 2019-03-01:
Which is what kernel version?

dernyn, 2019-03-01:
3.18.11-v7+

dernyn, 2019-03-01:
release 2015-02-16

dernyn, 2019-03-01:
https://downloads.raspberrypi.org/raspbian/images/raspbian-2015-02-17/2015-02-16-raspbian-wheezy.zip

pelwell, 2019-03-01:
Raspbian Wheezy 2015-02-16 shipped with a 3.18.7 kernel. It did not include the usb_f_fs module.

I think your image must have a custom kernel (what does `dmesg | head -5` report shortly after boot?), and I believe we have never included a usb_f_fs module in a kernel release. If we did, you should be able to find it on the [firmware repo](https://github.com/raspberrypi/firmware).

dernyn, 2019-03-17:
interesting...I may had been mistaken.
can it be included for the armv6 architecture in bcmrpi_defconfig ---or for the (Pi Zero W)

it would be ideal to have this native along with all the other dw2 modules natively.

dernyn, 2019-03-17:
[image]

dernyn, 2019-03-17:
[image]

at least in the full version?

JamesH65, 2019-09-19:
@dernyn Do you have any figures for impact on kernel size and performance? Given a limited resources device like the Pi we like to ensure that kernel changes for smaller use cases do not have a detrimental impact.

dernyn, 2020-01-03:
very minimal, it's a simple usb host support at kernel level to be enabled when needed by userspace.