raspberrypi/linux #2880: Allo Katana last commits not updated on 4.19.y

allocom, 2019-03-01:
Hi @pelwell   we are verifying the 4.19.25 latest kernel and noticed previous 2 commits not updated on this new one.
sound/soc/bcm/allo-katana-codec.c
commit e40c24aa06283bcabf4a391c46818c0684f5fca6
commit e004542a37e90d2791a2ceef32cbd5c13d5ad7a5

pelwell, 2019-03-01:
Downstream drivers such as this have to be maintained by us, and occasionally an upstream API change means they no longer compile. What seems to have happened here is that the initial commit was fixed up, but that caused the first patch (https://github.com/raspberrypi/linux/commit/e40c24aa06283bcabf4a391c46818c0684f5fca6) to fail to apply cleanly. The second commit (https://github.com/raspberrypi/linux/commit/e004542a37e90d2791a2ceef32cbd5c13d5ad7a5) did apply, and was subsequently squashed into the initial commit.

I've modified the "unmute" patch and applied it to rpi-4.19.y. Please test it and report back, then I'll cherry-pick to the newer kernels.

allocom, 2019-03-01:
Hi @pelwell 
thanks for your quick response. basic test ok , we will update you after detailed testing.

allocom, 2019-03-05:
Hi @pelwell 
yes all are working. Please upadte it in newer kernels.

closed 2019-03-05 by allocom