raspberrypi/linux #2883: How to link the WIFI with Chinese SSID?

Joker2770, 2019-03-06:
**Describe the bug**
There is no way to link the WIFI with Chinese SSID.

**System**

Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.8 (stretch)
Release:        9.8
Codename:       stretch

* Which model of Raspberry Pi? 
Pi3B+

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2018-06-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 0d34fb94398a4bd79711f21d6a0b5d35fe5f4a76, stage5

* Which firmware version (`vcgencmd version`)?
Feb 22 2019 18:10:04
Copyright (c) 2012 Broadcom
version c32b049542c0da4e13087d1043bbfa6c50b242f3 (clean) (release) (start_x)

* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.25-v7+ #1205 SMP Mon Feb 25 18:19:20 GMT 2019 armv7l GNU/Linux

JamesH65, 2019-09-19:
@Joker2770 Is this still an issue? There is s lack of detail in your report. What specifically is the problem? What process are you going through to try and link, and where does it go wrong?

Joker2770, 2019-09-19:
Win10 system can show the Chinese SSID like this:
[image]

but on my raspberrypi, it change to be as follows:
[image]

~~~
\123\343\445\566\678
~~~

Like some code form but I don't know.

pelwell, 2019-09-19:
In your screenshot you seem to be connected to the SSID, even though it doesn't look pretty. Is this just a limitation of the LXPanel WiFi plugin?

Joker2770, 2019-09-19:
I tried to input my password, but link failed. I'm ensure it's my wifi.
[image]

pelwell, 2019-09-19:
@spl237 says that parsing of the long octal escape sequences had a bug (not @spl237's bug, of course), and a fix went into the relevant package yesterday. Try an update & upgrade, and the "Invalid argument" may go away.

Joker2770, 2019-09-19:
> @spl237 says that parsing of the long octal escape sequences had a bug (not @spl237's bug, of course), and a fix went into the relevant package yesterday. Try an update & upgrade, and the "Invalid argument" may go away.

Thanks for your explain.

pelwell, 2019-09-19:
Just hold off on the update - the plugin author (again, not @spl237) has found another bug...

Joker2770, 2019-09-19:
That's OK.

JamesH65, 2019-09-19:
Original code repo is here https://roy.marples.name/cgit/dhcpcd-ui.git/ and shows the fix.

Joker2770, 2019-09-19:
> Original code repo is here https://roy.marples.name/cgit/dhcpcd-ui.git/ and shows the fix.

Thanks for leading sincerely.

JamesH65, 2020-02-13:
@Joker2770 Was this fixed? If so please close the issue.

Joker2770, 2020-02-14:
@JamesH65 I have tried an update & upgrade, but it does not work.

[image]