raspberrypi/linux #2885: 4.19.y: bcm2835_codec/v4l2 module unload oops

jjsuwa, 2019-03-07:
When vc_sm allocation is met on boot, unloading both `bcm2835_codec` and `bcm2835_v4l2` (in no particular order) will occur kernel oops in latter module.

```
[    1.633192] [vc_sm_connected_init]: start
[    1.640403] [vc_sm_connected_init]: end - returning 0
[    5.524657] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[    5.524670] [vc_sm_connected_init]: start
[    5.531295] [vc_sm_connected_init]: installed successfully
.....
$ modprobe -r bcm2835_v4l2
$ modprobe -r bcm2835_codec
.....
[ 1045.482001] bcm2835-codec bcm2835-codec: Removing bcm2835-codec
[ 1045.482419] bcm2835-codec bcm2835-codec: Removing bcm2835-codec
[ 1045.482714] bcm2835-codec bcm2835-codec: Removing bcm2835-codec
[ 1045.507402] Unable to handle kernel NULL pointer dereference at virtual address 00000004
[ 1045.517725] pgd = 49c43cec
[ 1045.521742] [00000004] *pgd=36545835, *pte=00000000, *ppte=00000000
[ 1045.529224] Internal error: Oops: 817 [#1] PREEMPT SMP ARM
[ 1045.535831] Modules linked in: arc4 ecb md4 md5 sha512_generic cmac hmac nls_utf8 cifs ccm brcmfmac brcmutil sha256_generic spidev cfg80211 v4l2_mem2mem evdev rfkill videobuf2_v4l2 videobuf2_common videodev snd_soc_pcm5102a snd_soc_rpi_simple_soundcard snd_soc_bcm2835_i2s media vc_sm_cma(C-) snd_soc_core gpio_keys spi_bcm2835 raspberrypi_hwmon snd_compress hwmon snd_pcm_dmaengine i2c_bcm2835 snd_pcm snd_timer snd uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6 [last unloaded: bcm2835_mmal_vchiq]
[ 1045.589342] CPU: 2 PID: 8530 Comm: modprobe Tainted: G         C        4.19.27-v7+ #1
[ 1045.599972] Hardware name: BCM2835
[ 1045.604738] PC is at misc_deregister+0x54/0xb0
[ 1045.610556] LR is at 0x100
[ 1045.614619] pc : [<805a6558>]    lr : [<00000100>]    psr: 60000013
[ 1045.622293] sp : b85dde68  ip : 00000200  fp : b85dde84
[ 1045.628937] r10: 00000081  r9 : b85dc000  r8 : ba9b0244
[ 1045.635562] r7 : 7f2b4014  r6 : 80e44a74  r5 : 0000003f  r4 : b9ba4e84
[ 1045.643499] r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : 80d643a0
[ 1045.651429] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 1045.659997] Control: 10c5383d  Table: 367d006a  DAC: 00000055
[ 1045.667187] Process modprobe (pid: 8530, stack limit = 0xf83dc3c7)
[ 1045.674835] Stack: (0xb85dde68 to 0xb85de000)
[ 1045.680622] de60:                   7f2b42c0 ba248c10 ba9b0210 7f2b4014 b85dde9c b85dde88
[ 1045.691670] de80: 7f2b1644 805a6510 ba248c10 ba248c10 b85ddeb4 b85ddea0 805b7ee8 7f2b161c
[ 1045.702797] dea0: ba248c10 ba248c44 b85ddedc b85ddeb8 805b681c 805b7ec0 ba248c10 7f2b4014
[ 1045.713960] dec0: 004e1c2c 00000081 801011c4 b85dc000 b85ddef4 b85ddee0 805b694c 805b66a4
[ 1045.725111] dee0: 7f2b4014 7f2b4080 b85ddf0c b85ddef8 805b5478 805b68ec 7f2b4014 7f2b4080
[ 1045.736261] df00: b85ddf24 b85ddf10 805b6e30 805b5428 80d04d48 7f2b4080 b85ddf34 b85ddf28
[ 1045.747423] df20: 805b7fc8 805b6e04 b85ddf44 b85ddf38 7f2b2b88 805b7fb8 b85ddfa4 b85ddf48
[ 1045.758658] df40: 801b77b4 7f2b2b7c 735f6376 6d635f6d 00000061 801011c4 b85dc000 00000006
[ 1045.770033] df60: b85ddf8c 80c85d98 80101068 00000001 00000006 801011c4 b85dc000 00000006
[ 1045.781514] df80: 005ddfac 89fa4d05 80204ea0 004e1bf0 004e1c2c 004e0ff0 00000000 b85ddfa8
[ 1045.793113] dfa0: 80101000 801b7620 004e1bf0 004e1c2c 004e1c2c 00000800 6c0b1b00 0003f06c
[ 1045.804796] dfc0: 004e1bf0 004e1c2c 004e0ff0 00000081 004e0f9c 004e0f60 004e0008 7e8af8f0
[ 1045.816542] dfe0: 0003ef9c 7e8ae3e4 00022020 76f4462c 20000010 004e1c2c 00000000 00000000
[ 1045.828362] [<805a6558>] (misc_deregister) from [<7f2b1644>] (bcm2835_vc_sm_cma_remove+0x34/0x5c [vc_sm_cma])
[ 1045.841976] [<7f2b1644>] (bcm2835_vc_sm_cma_remove [vc_sm_cma]) from [<805b7ee8>] (platform_drv_remove+0x34/0x4c)
[ 1045.855917] [<805b7ee8>] (platform_drv_remove) from [<805b681c>] (device_release_driver_internal+0x184/0x224)
[ 1045.869527] [<805b681c>] (device_release_driver_internal) from [<805b694c>] (driver_detach+0x6c/0xa0)
[ 1045.882441] [<805b694c>] (driver_detach) from [<805b5478>] (bus_remove_driver+0x5c/0xb4)
[ 1045.894219] [<805b5478>] (bus_remove_driver) from [<805b6e30>] (driver_unregister+0x38/0x58)
[ 1045.906326] [<805b6e30>] (driver_unregister) from [<805b7fc8>] (platform_driver_unregister+0x1c/0x20)
[ 1045.919259] [<805b7fc8>] (platform_driver_unregister) from [<7f2b2b88>] (bcm2835_vcsm_cma_driver_exit+0x18/0x490 [vc_sm_cma])
[ 1045.934309] [<7f2b2b88>] (bcm2835_vcsm_cma_driver_exit [vc_sm_cma]) from [<801b77b4>] (sys_delete_module+0x1a0/0x1f0)
[ 1045.948651] [<801b77b4>] (sys_delete_module) from [<80101000>] (ret_fast_syscall+0x0/0x28)
[ 1045.960622] Exception stack(0xb85ddfa8 to 0xb85ddff0)
[ 1045.967517] dfa0:                   004e1bf0 004e1c2c 004e1c2c 00000800 6c0b1b00 0003f06c
[ 1045.979330] dfc0: 004e1bf0 004e1c2c 004e0ff0 00000081 004e0f9c 004e0f60 004e0008 7e8af8f0
[ 1045.991107] dfe0: 0003ef9c 7e8ae3e4 00022020 76f4462c
[ 1045.997959] Code: e3046a74 e34860e4 e3a0ec01 e3a0cc02 (e5823004) 
[ 1046.010598] ---[ end trace 6b7f3fc94c92fea8 ]---
```

Under no vc_sm allocation config (eg. `gpu_mem=16`), such oops does not happen.

6by9, 2019-03-07:
I didn't know that modprobe -r unloaded dependent modules too. I've always used rmmod.

There is a bug in bcm2835_vc_sm_cma_remove as it isn't calling misc_register during probe, so the mismatched misc_deregister will cause issue. If you've set gpu_mem=16 and are therefore using the cutdown firmware then the service never connects and sm_inited will never be set.
I'll sort a patch tomorrow.

6by9, 2019-03-08:
Fix merged. Thanks for the report.

jjsuwa, 2019-03-08:
LGTM. Thank you for your quick response.

another regression info:
- after upgrading 4.14.y to 4.19.y, group of both `/sys/class/i2c-dev` and `spidev` (and their children) remains `root` (formerly `i2c` and `spi`).
 (requires to add the rules similar to `/sys/class/gpio`, to `/etc/udev/rules.d/99-com.rules`?)

6by9, 2019-03-08:
One problem per issue please.

closed 2019-03-08 by jjsuwa