raspberrypi/linux #2886: Technisat Skystar USB HD SNR measurement problem after upgrading to kernel 4.14.xx

Xf0ur, 2019-03-07:
_Hello,
I was using kernel v4.9.80 ... my Technisat Skystar USB HD used to work fine.
However, after i upgraded to kernel v4.14.xx i faced SNR measurement problem as you can see in the below screenshots.
1st screenshot: kernel v4.9.80
2nd screenshot: kernel v4.14.xx
Any help would be very appreciated._

[image]

[image]

Xf0ur, 2019-03-21:
Is there someone who owns this card in order to verify the problem?

6by9, 2019-03-21:
Almost all the media drivers are developed by the mainline kernel developers (https://linuxtv.org/), not Raspberry Pi. Whilst individuals may have access to some hardware, it is a long way from comprehensive coverage.

You could try building the upstream latest drivers, and if the problem still exists there then report it to the linux-media mailing list. Instructions on building media_tree are at https://www.linuxtv.org/wiki/index.php/How_to_Obtain,_Build_and_Install_V4L-DVB_Device_Drivers

As a simpler, half step, you could use ```sudo rpi-update``` which will pick up the testing 4.19 kernel. There are a couple of issues still lurking in it, so please be slightly cautious.

PLEASE do not do either steps on a critical SD card. Use a backup as recovering it if things go really wrong is likely to be painful.

Xf0ur, 2019-03-21:
Thanks for replying,
The last time i used rpi-update tool, my kernel was upgraded to v4.14.90.
After i faced that SNR measurement problem i kept downgrading to some older 4.14.xx kernels.
I realized the problem started to occur in kernel v4.14.18.

Therefore i decided to stick with kernel v4.9.80. 

I have doubt that upgrading to kernel v4.19.xx won't solve that issue because it seems that someone had modified something in the driver code starting from kernel v4.14.18 that needs to be reverted.

I will try your 1st suggestion though.

What if i reported the problem here?
https://github.com/Hexxeh/rpi-update

Will they have the same response as yours?

JamesH65, 2019-03-21:
This is the correct place to be reporting issues with the Raspbery Pi kernel, which includes drivers.

Xf0ur, 2019-03-21:
Okay, thanks.
The following patch was applied in kernel v4.14.xx.
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/media/dvb-frontends/stv090x.c?h=linux-4.14.y&id=386701892b4821c4d127710fa35b1309c4355bbf
Is it causing the problem?
I mean the patch was intended to fix something and what i have is just a side effect.

P33M, 2019-03-22:
Why is it an issue? You get a number returned from the driver that represents SNR. This number is calculated differently between 4.14 and 4.9 - but the actual signal is unchanged.

Arguably the 4.14 method is more correct as it translates "number from the hardware" into "number representing decibels SNR" more accurately.

Xf0ur, 2019-03-22:
I have Ubuntu 16.04, whether i use kernel  4.13 or 4.15 SNR  remains the same(sometimes slightly different).

But now all channels got SNR between 20-26% no matter what, i'm aware it should be displayed in decibles though.

As i wrote, something were modified starting from kernel 4.14.18 which caused that issue.

I will report it to linux-media mailing list and see if i will get any response.