raspberrypi/linux #2891: Full KMS not working @4.19.x

ezar, 2019-03-11:
**Describe the bug**
Using Full KMS, raspbi doesn't start.

**To reproduce**
dtoverlay=vc4-kms-v3d @ config.txt

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2018-06-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 0d34fb94398a4bd79711f21d6a0b5d35fe5f4a76, stage5
* Which firmware version (`vcgencmd version`)?
Mar  6 2019 14:47:17 
Copyright (c) 2012 Broadcom
version b403ee6ed819f9ac7a96834ac437b6cfdd4512ad (clean) (release) (start_x)
* Which kernel version (`uname -a`)?
Linux magicmirror 4.19.27-v7+ #1206 SMP Wed Mar 6 14:40:18 GMT 2019 armv7l GNU/Linux

**Logs**
No logs. Raspi doesnt boot :(

6by9, 2019-03-11:
Have you enabled it through raspi-config, or just altered config.txt? IIRC there are a couple of other parameters that get set up via raspi-config.

You've obviously used rpi-update to get the 4.19 kernel. Have you tried any other 4.19 kernels?
```sudo rpi-update 82f49d8``` would get the previous release.

4.19.27 does work with legacy, vc4-fkms-v3d, and vc4-kms-v3d, so something odd is going on with your system.

ezar, 2019-03-12:
Yes, I use raspi-config.
I try with previous 4.19.x version with same result.
Do you need my config.txt?
Regards

6by9, 2019-03-12:
If you've made your own modifications, then yes having your config.txt would be useful.

So this is a vanilla Raspbian download, updated via ```apt update``` and ```apt upgrade```, and then using Full KMS driver? Nothing else special? I'll try to reproduce today.

ezar, 2019-03-12:
Nothing special. Only I need to rotate screen.

config.txt:
> # For more options and information see
> # http://rpf.io/configtxt
> # Some settings may impact device functionality. See link above for details
> 
> # uncomment if you get no picture on HDMI for a default "safe" mode
> #hdmi_safe=1
> 
> # uncomment this if your display has a black border of unused pixels visible
> # and your display can output without overscan
> #disable_overscan=1
> 
> # uncomment the following to adjust overscan. Use positive numbers if console
> # goes off screen, and negative if there is too much border
> #overscan_left=16
> #overscan_right=16
> #overscan_top=16
> #overscan_bottom=16
> 
> # uncomment to force a console size. By default it will be display's size minus
> # overscan.
> #framebuffer_width=1280
> #framebuffer_height=720
> 
> # uncomment if hdmi display is not detected and composite is being output
> hdmi_force_hotplug=1
> 
> # uncomment to force a specific HDMI mode (this will force VGA)
> hdmi_group=2
> hdmi_mode=82
> 
> # uncomment to force a HDMI mode rather than DVI. This can make audio work in
> # DMT (computer monitor) modes
> hdmi_drive=1
> 
> # uncomment to increase signal to HDMI, if you have interference, blanking, or
> # no display
> #config_hdmi_boost=4
> 
> # uncomment for composite PAL
> #sdtv_mode=2
> 
> #uncomment to overclock the arm. 700 MHz is the default.
> #arm_freq=800
> 
> # Uncomment some or all of these to enable the optional hardware interfaces
> dtparam=i2c_arm=on,i2c_arm_baudrate=10000
> #dtparam=i2s=on
> #dtparam=spi=on
> 
> # Uncomment this to enable the lirc-rpi module
> #dtoverlay=lirc-rpi
> 
> # Additional overlays and parameters are documented /boot/overlays/README
> 
> # Enable audio (loads snd_bcm2835)
> dtparam=audio=on
> 
> # OpenGL
> #dtoverlay=vc4-kms-v3d
> dtoverlay=vc4-fkms-v3d
> 
> # Magic Mirror
> display_rotate=3
> avoid_warnings=1
> start_x=1
> gpu_mem=128

6by9, 2019-03-12:
Your title says "Full KMS not working", but your config.txt says ```vc4-fkms-v3d``` which is the fake or firmware KMS driver. Please confirm that you really mean fkms isn't working for you.

I was about to post that none of
```
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=82
hdmi_drive=1
display_rotate=3
```
will do anything in full KMS mode.

pelwell, 2019-03-12:
As Orson Welles (and Faith No More) would say, it's "F for Fake".

6by9, 2019-03-12:
Tested with  November 2018 Raspbian image, apt update, apt upgrade, sudo rpi-update
Using sudo raspi-config to enable either Full or Fake KMS drivers works fine with default config. 

It appears that ```display_rotate=3``` does upset Full KMS (vc4-kms-v3d), but as I noted earlier it is meaningless in the full KMS world. I'm therefore reluctant to spend much effort investigating why.
```xrandr --output HDMI-1 --rotate left``` is the command to amend the rotation in KMS, or stuff it in .bashrc if you want it permanently.

All your other hdmi options in config.txt are harmlessly ignored under full KMS.

6by9, 2019-03-12:
Actually I do know why it is failing.
```display_rotate=[1|3]``` requires the transposer, however because you've requested the full KMS driver the firmware will never touch it as the ARM "owns" it. Boot stalls waiting for a screen update to complete, which will never happen becuase it requires the unavailable tranposer.

Low priority to fix as there is an obvious workaround, but it probably ought to get done to avoid others falling into the same hole.

ezar, 2019-03-12:
> Your title says "Full KMS not working", but your config.txt says `vc4-fkms-v3d` which is the fake or firmware KMS driver. Please confirm that you really mean fkms isn't working for you.
>
> I was about [...]

Sorry about that. I'm using fake KMS because full fails. I correct the config file for testing purposes.

6by9, 2019-03-12:
> Sorry about that. I'm using fake KMS because full fails. I correct the config file for testing purposes.

No problem, it was just confusing.
The issue is now understood, and having said it's a low priority it's a two line fix (although I then also lose the rainbow screen).

ezar, 2019-03-12:
Ok. I will remove "display_rotate=3" :)

6by9, 2019-03-12:
Patch pushed internally so that "display_rotate=3" shouldn't stall boot with KMS.

Loss of the rainbow screen may just be one of those things as it isn't that easy to fix, and the configuration options make no sense.
It does have a minor impact on legacy and fkms systems in that it is only when the Linux frame buffer kicks the transposer function does the rainbow screen flash up several seconds into the kernel boot, and is then almost immediately replaced by the frame buffer. If there is an easy fix then I'll add it.

ezar, 2019-03-12:
I try it after eliminate display_rotate and works as expected.
The only problem is when I try to rotate screen using xrandr fails!!! Say “Can’t open display”.

pelwell, 2019-03-12:
`xrandr` needs the name of a display to control - see the [man page](https://www.x.org/releases/X11R7.5/doc/man/man1/xrandr.1.html). If you run it from a terminal opened on the desktop, the `DISPLAY` environment variable should be set to `:0.0`, and the command should work without an explicit display. If not, try adding `-display :0.0`.

ezar, 2019-03-12:
@pelwell works fantastic!
If I want to apply when rpi start?

pelwell, 2019-03-12:
I was going to suggest adding it to /etc/rc.local, but it could end up being started before the X server. [This page](https://developer.toradex.com/knowledge-base/how-to-autorun-application-at-the-start-up-in-linux) suggests putting it to ~/.config/autostart/<something>.desktop, or /etc/xdg/autostart/<something>.desktop, but I've not tried either.

vanfanel, 2019-04-30:
Full KMS is not working on 4.14.117 either. I have just built the latest kernel on the 4.14.y branch and I get a black screen unless I disable the vc4-kms-v3d overlay.

popcornmix, 2019-04-30:
What was the last version that did work @vanfanel ?

vanfanel, 2019-04-30:
@popcornmix : I am not building the kernel very often, but 4.14.72-v8+ (which was the version I was using previously) seems to work perfectly find with the vc4-kms-v3d.

6by9, 2019-05-01:
Are we happy to close this one?
ezar's issue with ```display_rotate=3``` is fixed, and is an invalid configuration anyway.
vanfanel's issue was user error in installing the modules in the wrong place. (It looks like it's on an aarch64 build as well based on 4.14.72-v8+)

Full KMS should be working fine on both 4.14 and 4.19.

closed 2019-05-01 by popcornmix

leragequit, 2020-06-26:
Hello there. 
For me Full KMS is still not working when using the Buster Lite Image and Kernel 
`Linux raspberrypi 5.4.47-v7+ #1322 SMP Wed Jun 17 17:52:39 BST 2020 armv7l GNU/Linux
`
The screen just stays dark, ssh works though.

It would be really nice if this could be fixed in the near future, as I am experiencing issues with the Fake KMS driver and QT.