raspberrypi/linux #2904: chromium crashes after today update

jmeile, 2019-03-22:
Hi

My Chromium kiosk was working before today's update. It was setup four days ago. The following packages were installed:
chromium-browser:armhf 65.0.3325.181-0+rpt4 72.0.3626.121-0+rpt3
chromium-codecs-ffmpeg-extra:armhf 65.0.3325.181-0+rpt4 72.0.3626.121-0+rpt3

I'm sure about this because minutes before the update, everything was working perfectly.

I have an nginx local web server, which loads a local php file. There I will rotate between different pages. The first page is normal html with a picture slider. After five minutes, a second page will be shown. There a html5 video should be played; however, instead of this, the following page will be shown:

""""
Aw, Snap!
Something went wrong while displaying this webpage.
"""

There is no error on the nginx log file nor the syslog. I was doing:
tail -f /var/log/nginx/error_log
tail -f /var/log/syslog

when testing and nothing was logged.

Here what I have:
Hardware: Raspberry Pi3B+

OS:
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2

Firmware:
Feb 12 2019 19:42:42
Copyright (c) 2012 Broadcom
version 8eff5e4023657a8b3b59e1f90dc966f62d74908c (clean) (release) (start)

Kernel:
Linux iwf-vitrine 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

Any way of reverting to the older version?

Here is what I had before according to: /var/log/apt/history.log
chromium-browser:armhf (65.0.3325.181-0+rpt4)
chromium-codecs-ffmpeg-extra:armhf (65.0.3325.181-0+rpt4, automatic)

Todays update looks like:
chromium-browser:armhf (65.0.3325.181-0+rpt4, 72.0.3626.121-0+rpt3)
chromium-codecs-ffmpeg-extra:armhf (65.0.3325.181-0+rpt4, 72.0.3626.121-0+rpt3)

I tried
apt-get install chromium-browser:armhf=65.0.3325.181-0+rpt4
apt-get install chromium-codecs-ffmpeg-extra:armhf=65.0.3325.181-0+rpt4

But the only thing I get is:
E: Version '65.0.3325.181-0+rpt4' for 'chromium-browser' was not found

Best regards
Josef

popcornmix, 2019-03-22:
This doesn't sound like a linux kernel issue. Try reporting here:
https://github.com/RPi-Distro/repo/issues

closed 2019-03-22 by popcornmix

jmeile, 2019-03-22:
Ok, thanks for your answer. I moved it here:
https://github.com/RPi-Distro/repo/issues/131