raspberrypi/linux #2905: xt_socket module no longer enabled in stretch

lcolitti, 2019-03-23:
I updated one of my 3b units to stretch, and my transparent proxy stopped working because the xt_socket iptables module is no longer enabled in the kernel configuration. This used to work on jessie (see below).

Could you set CONFIG_NETFILTER_XT_MATCH_SOCKET=m again like we had in 4.9? It's only 10k of code.

============
On jessie:

$ dmesg | grep Machine
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
$ uname -sr
Linux 4.9.35-v7+
$ zcat /proc/config.gz | grep MATCH_SOCKET
CONFIG_NETFILTER_XT_MATCH_SOCKET=m
$ ls -l /lib/modules/`uname -r`/kernel/net/netfilter/xt_socket.ko
-rw-r--r-- 1 root root 10348 Jul  3  2017 /lib/modules/4.9.35-v7+/kernel/net/netfilter/xt_socket.ko

On stretch:

$ dmesg | grep Machine
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
$ uname -sr
Linux 4.14.98-v7+
$ zcat /proc/config.gz | grep MATCH_SOCKET
$ ls -l /lib/modules/`uname -r`/kernel/net/netfilter/xt_socket.ko
ls: cannot access '/lib/modules/4.14.98-v7+/kernel/net/netfilter/xt_socket.ko': No such file or directory

pelwell, 2019-03-24:
Our defconfigs were broken by the 4.10 kernel with introduced a new dependency for CONFIG_NETFILTER_XT_MATCH_SOCKET on NF_SOCKET_IPV4 and NF_SOCKET_IPV6. Because those new dependencies weren't enabled the option got dropped. Then, in 4.18, the `depends on` was changed to a `select`, but by then the damage was done.

I'm happy to re-enable CONFIG_NETFILTER_XT_MATCH_SOCKET. The new `select`s do mean that your calculation of the code size should really include the extra 12kB of the nf_socket_ipv4 and nf_socket_ipv6 modules, but if the relevant Kconfig had moved straight to the 4.18 version from the 4.9 version we wouldn't have noticed the appearance of the extra modules so we can't object.

See https://github.com/raspberrypi/linux/commit/0db9356879bf923684d5397feb99171d3ee62e87.

lcolitti, 2019-03-28:
Thanks for quick turnaround! Would I expect to see 4.14 pick up this change too? The commits here all seem for newer kernels.

pelwell, 2019-03-28:
Not without a good reason. Backporting to rpi-4.14.y is simple enough, but binary releases are unlikely/rare.

lcolitti, 2019-05-26:
My 3b just got upgraded to 4.19.42-v7+ and the code works again. Thanks!

pelwell, 2019-05-26:
Please close the issue when you are satisfied.

closed 2019-05-27 by lcolitti