raspberrypi/linux #2906: module request: 'max17040_battary'

Fat-Zer, 2019-03-23:
Could you please include the `max17040_battary` module (config option `CONFIG_BATTERY_MAX17040`) to the default rpi-config (and as a result to the default kernel for raspbian). The chip is found at least on [this][1] UPS board.

Also it would be nice to have an associated device tree overlay. The chip utilizes i2c bus and has hard-wired address `0x36`, the datasheet can be found [here][2]. I may assist with testing/writing it, but I haven't found the original `*.dts` repository for the raspbery pi `*.dtbo`'s.

  [1]:https://github.com/linshuqin329/UPS-18650
  [2]:https://datasheets.maximintegrated.com/en/ds/MAX17040-MAX17041.pdf

pelwell, 2019-03-23:
Look at `arch/arm/boot/dts/overlays` in this repo. We'll consider the module request.

Fat-Zer, 2019-03-24:
Thanks, for the directions. The suggested patch, if you wondering (I grant it to the public domain or whatever if anybody wondering):

```
diff --git a/arch/arm/boot/dts/overlays/i2c-sensor-overlay.dts b/arch/arm/boot/dts/overlays/i2c-sensor-overlay.dts
index 4b5be8676d11..0e2efa6ac8ea 100644
--- a/arch/arm/boot/dts/overlays/i2c-sensor-overlay.dts
+++ b/arch/arm/boot/dts/overlays/i2c-sensor-overlay.dts
@@ -201,6 +201,21 @@
                };
        };

+       fragment@13 {
+               target = <&i2c_arm>;
+               __dormant__ {
+                       #address-cells = <1>;
+                       #size-cells = <0>;
+                       status = "okay";
+
+                       max17040: max17040@36 {
+                               compatible = "maxim,max17040";
+                               reg = <0x36>;
+                               status = "okay";
+                       };
+               };
+       };
+
        __overrides__ {
                addr =  <&bme280>,"reg:0", <&bmp280>,"reg:0", <&tmp102>,"reg:0",
                        <&lm75>,"reg:0", <&hdc100x>,"reg:0", <&sht3x>,"reg:0",
@@ -219,5 +234,6 @@
                veml6070 = <0>,"+10";
                sht3x = <0>,"+11";
                ds1621 = <0>,"+12";
+               max17040 = <0>,"+13";
        };
 };
```

PS: don't make a typo in your configs... just spent half a day wondering why it doesn't work because wrote "_i2c-sensor**s**_" for name of dt_overlay >_< (just wanted to share it).

pelwell, 2019-03-25:
Added to rpi-4.19.y. Please test, then close if happy.

Fat-Zer, 2019-03-25:
Thanks! The commits looks fine. I'll actually test it with next respbian kernel package update if you never mind.

JamesH65, 2019-09-19:
@Fat-Zer Can this issue be closed?

andrew-stclair, 2019-12-11:
is this feature implemented with `dtoverlay=i2c-sensor,max17040`?

if it is, I'm having issues using it, the Battery (pi-top / laptop) panel add-on for raspbian will always say Charging 0% irrespective of what is actually going on

The following python script does successfully return the voltage and charge percentage

```python
#!/usr/bin/python
import struct
import smbus
import sys
import time

def readVoltage(bus):
        "This function returns as float the voltage from the Raspi UPS Hat via the provided SMBus object"
        address = 0x36
        read = bus.read_word_data(address, 2)
        swapped = struct.unpack("<H", struct.pack(">H", read))[0]
        voltage = swapped * 1.25 /1000/16
        return voltage

def readCapacity(bus):
        "This function returns as a float the remaining capacity of the battery connected to the Raspi UPS Hat via the provided SMBus object"
        address = 0x36
        read = bus.read_word_data(address, 4)
        swapped = struct.unpack("<H", struct.pack(">H", read))[0]
        capacity = swapped/256
        return capacity

bus = smbus.SMBus(1)  # 0 = /dev/i2c-0 (port I2C0), 1 = /dev/i2c-1 (port I2C1)

print "++++++++++++++"
print "Voltage:%5.2fV" % readVoltage(bus)
print "Battery:%5i%%" % readCapacity(bus)
if readCapacity(bus) == 100:
  print "Battery:  FULL"

if readCapacity(bus) < 20:
  print "Battery:   LOW"

print "++++++++++++++"
```

I'm using the `UPS 17650` developed by @linshuqin329

Fat-Zer, 2020-02-21:
@andrew-stclair, Hi, yes, the kernel interface should be enbaled by `dtoverlay=i2c-sensor,max17040` in boot config. The chip supposed to correctly display only charge persentage, and a scaled value for voltage. 

I can't say why the applet doesn't work for you. If the applet utilises upower it should correctly display at least charge percentage. The state should likely be 'discharging' or 'unknown' (as its reported by the kernel, but it seems upower doesn't actually support that state). The voltage is received directly from the chip, so it won't be quite correct as well (it should be scaled by  `1.25/16*1000`).

You should check what it actually displays under the hood with next commands:

* for kernel interface: `grep -r . /sys/class/power_supply/battery`

        /sys/class/power_supply/battery/online:1
        /sys/class/power_supply/battery/capacity:101
        /sys/class/power_supply/battery/power/runtime_suspended_time:0
        /sys/class/power_supply/battery/power/runtime_active_time:0
        /sys/class/power_supply/battery/power/control:auto
        /sys/class/power_supply/battery/power/runtime_status:unsupported
        /sys/class/power_supply/battery/type:Battery
        /sys/class/power_supply/battery/status:Unknown
        /sys/class/power_supply/battery/voltage_now:53744
        /sys/class/power_supply/battery/uevent:POWER_SUPPLY_NAME=battery
        /sys/class/power_supply/battery/uevent:POWER_SUPPLY_STATUS=Unknown
        /sys/class/power_supply/battery/uevent:POWER_SUPPLY_ONLINE=1
        /sys/class/power_supply/battery/uevent:POWER_SUPPLY_VOLTAGE_NOW=53744
        /sys/class/power_supply/battery/uevent:POWER_SUPPLY_CAPACITY=101

* for upower interface: `upower -e`

        /org/freedesktop/UPower/devices/battery_battery
        /org/freedesktop/UPower/devices/DisplayDevice

    and then `upower -i /org/freedesktop/UPower/devices/battery_battery`

          native-path:          battery
          power supply:         yes
          updated:              Fri Feb 21 21:15:06 2020 (72 seconds ago)
          has history:          yes
          has statistics:       yes
          battery
            present:             yes
            rechargeable:        yes
            state:               discharging
            warning-level:       none
            energy:              0 Wh
            energy-empty:        0 Wh
            energy-full:         0 Wh
            energy-full-design:  0 Wh
            energy-rate:         0 W
            voltage:             0.053728 V
            percentage:          100%
            capacity:            100%
            icon-name:          'battery-full-symbolic'

Fat-Zer, 2020-02-21:
@pelwell, sorry I haven't replied for a while. Everything works fine for me, so I'm closing the bug. If there are problems with modules, I don't think there is much more can be done on the raspberry's kernel side.

It would be nice to integrate some pre-scaler for voltage, but it's a task for the mainline kernel IMHO.

closed 2020-02-21 by Fat-Zer