raspberrypi/linux #2907: USB Broken for CM3 post 4.14.18 through 4.19.y

grimepoch, 2019-03-23:
There is some major problem with the USB driver starting with the fix that went into 4.14.20.  I have verified that the problem comes from the changes of these two files:

drivers/usb/host/dwc_otg/dwc_otg_hcd.c
drivers/usb/host/dwc_otg/dwc_otg_hcd_linux.c

I replaced the files in 4.14.20 with those 2 files in 4.14.18 and the USB began working properly.

I then tested replacing those files in 4.14.59 that I am using on an embedded project, and again, the USB began working properly.

Specifically the problem is on hot swapping.  With the changes introduced in 4.14.20, if I removed a USB device while the system was running, there was a good chance of the whole system freezing.  (Nothing was written out to logs either).

Using a hub, the problem would haven less often, but still quite a bit.

The biggest problem I had, if I removed a WiFi USB adapter, it would NEVER recover from that.  With this roll back to the older code, if I reattach the WiFi adapter, it will reconnect and everything.

I am not sure the implications of what changed in the driver, it's possible it fixed a different problem and introduced this new one, but after a day of testing, the USB works much better with the older source on those two files.

4.19 has the same problem.  I did not test using the older code with 4.19 as we are not ready to move up to that yet, but just wanted to identify the problem still exists.

Hardware: CM3

6by9, 2019-03-25:
https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=230161

Specfically what Wifi adapter/devices are you using? ```lsusb -vvv```.

grimepoch, 2019-03-25:
What I have observed is it isn't only the WiFi adapter, if I have a MIDI device attached and connected, and I have it open, if I remove the device, with 4.14.20, it will lock the whole system as well (completely unresponsive pi).

If I use a small hub like what is listed below, the WiFi remove will almost always lock it.  The MIDI becomes more sporadic, but it still happens.

The way I was testing the MIDI removal, any MIDI device attached, then doing
`while true; do cat /dev/snd/midiC0D0; done`

In the console.  When you remove the MIDI device, it just loops saying device not found, when you insert one, it will stop and echo any messages you send with the MIDI device.

On 4.14.18, I can add and remove MIDI devices, never locks up the system.

On 4.14.20, it will lock up.

`Bus 001 Device 003: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x148f Ralink Technology, Corp.
  idProduct          0x5370 RT5370 Wireless Adapter
  bcdDevice            1.01
  iManufacturer           1 Ralink
  iProduct                2 802.11 n WLAN
  iSerial                 3 1.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           67
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              450mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           7
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              5 1.0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
[... 175 lines cut ...]
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered`

6by9, 2019-03-27:
What board are you plugging the CM3 in to? The standard CMIO or a custom one?

@P33M any thoughts?

pelwell, 2019-03-27:
It looks like commit 62bb67a2cc5d ("dwc_otg: Fix a regression when dequeueing isochronous transfers") is the cause. I would guess there is a path followed during device disconnection that leaves the fiq_state lock held.

pelwell, 2019-03-27:
Or the FIQ_HS_ISOC_ABORTED handling code (which won't have been used before) needs to actually do something (e.g. `handled = 1;`).

grimepoch, 2019-03-27:
This is a custom board that I designed.  In the USB path I've included a USB current limiter as well to prevent any of the issues that can happen with USB cables and power/gnd. I also have 22ohm resistors inline with Dn/Dp and I have a diode back for protection across the USB signals (again, it's designed specifically for USB).

I made the USB robust because of reports of sensitivity on the Dn/Dp lines I've read in other posts, so I wanted to make sure that wasn't going to be an issue for us.

Last night, I took the 4.14.59-v7+ that I've been using for the last year and I completely replaced the dwc_otg subdirectory with the same contents of 4.14.18, rebuilt everything, and installed.

I tested this last night by inserting and removing both WiFi and MIDI devices and the system was solid, no hangs.  

So those changes in 4.14.20 definitely cause a problem.  If someone puts in some fixes for me to test, I can build up a 4.19.y variant just to test the USB issues.

popcornmix, 2019-04-01:
@pelwell's suggestion was 62bb67a was probably responsible.
If you are building your own kernel then building without that and confirming the issue is gone would be use. e.g.
```git revert 62bb67a```

grimepoch, 2019-04-01:
Just to be clear, when I installed the precompiled kernel 4.14.18, USB worked without ever locking up the system adding/removing devices.  When I installed precompiled binary 4.14.20, USB would lock up the system when removing Wireless or MIDI devices that were being used (in other words, they were open).

A USB keyboard NEVER locked the system adding/removing.

I have not run with reverting that change, I'll try that this week.

I noticed there appear to be other USB changes that happened since 4.14.20 as well when I was looking up the 4.19.y tree, but I did install the 4.19.y binary (I forget which version) and the problem still existed.

P33M, 2019-04-08:
I can reproduce: do `fswebcam -S 10000` on a Pi3A+ with a webcam and disconnect it. System wedges about 50% of the time.

P33M, 2019-04-08:
> Or the FIQ_HS_ISOC_ABORTED handling code (which won't have been used before) needs to actually do something (e.g. `handled = 1;`).

The "handled" flag means that there's no state for the upstream driver to process, thus no interrupts are propagated to the HCD. It's safe to not handle a host channel interrupt, it is instead masked and passed to the HCD via triggering the fake interrupt.

If I look at the register set in the wedged state, then it looks like I have a stuck SOF interrupt along with pending-but-masked host port disconnect and OTG interrupts. The host all-channels interrupt is 0.

P33M, 2019-04-23:
Please retest after running rpi-update and close if fixed.

grimepoch, 2019-04-24:
I will get this tested next week when I am back in the lab, I will update accordingly, thanks for the work P33M!

Jmennius, 2019-05-21:
@grimepoch any luck with testing?
I guess this issue is one of the blockers for 4.19?

JamesH65, 2019-05-21:
@Jmennius We believe this is fixed, not a blocker. I suggest trying it yourself to confirm.

Jmennius, 2019-05-21:
@JamesH65 Ok, I didn't experience this issue, just wondering about 4.19 status mostly.
I was suspecting USB/Eth issue (which made me look into kernel updates and USB/Eth issues) but it turned out to be unrelated in my case.

Thank you!

JamesH65, 2019-09-19:
@grimepoch Can this be closed?

P33M, 2021-03-15:
OP never reported back, and a positively identified bug was fixed. Closing.

closed 2021-03-15 by P33M