raspberrypi/linux #2917: USB touch delay when connected directly to CM3+

Lauszus, 2019-04-02:
I've got a strange issue, as there is a noticeable delay when I connect a USB touch screen directly to the USB port on a CM3+ module and the official IO board:

<https://www.youtube.com/watch?v=kl8zyjY_Czw>

However the delay disappears when the USB touch screen is connected via a FS USB hub:

<https://www.youtube.com/watch?v=bKs07fV0CN0>

I'm running the following firmware and kernel versions:

```bash
version ed5baf9520a3c4ca82ba38594b898f0c0446da66 (clean) (release)

Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
```

I've tried upgrading using ```rpi-update``` however the issue is still present. I have confirmed the same issue on a CM3.

dmesg does not report any issues.

<details><summary>lsusb for touch screen:</summary>

```bash
Bus 001 Device 005: ID 0eef:c000 D-WAV Scientific Co., Ltd
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0eef D-WAV Scientific Co., Ltd
  idProduct          0xc000
  bcdDevice           83.00
  iManufacturer           1 eGalax Inc.
  iProduct                2 eGalaxTouch EXC3110-3883-08.00.00
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          1 eGalax Inc.
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     494
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
````

</details>

<details><summary>lsusb for usb hub:</summary>

```bash
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1a40 Terminus Technology Inc.
  idProduct          0x0101 Hub
  bcdDevice            1.11
  iManufacturer           0
  iProduct                1 USB 2.0 Hub
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           25
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             4
  wHubCharacteristic 0x0000
    Ganged power switching
    Ganged overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent    100 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
   Port 2: 0000.0100 power
   Port 3: 0000.0503 highspeed power enable connect
   Port 4: 0000.0103 power enable connect
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         0 Full speed (or root) hub
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0001
  Self Powered
```

</details>

I've tried various combinations of the dwc_otg flags: 

```bash
dwc_otg.speed
dwc_otg.fiq_fix_enable
dwc_otg.nak_holdoff
dwc_otg.nak_holdoff_enable
dwc_otg.fiq_fsm_mask
dwc_otg.fiq_fsm_enable
```

The only thing I learned was that I was able to replicate the issue on the hub if I set ```dwc_otg.speed=1```.

Let me know if you want any more details.

Lauszus, 2019-04-02:
Here's some Python code that can be used to verify the delay:

```python
from mtdev import Device

device = Device('/dev/input/event0')
while True:
    while device.idle(1000):
        continue
    data = device.get()
    if data is None:
        continue
    print( data.value)
```

The mtdev module can be downloaded from the Kivy project: <https://github.com/kivy/kivy/blob/master/kivy/lib/mtdev.py>.

When running the code the delay is quite noticeable.

It can also verified by using ```mtdev-test```:

```bash
sudo mtdev-test /dev/input/event0
```

Note that the following package has to be installed:

```bash
sudo apt-get install libmtdev1 mtdev-tools
```

Lauszus, 2019-04-02:
Hmm maybe the kernel has issue with reading the touch screen at a bInterval of 1 ms when it is connected directly?

Lauszus, 2019-04-02:
Here's a minimal C example as well:

```C
#include <stdio.h>
#include <linux/input.h>
#include <signal.h>
#include <fcntl.h>
#include <unistd.h>

volatile sig_atomic_t run = 1;

void sigint(int sig){
    run = 0;
}

int main() {
    signal(SIGINT, sigint);
    int fileno = open("/dev/input/event0", /*O_NONBLOCK |*/ O_RDONLY);
    struct input_event ev;
    while (run) {
        int num_bytes = read(fileno, &ev, sizeof(struct input_event));
        if (num_bytes != sizeof(struct input_event)) {
            printf("Failed to read device\n");
            return 1;
        }
        printf("%d, %d, %d, %d\n", ev.time, ev.type, ev.code, ev.value);
    }
    printf("Closing device\n");
    close(fileno);
    return 0;
}
```

Lauszus, 2019-04-02:
I found a workaround!

I works as expected by using the dwc2 USB controller driver overlay: <https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/dwc2-overlay.dts>.

Lauszus, 2019-04-03:
I got it working with the dwc_otg driver by disabling the microframe scheduling i.e. ```dwc_otg.microframe_schedule=0```. I suspect that the driver might not deal with full speed devices connected directly properly.

It should be noted that the same issues is present if a FS hub is used.

I found a potential cause here: 

https://github.com/raspberrypi/linux/blob/e2d2941326922b63d722ebc46520c3a2287b675f/drivers/usb/host/dwc_otg/dwc_otg_hcd_queue.c#L194-L203

as far I understand it should check if the hub is a high speed hub as well i.e. ```hprt.b.prtspd == DWC_HPRT0_PRTSPD_HIGH_SPEED```. This is also what is done in the dwc2 driver:

https://github.com/raspberrypi/linux/blob/e2d2941326922b63d722ebc46520c3a2287b675f/drivers/usb/dwc2/hcd_queue.c#L1518-L1519

JamesH65, 2019-04-03:
@P33M  Any thoughts?

P33M, 2019-04-03:
What's the output of lsusb -t in the failing case?

Lauszus, 2019-04-03:
lsusb for setup with FS hub:

```bash
$ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 12M
        |__ Port 1: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
```

I.e.  the setup that does NOT work with microframe scheduling enabled.

lsusb for setup with HS hub:

```bash
$ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
```

I.e. the setup that works with default dwc_otg settings.

P33M, 2019-04-03:
Yep this is a bug. dwc-otg will erroneously try to issue start-splits based on hub address and device speed. dwc2 is doing the "right" thing but the fix will need testing on dwc-otg with a full-speed hub...

Lauszus, 2019-04-03:
Yeah exactly. I can test it on both a FS and HS hub :)

P33M, 2019-04-09:
Candidate fix in https://github.com/raspberrypi/linux/pull/2925

P33M, 2019-04-10:
Can you run rpi-update and re-test?

Lauszus, 2019-04-11:
@P33M no unfortunately that didn't solve the issue.

P33M, 2019-04-23:
To be clear, in the failing case you can still fix the behaviour by disabling microframe_schedule?

Lauszus, 2019-04-23:
@P33M yes, setting ```dwc_otg.microframe_schedule=0``` works around the issue.

JamesH65, 2019-09-19:
@P33M @Lauszus Did this get any further?

Lauszus, 2019-09-21:
@JamesH65 no I did not look into this any further, as I worked around it by disabling microframe scheduling in our product.

P33M, 2021-05-25:
I'm closing this as the OP has implemented a workaround. dwc_otg is effectively in maintenance mode, which means there needs to be a compelling reason to disturb code that could affect a plethora of use-cases.

closed 2021-05-25 by P33M