raspberrypi/linux #2918: Cannot use a WiFi name with an apostrophe in it

jimbobbennett, 2019-04-02:
**Describe the bug**
The Pi cannot connect to a WiFi SSID with an apostrophe in the name.

**To reproduce**
Have a WiFi SSID with an apostrophe in the name (for example Jim's iPhone. the default personal hotspot name on my iPhone).
Try to connect to this.

**Expected behaviour**
The WiFi name should appear correctly and the Pi should be able to connect to it.

**Actual behaviour**
The WiFi name appears wrong in the SSID list, with the apostrophe replaced with `\342\200\231`. 

[image]

After entering the password for this WiFi connection, an error pops up saying **Invalid Argument** and a connection cannot be made.

[image]

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW - P13B+
* Which OS and version (`cat /etc/rpi-issue`)? 
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

pelwell, 2019-04-02:
Does it work if you add the entry manually to wpa_supplicant.conf?

JamesH65, 2019-04-02:
There's no RFC for which characters can be used in SSID's, but the recommendation appears to be to avoid special characters. Looks like we are not the only device that complains when apostrophes are used...! A quick Google will show quite a few issues elsewhere.

carmeli-tamir, 2019-04-14:
Hi, not sure what is the contributions policy but I'm willing to check this bug out. May I be assigned to the issue? Contact me if necessary carmeli.tamir@gmail.com

JamesH65, 2019-09-19:
@carmeli-tamir Feel free to investigate and provide a PR if you get anywhere. No need to assign it.

JamesH65, 2019-09-19:
@spl237 Is this in anyway related to that recent fix?

pelwell, 2019-09-19:
Yes - this is almost certainly a duplicate of https://github.com/raspberrypi/linux/issues/2883.

spl237, 2019-09-19:
Three octal characters with the digit 2 in - yes, almost certainly the same bug!

spl237, 2019-09-19:
Fix should be in apt by this evening if it isn't there already.

JamesH65, 2019-09-19:
@jimbobbennett It appears that there will be a possible fix for this in tonight apt update. So if you could update tomorrow using `sudo apt update`, `sudo apt upgrade` and retest and provide results that would be most excellent.

jimbobbennett, 2019-09-20:
I can now connect. The WiFi SSID is still displayed wrong in the list, but I can successfully connect.

JamesH65, 2019-09-20:
@spl237 One for you perhaps?

spl237, 2019-09-20:
I suspect the wrong display is due to the AP rendering the apostrophe in Unicode, which the plugin / dhcpcd doesn't support. If the OP could confirm how the SSID name appears (and what it should be), I can pass this on to the author of dhcpcd.

(I suspect the AP name includes a ` rather than a standard ' apostrophe - that quite often gets rendered badly.)

jimbobbennett, 2019-09-20:
It's the same as in the original issue, the apostrophe is replaced with `\342\200\231`. 

[image]

spl237, 2019-09-20:
Right, so that's E2 80 99 in hex, which is the Unicode for a "right single quotation mark". That's an extended character set symbol - the apostrophe is 27 in hex in Unicode. They look similar, but are not the same character.

iOS is using a bad character in the SSID name - that's not something which can be fixed without adding full Unicode support to the SSID parser, which is a non-trivial job. I'm pointing the finger of blame at Apple here... ;)

jimbobbennett, 2019-09-20:
Damn you Tim Apple!!!!!

jimbobbennett, 2019-09-20:
I guess the important thing here is fixed - users can connect a Pi to an iOS personal hotspot. Users savvy enough to do this should be ok to recognize their own hotspot if it has the name mangled around the apostrophe, so although it's not a perfect experience, it's not a show stopper.

Thanks @spl237

pelwell, 2019-09-20:
That's progress. The remaining issue is a Unicode handling problem in the LXPanel plugin; that apostrophe isn't really an apostrophe (or so I've read), it's a single right quote which puts it into the Unicode character set.

pelwell, 2019-09-20:
Apologies for the echo - it's one of the perils of patchy connectivity.

JamesH65, 2020-02-13:
@spl Anything left for us to do, or can this be closed?

spl237, 2020-02-13:
The show-stopper bug in the octal parser is fixed; Unicode support hasn't been added to the dhcpcd code, but that isn't likely to happen in the short term. I think we can close this for now.

JamesH65, 2020-02-13:
Closing this issue as questions answered/issue resolved.

closed 2020-02-13 by JamesH65

csz5024, 2022-12-01:
Just wanted to add to this thread for posterity: 

I added a WiFi SSID with the same apostrophie in my wpa_supplicant.conf (or in the advanced page of the RPi imaging software) and on first run, Raspberry Pi would not boot.

Removed "quiet" and "splash" from my cmdline.txt to see what the problem was, and sure enough there was an EOF error in my /boot/firstrun.sh file that was caused by the apostrophie in the SSID that I added to the wpa_supplicant. Seems like a pretty fatal bug, and thought it should be documented somewhere. 

The physical behavior of the Pi is that it would boot normally, then power off after the splash screen with 10 short green light blinks indicating HALT state. 

TLDR: Dont put apostrophies in your WiFi SSID.

pelwell, 2022-12-01:
I'm reminded of little Johnny Drop Tables: https://xkcd.com/327/

JosephCatrambone, 2022-12-06:
Failing to boot is a really annoying issue.  I couldn't figure out why my Zero W wouldn't start or connect until I plugged in a monitor and keyboard.  Saw the EOF -> HALT problem and ended up here.  

Perhaps the imager could alert that "an apostrophe in SSID will prevent the device from booting" or could disallow characters that will prevent the device from starting?  It's not exactly a workaround for the problem, but maybe a little safety for what would be a nasty issue?