raspberrypi/linux #2924: 2 usb ethernet adapters: cable plug-in causes kernel-4.19.x to crash

m1027, 2019-04-08:
No issues with kernel-4.14.98.

Tested setup:
- kernel-4.19.30 to 4.19.32 and according firmware
- rpi 3b or 3b+
- booting from HDD via USB SATA adapter, no sd-card
- 1x onboard ethernet connected (short cable)
- 2x USB ethernet via 0b95:772b ASIX Electronics Corp. AX88772B
- alternatively tested: 2x USB ethernet via 0b95:1790 ASIX Electronics Corp.
  AX88179 Gigabit Ethernet
- ethernet cables to the 2x USB adapters quite long (approx. 20m each)
- tested with other/new cables (1 short at usb, 2 long at usb/onboard)
- 2.5 A current

To trigger the kernel crash:
- fresh boot with all devices plugged into the rpi but one ethernet cable
  not yet connected to USB adapter
- then connect ethernet cable to USB adapter: crash soon after adapter is
  broad up

Immediate dmesg message after plugging the ethernet cable:
[   32.537340] ERROR::assign_and_init_hc:1394: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case

Kernel crash some seconds later:
[   78.078870] ------------[ cut here ]------------
[   78.083672] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:461 dev_watchdog+0x298/0x29c
[   78.092196] NETDEV WATCHDOG: eth_inet (ax88179_178a): transmit queue 0 timed out
[   78.099835] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.19.32-v7+ #37
[   78.106456] Hardware name: BCM2835
[   78.109977] [<80110070>] (unwind_backtrace) from [<8010c82c>] (show_stack+0x20/0x24)
[   78.117941] [<8010c82c>] (show_stack) from [<80ac87a8>] (dump_stack+0xcc/0x110)
[   78.125399] [<80ac87a8>] (dump_stack) from [<80127898>] (__warn.part.3+0xc8/0xe4)
[   78.133033] [<80127898>] (__warn.part.3) from [<8012791c>] (warn_slowpath_fmt+0x68/0x70)
[   78.141286] [<8012791c>] (warn_slowpath_fmt) from [<809260bc>] (dev_watchdog+0x298/0x29c)
[   78.152419] [<809260bc>] (dev_watchdog) from [<801995c0>] (call_timer_fn+0x3c/0x1a4)
[   78.166200] [<801995c0>] (call_timer_fn) from [<80199820>] (expire_timers+0xf8/0x168)
[   78.180364] [<80199820>] (expire_timers) from [<80199934>] (run_timer_softirq+0xa4/0x1c0)
[   78.194954] [<80199934>] (run_timer_softirq) from [<80102370>] (__do_softirq+0x188/0x410)
[   78.209607] [<80102370>] (__do_softirq) from [<8012d3f0>] (irq_exit+0xf8/0x134)
[   78.220293] [<8012d3f0>] (irq_exit) from [<80183618>] (__handle_domain_irq+0x70/0xc4)
[   78.234765] [<80183618>] (__handle_domain_irq) from [<8010219c>] (bcm2836_arm_irqchip_handle_irq+0x60/0xa8)
[   78.251219] [<8010219c>] (bcm2836_arm_irqchip_handle_irq) from [<801019bc>] (__irq_svc+0x5c/0x7c)
[   78.266635] Exception stack(0x81001ee8 to 0x81001f30)
[   78.274964] 1ee0:                   801092a4 00000000 40000093 40000093 ffffe000 8100616c
[   78.289630] 1f00: 810061b4 00000001 00000001 8109cd8d 80cc1a74 81001f44 81000000 81001f38
[   78.304380] 1f20: 00000000 801092a8 40000013 ffffffff
[   78.312744] [<801019bc>] (__irq_svc) from [<801092a8>] (arch_cpu_idle+0x34/0x4c)
[   78.326558] [<801092a8>] (arch_cpu_idle) from [<80ae3000>] (default_idle_call+0x40/0x48)
[   78.341068] [<80ae3000>] (default_idle_call) from [<80157144>] (do_idle+0x134/0x174)
[   78.355248] [<80157144>] (do_idle) from [<80157420>] (cpu_startup_entry+0x28/0x2c)
[   78.369266] [<80157420>] (cpu_startup_entry) from [<80adcc34>] (rest_init+0xb8/0xbc)
[   78.383454] [<80adcc34>] (rest_init) from [<80f01020>] (start_kernel+0x3b4/0x3c8)
[   78.397396] ---[ end trace 19ce283ffed865af ]---
[   78.405259] ERROR::assign_and_init_hc:1394: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[   78.405259]
[   78.428274] ERROR::assign_and_init_hc:1394: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case

P33M, 2019-04-09:
Please post a full dmesg listing. Running out of memory in this context is likely to be fatal to other consumers, not just USB.

m1027, 2019-04-14:
Thanks for your reply.

Here the full dmesg. You'll see:

- fresh boot
- memory alloc failed
- then my interactive login, just to be prepared for shutdown later
- then the kernel crash
- then I issue: systemctl poweroff
- powerdown still works if I do not wait for more crashs

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.32-v7+ (root@rpi) (gcc version 8.2.0 (Gentoo 8.2.0-r6 p1.7)) #54 SMP Thu Apr 4 10:48:40 CEST 2019
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3dc00000
[    0.000000] percpu: Embedded 15 pages/cpu @(ptrval) s40204 r0 d21236 u61440
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 253242
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  console=ttyAMA0,115200n8 console=tty1 root=/dev/sda3 rootwait init=/usr/lib/systemd/systemd usbcore.autosuspend=-1 rw
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 987108K/1021952K available (10240K kernel code, 689K rwdata, 2464K rodata, 1024K init, 434K bss, 26652K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbe600000   ( 998 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (11232 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 690 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 435 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 35411 entries in 104 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000278] Console: colour dummy device 80x30
[    0.000860] console [tty1] enabled
[    0.000917] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000964] pid_max: default: 32768 minimum: 301
[    0.001175] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001210] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002306] Disabling memory control group subsystem
[    0.002392] CPU: Testing write buffer coherency: ok
[    0.002855] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003504] Setting up static identity map for 0x100000 - 0x10003c
[    0.003670] rcu: Hierarchical SRCU implementation.
[    0.004467] smp: Bringing up secondary CPUs ...
[    0.005297] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006194] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007024] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.007141] smp: Brought up 1 node, 4 CPUs
[    0.007224] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.007248] CPU: All CPU(s) started in HYP mode.
[    0.007269] CPU: Virtualization extensions available.
[    0.008265] devtmpfs: initialized
[    0.018941] random: get_random_u32 called from bucket_table_alloc+0x94/0x1a4 with crng_init=0
[    0.019650] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.019902] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.019981] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.020585] xor: measuring software checksum speed
[    0.120026]    arm4regs  :   920.000 MB/sec
[    0.220112]    8regs     :   685.200 MB/sec
[    0.320196]    32regs    :   659.600 MB/sec
[... 308 lines cut ...]
[   84.326816] [<80ae3000>] (default_idle_call) from [<80157144>] (do_idle+0x134/0x174)
[   84.341634] [<80157144>] (do_idle) from [<80157420>] (cpu_startup_entry+0x28/0x2c)
[   84.356304] [<80157420>] (cpu_startup_entry) from [<80adcc34>] (rest_init+0xb8/0xbc)
[   84.371161] [<80adcc34>] (rest_init) from [<80f01020>] (start_kernel+0x3b4/0x3c8)
[   84.385817] ---[ end trace a3e1fa7b43dc358e ]---
[   84.394793] ERROR::assign_and_init_hc:1394: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[   84.394793] 

root@rpi ~ # systemctl poweroff
[   98.117761] systemd-shutdow: 29 output lines suppressed due to ratelimiting
[   98.141589] systemd-shutdown[1]: Syncing filesystems and block devices.
[   98.388066] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[... cut ...]
[   98.548451] BTRFS info (device sda3): disk space caching is enabled
[   98.748321] systemd-shutdown[1]: All filesystems unmounted.
[   98.756129] systemd-shutdown[1]: Deactivating swaps.
[   98.763464] systemd-shutdown[1]: All swaps deactivated.
[   98.770888] systemd-shutdown[1]: Detaching loop devices.
[   98.781066] systemd-shutdown[1]: All loop devices detached.
[   98.788254] systemd-shutdown[1]: Detaching DM devices.
[   98.798600] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[   98.848792] reboot: Power down

mkosch, 2019-04-27:
I experience a similar problem since kernel 4.19.23 and newer. Everything works up to kernel 4.14.98.

Tested setup:
- Kernel 4.19.23 and newer
- Two "slave" Raspberry Pi Zero W (each with camera module) setup as USB Ethernet gadget (g_ether) with USB connection to a "main" Raspberry Pi 3B+
- All Raspberry Pis boot without GUI using SD card
- No Ethernet cable connected to main Raspberry Pi, instead using WLAN
- Radio devices of both slave Raspberry Pis are disabled
- Main Raspberry Pi powered with Pichler's BEC BOY (UBEC 3A) from Mignon batteries or 6A power adapter
- Both slave Raspberry Pis powered with USB from main Raspberry Pi
- SSH connection to main Raspberry Pi

What happens:
- When only one slave Raspberry Pi is connected to main Raspberry Pi, then the Ethernet connection seems fine and I can ping the slave Raspberry Pi from the main Raspberry Pi
- When both slave Raspberry Pis are connected to main Raspberry Pi, then no Ethernet connection seems to be possible to at least one of the slave Raspberry Pis. It may be possible to ping the other slave Raspberry Pi, possibly with packet loss. I didn't explore this extensively.
- No crash occurs.

Test case:
- I start the main Raspberry Pi without slave Raspberry Pis connected.
- After connecting the first slave Raspberry Pi I get the following "dmesg" messages:
```
[  291.844239] usb 1-1.3: new high-speed USB device number 5 using dwc_otg
[  292.224254] usb 1-1.3: device descriptor read/64, error -71
[  292.664253] usb 1-1.3: device descriptor read/64, error -71
[  292.884238] usb 1-1.3: new high-speed USB device number 6 using dwc_otg
[  293.015142] usb 1-1.3: New USB device found, idVendor=0525, idProduct=a4a2, bcdDevice= 4.19
[  293.015155] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  293.015165] usb 1-1.3: Product: RNDIS/Ethernet Gadget
[  293.015175] usb 1-1.3: Manufacturer: Linux 4.19.27+ with 20980000.usb
[  293.071965] cdc_ether 1-1.3:1.0 usb0: register 'cdc_ether' at usb-3f980000.usb-1.3, CDC Ethernet Device, 76:a0:f6:6a:35:7f
[  293.072195] usbcore: registered new interface driver cdc_ether
[  293.072287] usbcore: registered new interface driver cdc_subset
[  293.149633] br0: port 1(usb0) entered blocking state
[  293.149646] br0: port 1(usb0) entered disabled state
[  293.150034] device usb0 entered promiscuous mode
[  293.158377] IPv6: ADDRCONF(NETDEV_UP): usb0: link is not ready
[  293.158398] br0: port 1(usb0) entered blocking state
[  293.158408] br0: port 1(usb0) entered forwarding state
[  294.154416] br0: port 1(usb0) entered disabled state
[  306.071431] IPv6: ADDRCONF(NETDEV_CHANGE): usb0: link becomes ready
[  306.071577] br0: port 1(usb0) entered blocking state
[  306.071587] br0: port 1(usb0) entered forwarding state
```
- After connecting the second slave Raspberry Pi I get the following "dmesg" messages:
```
[  433.941518] usb 1-1.3: USB disconnect, device number 6
[  433.941760] cdc_ether 1-1.3:1.0 usb0: unregister 'cdc_ether' usb-3f980000.usb-1.3, CDC Ethernet Device
[  433.967139] br0: port 1(usb0) entered disabled state
[  433.980661] device usb0 left promiscuous mode
[  433.980678] br0: port 1(usb0) entered disabled state
[  442.364891] usb 1-1.3: new high-speed USB device number 7 using dwc_otg
[  442.495651] usb 1-1.3: New USB device found, idVendor=0525, idProduct=a4a2, bcdDevice= 4.19
[  442.495666] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  442.495676] usb 1-1.3: Product: RNDIS/Ethernet Gadget
[  442.495688] usb 1-1.3: Manufacturer: Linux 4.19.27+ with 20980000.usb
[  442.505158] cdc_ether 1-1.3:1.0 usb0: register 'cdc_ether' at usb-3f980000.usb-1.3, CDC Ethernet Device, 76:a0:f6:6a:35:7f
[  442.609940] br0: port 1(usb0) entered blocking state
[  442.609955] br0: port 1(usb0) entered disabled state
[  442.610337] device usb0 entered promiscuous mode
[  442.615784] IPv6: ADDRCONF(NETDEV_UP): usb0: link is not ready
[  442.615804] br0: port 1(usb0) entered blocking state
[  442.615813] br0: port 1(usb0) entered forwarding state
[  442.616160] br0: port 1(usb0) entered disabled state
[  442.814795] usb 1-1.1.2: new high-speed USB device number 8 using dwc_otg
[  442.945540] usb 1-1.1.2: New USB device found, idVendor=0525, idProduct=a4a2, bcdDevice= 4.19
[  442.945551] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  442.945556] usb 1-1.1.2: Product: RNDIS/Ethernet Gadget
[  442.945563] usb 1-1.1.2: Manufacturer: Linux 4.19.27+ with 20980000.usb
[  442.947450] cdc_ether 1-1.1.2:1.0 usb1: register 'cdc_ether' at usb-3f980000.usb-1.1.2, CDC Ethernet Device, 62:35:4c:93:cb:9b
[  443.015159] br0: port 2(usb1) entered blocking state
[... 26 lines cut ...]
[  474.075984] [<8019602c>] (expire_timers) from [<80196144>] (run_timer_softirq+0xb8/0x1ec)
[  474.076009] [<80196144>] (run_timer_softirq) from [<80102388>] (__do_softirq+0x190/0x3f0)
[  474.076032] [<80102388>] (__do_softirq) from [<80126384>] (irq_exit+0xfc/0x120)
[  474.076056] [<80126384>] (irq_exit) from [<8017e7ac>] (__handle_domain_irq+0x70/0xc4)
[  474.076075] [<8017e7ac>] (__handle_domain_irq) from [<801021b4>] (bcm2836_arm_irqchip_handle_irq+0x60/0xa4)
[  474.076090] [<801021b4>] (bcm2836_arm_irqchip_handle_irq) from [<801019bc>] (__irq_svc+0x5c/0x7c)
[  474.076105] Exception stack(0x80d01ee8 to 0x80d01f30)
[  474.076121] 1ee0:                   80109a04 00000000 40000093 40000093 80d04d70 80d00000
[... cut ...]
[  629.036245] ERROR::assign_and_init_hc:1390: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[  729.036718] ERROR::assign_and_init_hc:1390: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[  889.037570] ERROR::assign_and_init_hc:1390: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[  964.406130] ERROR::assign_and_init_hc:1390: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
[ 1001.517575] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -110
```

m1027, 2019-04-29:
Updates:

(1) The issue (summary)

- Rpi 3 or 3b+
- 3x ethernet connected (2 additional USB/ethernet adapters)
- kernel 4.19 crash soon after "ERROR::assign_and_init_hc:1391 ..."
- No issue with kenel 4.14 or when leaving out one USB connector

(2) Test with newer kernel

Same issues with newest kernel/firmware 4.19.36 (was 4.19.32).

(3) Low memory?

> Running out of memory in this context is likely to be fatal to other
> consumers, not just USB.

So, I did the following:

(3.1) Checking memory with free command

The free command before/after the first kernel error message does not reveal
any low mem situation:

free
             total        used        free      shared  buff/cache   available
 Mem:        1000900       45348      893316        5684       62236      872492
 Swap:       2097148           0     2097148
			  
[   79.535682] asix 1-1.5:1.0 eth_inet: link up, 100Mbps, full-duplex, lpa 0xCDE1
[  216.853593] ERROR::assign_and_init_hc:1391: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case

free
              total        used        free      shared  buff/cache   available
  Mem:        1000900       38780      892936       10472       69184      873200
  Swap:       2097148           0     2097148

(3.2) Booting with nearly no user space daemon (no sshd, no apache, etc)

Still the same issue.

(4) New hints on what can trigger the issue

To recall: In my OP I saw a work-around by not connecting one of the two
USB/ethernet adapters.

One of the adapters is connected to a WLAN router.  Now I see that I *can*
boot with all adapters connected without the issue, and later trigger the
above error message immediately when the first laptop connects over WLAN and
tries to send/receive some data (like connecting to a website in my case).

I repeated that several times by rebooting the RPI, waiting an arbitrary time
and then connecting a laptop over WLAN / opening a website.

(4) Conclusion for now

I cannot confirm a memory issue yet (always note that kernel 4.14.x works).

I can trigger the error by the first network packets sent over one of the
adapters.

P33M, 2019-04-29:
I can replicate with 2x Pi zero plugged into a 3B+ with the zero devices acting as CDC-ether gadgets.

P33M, 2019-04-29:
Oh this is fun. In 4.14.122, there are 3 allocations for non-aligned buffers across three Ethernet devices (lan95xx plus two cdc_ether devices). Each of these is a 64k-1b buffer. In 4.19, the number of allocations grows to 5, with 2 of the requests failing. Increasing the CMA region size doesn't help; even with a 32MB region the same number of failures occurs.

There's two worrying implications here:
- usbnet has always been allocating misaligned buffers, such that for every network packet received or transmitted there's an additional memcpy to or from dwc_otg's align_buf
- the maximum number of these sorts of buffers that you can align is severely restricted, probably because we're doing what's essentially dma_alloc_contiguous() in an atomic context

P33M, 2019-04-29:
@m1027 @mkosch can you try adding the parameter "coherent_pool=4M" to /boot/cmdline.txt?

6by9, 2019-04-29:
4.20 gains https://github.com/raspberrypi/linux/commit/1ee5c23eb5b7bb3ac8a6848e91265a294a811246 which signals up the stack that some part of the buffers can be aligned. Does that help in this context? 
I think the same thing is valid for lan7800.c if we're seeing the issue on both 3B and 3B+.

m1027, 2019-04-29:
> @m1027 @mkosch can you try adding the parameter "coherent_pool=4M" to /boot/cmdline.txt?

Man, this is awesome.

It seems we can finally upgrade the kernel from 4.14 to 4.19.!

No error message, no crash since 20 minutes, and also some reboots.

In case you want to drop some lines about what this actually changes, when
(not) to apply this setting, possible drawbacks... that would help, too, as
there are several slightly different setups here.

mkosch, 2019-04-29:
> @m1027 @mkosch can you try adding the parameter "coherent_pool=4M" to /boot/cmdline.txt?
It seems to work. Thanks.

I can ping the slave Raspberry Pis and connect to it via SSH with the latest kernel. It even works with a third USB Ethernet device attached.

P33M, 2019-04-30:
Increasing the coherent pool size reduces total available RAM for other users, so this is a diagnostic test rather than a fix. This is a boot-time allocated pool of memory for DMA-coherent buffers allocated in atomic context. The way dwc_otg is using this is suboptimal - the allocations it uses are large and persist until the device is disconnected. The aligned buffer also incurs an extra memcpy to/from the provided buffer, so this also adds a performance penalty. 

Usbnet should make a better effort at allocating aligned RX buffers, since that's gone backwards since 4.14.

P33M, 2019-04-30:
Naively forcing NET_IP_ALIGN to be 0 stops RX buffers being misaligned, but ipv6 multicast tx is giving us misaligned buffers - so we still get one 64k allocation per net device. Somehow I think we're never going to completely get rid of the requirement to allocate a bounce buffer, but at the very least we need to stop memcpying every single RX packet.

I'll do some profiling to see which is the least worst option.

JamesH65, 2019-04-30:
#2599 Is this related?

P33M, 2019-04-30:
There's a difference in nomenclature here - from the point of view of the USB hardware, it needs the start of a read or write buffer to be 32-bit aligned. I'm not tracking CPU alignment faults, but there's potentially knobs in the USB-ethernet drivers for changing the offset of packet data start relative to the buffer data start if changing the URB alignment impacts the frequency of CPU faults.

P33M, 2019-04-30:
I've done various trials monitoring how often an unaligned URB is passed to dwc_otg. Between 4.14 and 4.19, usb-ethernet RX buffers became "unaligned" for ASIX and cdc_ether drivers, but unchanged for smsc95xx. In both 4.14 and 4.19 tx buffers are unaligned in almost all cases, with ASIX being the exception for ipv4 packets.

Setting NET_IP_ALIGN=0 reverts 4.19 to the old behaviour, as in no unaligned RX buffers. However, with the misalignment of IP headers we now get a lot more faults being handled - I believe one for every ipv6 packet received.

It also looks like for ASIX and cdc_ether, the tx header must immediately precede the data so there's no scope for adding a padding fudge factor. The smsc95xx and lan78xx drivers need further investigation.

P33M, 2019-04-30:
smsc95xx can have a variable start offset of up to 3 bytes between the 2 tx words and the start of data, specified in tx_cmd_a[17:16] - so we can mangle TX buffers presented to dwc_otg to be aligned to a word boundary. In lan78xx, there's multiple undefined bitfields in the TX command words that could be start-of-data offset fields. Assuming we can fixup both of these drivers, the default Ethernet interfaces should no longer require bounce buffers.

For an immediate "make 4.19 be less broken than 4.14" patch, I propose expanding the atomic allocation pool to 512K and moving the allocation code in dwc_otg up several levels so that USB device drivers get told about failure.

P33M, 2019-05-02:
Please retest with latest rpi-update firmware.

mkosch, 2019-05-02:
> Please retest with latest rpi-update firmware.

It doesn't work with latest kernel 4.19.37-v7+ (hash 18e0a0f9a31e7a3a47d9c4301c7705b980ab0516) and without specifying the command line parameter "coherent_pool". I can (more or less) ping the first slave Raspberry Pi, but cannot ping the second one.

Output from "dmesg":
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.37-v7+ (dom@dom-XPS-13-9370) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1216 SMP Thu May 2 13:32:40 BST 2019
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x37800000
[    0.000000] On node 0 totalpages: 229376
[    0.000000]   Normal zone: 2016 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 229376 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632
[    0.000000] pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 227360
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=6bda1aab-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 886304K/917504K available (8192K kernel code, 627K rwdata, 2164K rodata, 1024K init, 821K bss, 23008K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xb8800000 - 0xff800000   (1136 MB)
                   lowmem  : 0x80000000 - 0xb8000000   ( 896 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 628 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 822 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26225 entries in 77 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000023] Switching to timer-based delay loop, resolution 52ns
[    0.000287] Console: colour dummy device 80x30
[    0.000903] console [tty1] enabled
[    0.000964] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001009] pid_max: default: 32768 minimum: 301
[    0.001368] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001404] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002334] CPU: Testing write buffer coherency: ok
[    0.002817] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003482] Setting up static identity map for 0x100000 - 0x10003c
[    0.003646] rcu: Hierarchical SRCU implementation.
[    0.004455] smp: Bringing up secondary CPUs ...
[    0.005323] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006243] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007105] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.007225] smp: Brought up 1 node, 4 CPUs
[    0.007307] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.007331] CPU: All CPU(s) started in HYP mode.
[    0.007351] CPU: Virtualization extensions available.
[    0.008324] devtmpfs: initialized
[    0.020141] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.020412] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.020460] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.021065] pinctrl core: initialized pinctrl subsystem
[    0.021956] NET: Registered protocol family 16
[    0.024215] DMA: preallocated 256 KiB pool for atomic coherent allocations
[... 280 lines cut ...]
[   46.072525] [<80120904>] (__warn) from [<80120974>] (warn_slowpath_fmt+0x58/0x74)
[   46.072540] [<80120974>] (warn_slowpath_fmt) from [<807352b0>] (dev_watchdog+0x294/0x298)
[   46.072561] [<807352b0>] (dev_watchdog) from [<8019654c>] (call_timer_fn+0x3c/0x198)
[   46.072580] [<8019654c>] (call_timer_fn) from [<80196794>] (expire_timers+0xec/0x14c)
[   46.072597] [<80196794>] (expire_timers) from [<801968ac>] (run_timer_softirq+0xb8/0x1ec)
[... cut ...]
[   46.072800] [<80152478>] (do_idle) from [<801527c8>] (cpu_startup_entry+0x28/0x2c)
[   46.072816] [<801527c8>] (cpu_startup_entry) from [<80824fe4>] (rest_init+0xbc/0xc0)
[   46.072835] [<80824fe4>] (rest_init) from [<80c00fb0>] (start_kernel+0x484/0x4b4)
[   46.072848] ---[ end trace f91d15236d9d5cd9 ]---
[   46.073615] ERROR::assign_and_init_hc:1390: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case

[   76.068545] ERROR::assign_and_init_hc:1390: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case
```

pelwell, 2019-05-02:
I'm not surprised it fails:
```
    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1
```
The fix was to modify the dtb to include the necessary command line parameter, but clearly for some reason that hasn't worked.

m1027, 2019-05-02:
> > Please retest with latest rpi-update firmware.
> 
> It doesn't work with latest kernel 4.19.37-v7+ (hash
> 18e0a0f9a31e7a3a47d9c4301c7705b980ab0516) and without specifying the
> `/boot/cmdline.txt` parameter "coherent_pool". I can (more or less) ping the
> first slave Raspberry Pi, but cannot ping the second one.

Similar here. With coherent_pool it works, without I get this:

[   36.289695] ERROR::assign_and_init_hc:1390: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case

[  203.127427] ------------[ cut here ]------------
[  203.132153] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:461 dev_watchdog+0x288/0x28c
[  203.140554] NETDEV WATCHDOG: eth_lan0 (asix): transmit queue 0 timed out
[  203.147387] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.19.37-v7+ #42
[  203.153908] Hardware name: BCM2835
[  203.157375] [<8010d3b4>] (unwind_backtrace) from [<8010ac6c>] (show_stack+0x10/0x14)
[  203.165218] [<8010ac6c>] (show_stack) from [<8083f724>] (dump_stack+0x98/0xb8)
[  203.172482] [<8083f724>] (dump_stack) from [<80121e6c>] (__warn+0xd0/0xec)
[  203.179395] [<80121e6c>] (__warn) from [<80121b20>] (warn_slowpath_fmt+0x38/0x48)
[  203.186922] [<80121b20>] (warn_slowpath_fmt) from [<806fff54>] (dev_watchdog+0x288/0x28c)
[  203.195149] [<806fff54>] (dev_watchdog) from [<801824cc>] (call_timer_fn+0x24/0x94)
[  203.202850] [<801824cc>] (call_timer_fn) from [<801825d8>] (expire_timers+0x9c/0xa8)
[  203.212948] [<801825d8>] (expire_timers) from [<80182678>] (run_timer_softirq+0x94/0x1a8)
[  203.226018] [<80182678>] (run_timer_softirq) from [<8010210c>] (__do_softirq+0x114/0x278)
[  203.239362] unwind: Unknown symbol address 8010210c
[  203.246870] unwind: Index not found 8010210c
[  203.253697] ---[ end trace 8207672d8a842813 ]---
[  204.908228] ERROR::assign_and_init_hc:1390: assign_and_init_hc: Failed to allocate memory to handle non-dword aligned buffer case

My cmdline is:

console=ttyAMA0,115200n8 console=tty1 root=/dev/sda3 rootfstype=btrfs rootwait init=/usr/lib/systemd/systemd usbcore.autosuspend=-1 coherent_pool=4M

pelwell, 2019-05-02:
It appears that only the non-WiFi DTBs have been updated in the firmware repo - bm2710-rpi-3-b.dtb, bcm2710-rpi-3-b-plus.dtb and bcm2708-rpi-0-w.dtb are unchanged. Building the DTBs myself gives the same result - doh! The WiFi/BT equipped Pis use the other UART - an 8250 clone - which requires that "8250.nr_uarts=1" is in the kernel command line. I told @P33M that the firmware has special handling for the bootargs property that allows overlays to append to the kernel command line. However, this doesn't apply during the building of the DTBs, so the updating of bootargs to add the 8250 parameter overwrites the coherent_pool parameter.

591bd5452ad fixes the DTBs for the remaining Pis by explicitly including the coherent_pool setting wherever bootargs is defined.

P33M, 2019-05-03:
And of course I was testing on a Pi2 due to simultaneous debugging of smsc95xx and not a 3b+... derp

m1027, 2019-05-05:
> [591bd54](https://github.com/raspberrypi/linux/commit/591bd5452ad361deedc19d21d3f98bf1eab623ca)
> fixes the DTBs for the remaining Pis by explicitly including the
> coherent_pool setting wherever bootargs is defined.

Ok, for completeness: After applying the abovementioned commit, coherent_pool
does obviously not need to be configured on the cmdline anymore as you apply
it internally in any case.

Thank's a lot for all your support on fixing this issue.

As my box is doing well without issues I'll close this issue now.

macmpi, 2019-05-06:
Shouldn't we keep this issue opened until a **definitive fix** is applied, since current solution seems more of **workaround** at this point?

P33M, 2019-05-07:
Correct, the coherent pool is a memory reservation that's unavailable to other users. Ideally dwc_otg should not allocate in atomic context at all, but there's a pile of edge cases under the heading "we're partway through a transfer and find we have to do an unaligned DMA" that need understanding.

macmpi, 2019-06-07:
> Ideally dwc_otg should not allocate in atomic context at all
Is this been tracked/worked-out upstream somewhere? Thanks.

pelwell, 2019-06-07:
> Is this been tracked/worked-out upstream somewhere? Thanks.

dwc_otg is a downstream-only driver, so no.

andrum99, 2019-06-18:
I'm getting a similar stack trace on my Pi 1:

`Jun 14 15:50:41 raspi2 kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jun 14 15:50:41 raspi2 kernel: [    0.000000] Linux version 4.19.42+ (dom@dom-XPS-13-9370) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1219 Tue May 14 21:16:38 BST 2019
Jun 14 15:50:41 raspi2 kernel: [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Jun 14 15:50:41 raspi2 kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Jun 14 15:50:41 raspi2 kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi Model B Rev 1
Jun 14 15:50:41 raspi2 kernel: [    0.000000] Memory policy: Data cache writeback
Jun 14 15:50:41 raspi2 kernel: [    0.000000] cma: Reserved 8 MiB at 0x0e400000
Jun 14 15:50:41 raspi2 kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x49c with crng_init=0
Jun 14 15:50:41 raspi2 kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 60900
Jun 14 15:50:41 raspi2 kernel: [    0.000000] Kernel command line: coherent_pool=1M bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0xfa00000 vc_mem.mem_size=0x10000000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=63242b5f-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait ipv6.disable=1
Jun 14 15:50:41 raspi2 kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Jun 14 15:50:41 raspi2 kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Jun 14 15:50:41 raspi2 kernel: [    0.000000] Memory: 223564K/245760K available (6932K kernel code, 633K rwdata, 2068K rodata, 456K init, 797K bss, 14004K reserved, 8192K cma-reserved)
Jun 14 15:50:41 raspi2 kernel: [    0.000000] Virtual kernel memory layout:
Jun 14 15:50:41 raspi2 kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Jun 14 15:50:41 raspi2 kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Jun 14 15:50:41 raspi2 kernel: [    0.000000]     vmalloc : 0xcf800000 - 0xff800000   ( 768 MB)
Jun 14 15:50:41 raspi2 kernel: [    0.000000]     lowmem  : 0xc0000000 - 0xcf000000   ( 240 MB)
Jun 14 15:50:41 raspi2 kernel: [    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
Jun 14 15:50:41 raspi2 kernel: [    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (6934 kB)
Jun 14 15:50:41 raspi2 kernel: [    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   ( 456 kB)
Jun 14 15:50:41 raspi2 kernel: [    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 634 kB)
Jun 14 15:50:41 raspi2 kernel: [    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 798 kB)
Jun 14 15:50:41 raspi2 kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Jun 14 15:50:41 raspi2 kernel: [    0.000000] ftrace: allocating 25044 entries in 74 pages
Jun 14 15:50:41 raspi2 kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Jun 14 15:50:41 raspi2 kernel: [    0.000029] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Jun 14 15:50:41 raspi2 kernel: [    0.000086] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
Jun 14 15:50:41 raspi2 kernel: [    0.000194] bcm2835: system timer (irq = 27)
Jun 14 15:50:41 raspi2 kernel: [    0.000794] Console: colour dummy device 80x30
Jun 14 15:50:41 raspi2 kernel: [    0.001423] console [tty1] enabled
Jun 14 15:50:41 raspi2 kernel: [    0.001505] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
Jun 14 15:50:41 raspi2 kernel: [    0.060374] pid_max: default: 32768 minimum: 301
Jun 14 15:50:41 raspi2 kernel: [    0.060964] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Jun 14 15:50:41 raspi2 kernel: [    0.061023] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Jun 14 15:50:41 raspi2 kernel: [    0.062485] CPU: Testing write buffer coherency: ok
Jun 14 15:50:41 raspi2 kernel: [    0.063881] Setting up static identity map for 0x8200 - 0x8238
Jun 14 15:50:41 raspi2 kernel: [    0.065310] devtmpfs: initialized
Jun 14 15:50:41 raspi2 kernel: [    0.076164] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
Jun 14 15:50:41 raspi2 kernel: [    0.076650] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Jun 14 15:50:41 raspi2 kernel: [    0.076726] futex hash table entries: 256 (order: -1, 3072 bytes)
Jun 14 15:50:41 raspi2 kernel: [    0.078122] pinctrl core: initialized pinctrl subsystem
Jun 14 15:50:41 raspi2 kernel: [    0.079732] NET: Registered protocol family 16
Jun 14 15:50:41 raspi2 kernel: [    0.082977] DMA: preallocated 1024 KiB pool for atomic coherent allocations
Jun 14 15:50:41 raspi2 kernel: [    0.090135] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
Jun 14 15:50:41 raspi2 kernel: [    0.090212] hw-breakpoint: maximum watchpoint size is 4 bytes.
Jun 14 15:50:41 raspi2 kernel: [    0.090384] Serial: AMBA PL011 UART driver
Jun 14 15:50:41 raspi2 kernel: [    0.094071] bcm2835-mbox 2000b880.mailbox: mailbox enabled
Jun 14 15:50:41 raspi2 kernel: [    0.136638] bcm2835-dma 20007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
Jun 14 15:50:41 raspi2 kernel: [    0.139219] SCSI subsystem initialized
Jun 14 15:50:41 raspi2 kernel: [    0.139543] usbcore: registered new interface driver usbfs
[... cut ...]
Jun 14 23:58:49 raspi2 kernel: [29284.910821] [<c06cb918>] (default_idle_call) from [<c0050130>] (do_idle+0x94/0xf4)
Jun 14 23:58:49 raspi2 kernel: [29284.910850] [<c0050130>] (do_idle) from [<c0050434>] (cpu_startup_entry+0x1c/0x20)
Jun 14 23:58:49 raspi2 kernel: [29284.910874] [<c0050434>] (cpu_startup_entry) from [<c06c5c54>] (rest_init+0x7c/0x9c)
Jun 14 23:58:49 raspi2 kernel: [29284.910899] [<c06c5c54>] (rest_init) from [<c0964ed4>] (start_kernel+0x3e8/0x49c)
Jun 14 23:58:49 raspi2 kernel: [29284.910913] ---[ end trace 4b5e7be83442ccbe ]---
`

andrum99, 2019-06-19:
I've also got a stack trace from a Pi 3B+. As far as I can tell there was nothing particularly happening on my Pi at the time, so from my analysis it looks somewhat random as to when this occurs. Can someone tell me what effect this has on network traffic? Does the network get reset, or does the driver just carry on as if nothing has happened?

Stack trace:

Jun 15 01:28:55 livingroom kernel: [14175.155918] ------------[ cut here ]------------
Jun 15 01:28:55 livingroom kernel: [14175.155961] WARNING: CPU: 1 PID: 85 at net/sched/sch_generic.c:461 dev_watchdog+0x294/0x298
Jun 15 01:28:55 livingroom kernel: [14175.155971] NETDEV WATCHDOG: eth0 (lan78xx): transmit queue 0 timed out
Jun 15 01:28:55 livingroom kernel: [14175.155977] Modules linked in: btrfs xor xor_neon zstd_decompress zstd_compress xxhash lzo_compress raid6_pq zlib_deflate evdev sg uas snd_soc_pcm512x_i2c snd_soc_bcm2835_i2s snd_soc_iqaudio_dac snd_soc_pcm512x bcm2835_v4l2(C) bcm2835_codec(C) snd_soc_core v4l2_mem2mem v4l2_common bcm2835_mmal_vchiq(C) snd_compress videobuf2_vmalloc videobuf2_dma_contig snd_pcm_dmaengine videobuf2_memops videobuf2_v4l2 snd_pcm videobuf2_common videodev snd_timer media snd raspberrypi_hwmon hwmon i2c_bcm2835 vc_sm_cma(C) uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
Jun 15 01:28:55 livingroom kernel: [14175.156243] CPU: 1 PID: 85 Comm: kworker/1:1H Tainted: G         C        4.19.42-v7+ #1219
Jun 15 01:28:55 livingroom kernel: [14175.156249] Hardware name: BCM2835
Jun 15 01:28:55 livingroom kernel: [14175.156271] Workqueue:    (null) (kblockd)
Jun 15 01:28:55 livingroom kernel: [14175.156311] [<80111eac>] (unwind_backtrace) from [<8010d430>] (show_stack+0x20/0x24)
Jun 15 01:28:55 livingroom kernel: [14175.156330] [<8010d430>] (show_stack) from [<8080ec80>] (dump_stack+0xd4/0x118)
Jun 15 01:28:55 livingroom kernel: [14175.156350] [<8080ec80>] (dump_stack) from [<80120904>] (__warn+0x104/0x11c)
Jun 15 01:28:55 livingroom kernel: [14175.156366] [<80120904>] (__warn) from [<80120974>] (warn_slowpath_fmt+0x58/0x74)
Jun 15 01:28:55 livingroom kernel: [14175.156384] [<80120974>] (warn_slowpath_fmt) from [<807355b8>] (dev_watchdog+0x294/0x298)
Jun 15 01:28:55 livingroom kernel: [14175.156404] [<807355b8>] (dev_watchdog) from [<801965a0>] (call_timer_fn+0x3c/0x198)
Jun 15 01:28:55 livingroom kernel: [14175.156424] [<801965a0>] (call_timer_fn) from [<801967e8>] (expire_timers+0xec/0x14c)
Jun 15 01:28:55 livingroom kernel: [14175.156441] [<801967e8>] (expire_timers) from [<80196900>] (run_timer_softirq+0xb8/0x1ec)
Jun 15 01:28:55 livingroom kernel: [14175.156459] [<80196900>] (run_timer_softirq) from [<80102388>] (__do_softirq+0x190/0x3f0)
Jun 15 01:28:55 livingroom kernel: [14175.156478] [<80102388>] (__do_softirq) from [<801268dc>] (irq_exit+0xfc/0x120)
Jun 15 01:28:55 livingroom kernel: [14175.156501] [<801268dc>] (irq_exit) from [<8017ef24>] (__handle_domain_irq+0x70/0xc4)
Jun 15 01:28:55 livingroom kernel: [14175.156522] [<8017ef24>] (__handle_domain_irq) from [<801021b4>] (bcm2836_arm_irqchip_handle_irq+0x60/0xa4)
Jun 15 01:28:55 livingroom kernel: [14175.156537] [<801021b4>] (bcm2836_arm_irqchip_handle_irq) from [<801019bc>] (__irq_svc+0x5c/0x7c)
Jun 15 01:28:55 livingroom kernel: [14175.156545] Exception stack(0xb63dfe20 to 0xb63dfe68)
Jun 15 01:28:55 livingroom kernel: [14175.156562] fe20: 8014ac30 00000000 40000093 40000093 80c92e40 00000000 a48f1e00 b6fd7e40
Jun 15 01:28:55 livingroom kernel: [14175.156575] fe40: 80826430 b63c8f00 00000002 b63dfeb4 80d0517c b63dfe70 00000000 8014ac34
Jun 15 01:28:55 livingroom kernel: [14175.156581] fe60: 40000013 ffffffff
Jun 15 01:28:55 livingroom kernel: [14175.156597] [<801019bc>] (__irq_svc) from [<8014ac34>] (finish_task_switch+0x9c/0x230)
Jun 15 01:28:55 livingroom kernel: [14175.156615] [<8014ac34>] (finish_task_switch) from [<80826430>] (__schedule+0x390/0x9ac)
Jun 15 01:28:55 livingroom kernel: [14175.156631] [<80826430>] (__schedule) from [<80826a9c>] (schedule+0x50/0xa8)
Jun 15 01:28:55 livingroom kernel: [14175.156646] [<80826a9c>] (schedule) from [<8013c174>] (worker_thread+0x110/0x5a4)
Jun 15 01:28:55 livingroom kernel: [14175.156665] [<8013c174>] (worker_thread) from [<801423ac>] (kthread+0x138/0x168)
Jun 15 01:28:55 livingroom kernel: [14175.156684] [<801423ac>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
Jun 15 01:28:55 livingroom kernel: [14175.156692] Exception stack(0xb63dffb0 to 0xb63dfff8)
Jun 15 01:28:55 livingroom kernel: [14175.156701] ffa0:                                     00000000 00000000 00000000 00000000
Jun 15 01:28:55 livingroom kernel: [14175.156713] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jun 15 01:28:55 livingroom kernel: [14175.156723] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
Jun 15 01:28:55 livingroom kernel: [14175.156793] ---[ end trace 7ebdcdba1beb5bca ]---

andrum99, 2019-06-19:
Just to be clear - my stack traces were not resulting in any crash - I just get the above warning and stack trace, then things seem to carry on as normal.

mkosch, 2019-08-15:
I tested my previous setup with a Raspberry Pi 4 B. The two slave Raspberry Pi Zero W are connected to the USB 3.0 ports as USB Ethernet gadgets.

Currently I've got two issues:

***First issue:*** At least after the first boot it appears that sometimes (?) the USB Ethernet devices are not recognized. In this case they don't show up in the output of the `ifconfig` command.

***Second issue:*** When the USB Ethernet devices are recognized then I can only ping one of them. When pinging the other one I get a "Destination Host Unreachable" error and `dmesg` shows "transmit queue 0 timed out". I've also tried the parameter `coherent_pool=4M` inside the configuration file `cmdline.txt` but this didn't work.

Output from "dmesg" in case the USB Ethernet devices are not recognized:
```
[    0.000000] Booting Linux on physical CPU 0x0                                                                                                                                 
[    0.000000] Linux version 4.19.58-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1245 SMP Fri Jul 12 17:31:45 BST 2019               
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d                                                                                                   
[    0.000000] CPU: div instructions available: patching division code                                                                                                           
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache                                                                                                   
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1                                                                                                            
[    0.000000] Memory policy: Data cache writealloc                                                                                                                              
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000                                                                                                                       
[    0.000000] On node 0 totalpages: 999424                                                                                                                                      
[    0.000000]   DMA zone: 1728 pages used for memmap                                                                                                                            
[    0.000000]   DMA zone: 0 pages reserved                                                                                                                                      
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63                                                                                                                           
[    0.000000]   HighMem zone: 802816 pages, LIFO batch:63                                                                                                                       
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e8 with crng_init=0                                                                                     
[    0.000000] percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632                                                                                                          
[    0.000000] pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096                                                                                                              
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3                                                                                                                               
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 997696                                                                                                     
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M video=HDMI-A-1:640x480@60 smsc95xx.macaddr=DC:A6:32:0D:A3:94 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait                   
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 3682812K/3997696K available (8192K kernel code, 658K rwdata, 2340K rodata, 2048K init, 850K bss, 52740K reserved, 262144K cma-reserved, 3211264K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 659 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 851 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[... cut ...]
[   42.073381] [<c02019bc>] (__irq_svc) from [<c0209a88>] (arch_cpu_idle+0x34/0x4c)
[   42.073404] [<c0209a88>] (arch_cpu_idle) from [<c09935a4>] (default_idle_call+0x34/0x48)
[   42.073426] [<c09935a4>] (default_idle_call) from [<c025423c>] (do_idle+0xec/0x17c)
[   42.073445] [<c025423c>] (do_idle) from [<c025458c>] (cpu_startup_entry+0x28/0x2c)
[   42.073461] [<c025458c>] (cpu_startup_entry) from [<c098cc80>] (rest_init+0xbc/0xc0)
[   42.073477] [<c098cc80>] (rest_init) from [<c0e00fe4>] (start_kernel+0x4b8/0x4e8)
[   42.073493] ---[ end trace 823697e2090bfb7d ]---
```

JamesH65, 2020-02-13:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

soxrok2212, 2020-08-24:
Hi, I managed to get it working by adding `coherent_pool=4M` to my boot config and it's now able to get a DHCP address and appears to be working.

Also, I am aware that I won't see gigabit speeds... I'll be using significantly less bandwidth ;)