raspberrypi/linux #2928: „lan78xx: kevent0 may have been dropped“ on Pi 3 B+

bcutter, 2019-04-09:
Symptoms:
- Kern.log is flooded with the message entries
- System completely dead in terms of network connectivity (no input no output, even outgoing Ping impossible - ifconfig and all network services looking okay by the way)
- Rebooting the whole system (only using console of course, therefore in some situations like being on work or holiday the systems dead for a longer period of time!) works around this issue

It happens pretty randomly. In some cases it happens when transferring a significant amount of data using the Ethernet interface, but not every time.

Similar issues:
https://github.com/raspberrypi/linux/issues/2447 (different Kernel Event, Same Chip)
https://github.com/raspberrypi/linux/issues/2027 (different Chip, Same Kernel Event)

It‘s really freaking me out especially nobody‘s looking into this so far and the issue linked above with the same kernel event on a different chip is quite old and STILL not fixed. So after the memory leak my Pi 3B+ was suffering on kernel 4.19 line for MONTHS and has been „fixed“ at the end of March in 4.29 kernel update this is the next issue keeping my system from being reliable.

Possible relevant information:
- Having three hard drives (SATA/USB converter) attached to the Pi
- Using a second virtual interface („eth0:0“) with a second IP address on the same system
- No other modifications I can think of being relevant on this issue.

Kernel currently (already happened on earlier versions of 4.19 path): 4.19.30

Example of kern.log:
‘‘‘
Feb 26 14:29:14 raspberrypi kernel: [198193.678073] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.428064] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.428089] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.428182] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.428226] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628603] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628678] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628717] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628765] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628804] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628841] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628888] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628927] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.628965] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.629013] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.629052] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.829514] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.829623] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.829669] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.829722] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:30:01 raspberrypi kernel: [198240.829771] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:34:24 raspberrypi kernel: [198503.992292] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:39:29 raspberrypi kernel: [198808.894392] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:39:29 raspberrypi kernel: [198808.894574] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:39:34 raspberrypi kernel: [198813.899593] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:39:34 raspberrypi kernel: [198813.899807] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:39:39 raspberrypi kernel: [198818.904917] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:39:39 raspberrypi kernel: [198818.905092] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:41:14 raspberrypi kernel: [198913.742733] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199135.864108] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199135.864213] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199135.864301] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199135.864422] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199135.864466] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.064917] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065041] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065088] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065205] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065299] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065506] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065611] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065747] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065838] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:56 raspberrypi kernel: [199136.065937] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:57 raspberrypi kernel: [199136.266493] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:57 raspberrypi kernel: [199136.266596] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:57 raspberrypi kernel: [199136.266690] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 14:44:57 raspberrypi kernel: [199136.266806] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
[... 684 lines cut ...]
Feb 26 20:37:01 raspberrypi kernel: [220261.055248] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055270] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055356] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055394] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055437] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055478] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055521] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055578] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055620] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055665] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055719] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055807] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055852] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055895] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.055945] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
[... cut ...]
Feb 26 20:37:01 raspberrypi kernel: [220261.056188] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.056233] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.056280] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:01 raspberrypi kernel: [220261.056332] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 20:37:30 raspberrypi kernel: [    0.000000] Booting Linux on physical CPU 0x0
‘‘‘

P33M, 2019-04-10:
When in the "dead" state, what is the output of `sudo slabtop`?

bcutter, 2019-04-10:
I‘ll investigate this on the next occurrence. Anything else I could check or collect in that situation?

Side note: Kernel back on latest 4.14(.98).

P33M, 2019-04-10:
The contents of `/proc/buddyinfo` would also be a useful data point to gather.

lategoodbye, 2019-04-14:
Could you please provide more information about the SATA/USB converters?

bcutter, 2019-04-14:
@P33M: Will also provide this on next occurence.

@lategoodbye:
Filtered output of `lsusb` is
```
Bus 001 Device 010: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge
Bus 001 Device 009: ID 152d:2329 JMicron Technology Corp. / JMicron USA Technology Corp. JM20329 SATA Bridge
Bus 001 Device 008: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge
```

bcutter, 2019-04-27:
Happened again few minutes ago (started at 00:18) when streaming media via SAMBA from one of the attached HDDs (no idea if this is the reason, but nothing else happened at that time).

The information you asked for (and some additional for my own documentation):

```
uptime
00:26:19 up 17 days, 9:59, 6 users, load average: 0,46, 0,33, 0,37
```

```
uptime -s
2019-04-10 14:27:01
```
```
sudo slabtop
```
--> See [this console picture](https://www.dropbox.com/s/trgix2kgdpj50iq/raspberrypi_issue%232928_2019-04-28_sudo%20slabtop.JPG?dl=0)

```
sudo cat /proc/buddyinfo
Node 0, zone   Normal   3287   4775   672   250   55   16    1    2    0    1   0
```
```
sudo cat /var/log/kern.log
Apr 21 22:25:16 raspberrypi kernel: [979103.054025] usb 1-1.1.3.4: reset high-speed USB device number 10 using dwc_otg
Apr 22 03:00:08 raspberrypi kernel: [995595.641954] EXT4-fs (sdb2): mounted filesystem with ordered data mode. Opts: (null)
Apr 22 03:01:40 raspberrypi kernel: [995687.483999] usb 1-1.1.3.4: reset high-speed USB device number 10 using dwc_otg
Apr 22 03:01:40 raspberrypi kernel: [995688.024086] usb 1-1.1.3.4: device not accepting address 10, error -71
Apr 22 03:01:40 raspberrypi kernel: [995688.124039] usb 1-1.1.3.4: reset high-speed USB device number 10 using dwc_otg
Apr 22 03:01:41 raspberrypi kernel: [995688.664038] usb 1-1.1.3.4: device not accepting address 10, error -71
Apr 22 03:01:41 raspberrypi kernel: [995688.764028] usb 1-1.1.3.4: reset high-speed USB device number 10 using dwc_otg
Apr 22 03:01:42 raspberrypi kernel: [995689.204037] usb 1-1.1.3.4: device not accepting address 10, error -71
Apr 22 03:01:42 raspberrypi kernel: [995689.304032] usb 1-1.1.3.4: reset high-speed USB device number 10 using dwc_otg
Apr 23 03:00:08 raspberrypi kernel: [1081996.175860] EXT4-fs (sdb2): mounted filesystem with ordered data mode. Opts: (null)
Apr 24 03:00:08 raspberrypi kernel: [1168397.369210] EXT4-fs (sdb2): mounted filesystem with ordered data mode. Opts: (null)
Apr 25 03:00:09 raspberrypi kernel: [1254798.429777] EXT4-fs (sdb2): mounted filesystem with ordered data mode. Opts: (null)
Apr 25 13:21:40 raspberrypi kernel: [1292090.603075] lan78xx 1-1.1.1:1.0 eth0: Failed to read register index 0x00000120. ret = -71
Apr 26 03:00:03 raspberrypi kernel: [1341193.516284] usb 1-1.1.3.2: reset high-speed USB device number 8 using dwc_otg
Apr 26 03:00:08 raspberrypi kernel: [1341198.415780] EXT4-fs (sdb2): mounted filesystem with ordered data mode. Opts: (null)
Apr 26 04:33:12 raspberrypi kernel: [1346782.913556] usb 1-1.1.3.3: reset high-speed USB device number 9 using dwc_otg
Apr 26 04:37:06 raspberrypi kernel: [1347016.405525] usb 1-1.1.3.3: reset high-speed USB device number 9 using dwc_otg
Apr 26 21:51:38 raspberrypi kernel: [1409088.864196] dwc_otg 3f980000.usb: dev 1.1.3.4 ep1in scatterlist error 0/-71
Apr 26 21:51:38 raspberrypi kernel: [1409088.864236] dwc_otg 3f980000.usb: dev 1.1.3.4 ep1in scatterlist error 0/-71
Apr 26 21:51:38 raspberrypi kernel: [1409088.864267] dwc_otg 3f980000.usb: dev 1.1.3.4 ep1in scatterlist error 0/-71
Apr 26 21:51:38 raspberrypi kernel: [1409088.864286] dwc_otg 3f980000.usb: dev 1.1.3.4 ep1in scatterlist error 0/-71
Apr 26 21:51:38 raspberrypi kernel: [1409088.864305] dwc_otg 3f980000.usb: dev 1.1.3.4 ep1in scatterlist error 0/-71
Apr 26 21:51:38 raspberrypi kernel: [1409088.955076] usb 1-1.1.3.4: reset high-speed USB device number 10 using dwc_otg
Apr 26 21:51:38 raspberrypi kernel: [1409089.055033] usb 1-1.1.3.4: device descriptor read/64, error -71
Apr 27 03:00:08 raspberrypi kernel: [1427599.199898] EXT4-fs (sdb2): mounted filesystem with ordered data mode. Opts: (null)
Apr 28 00:18:49 raspberrypi kernel: [1504321.249878] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.249930] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.249993] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.250055] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.250097] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.250130] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.250268] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.250274] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.250362] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.250446] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.250508] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.250555] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.250615] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.250650] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.250687] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:18:49 raspberrypi kernel: [1504321.250717] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
[... cut ...]
Apr 28 00:32:25 raspberrypi kernel: [1505137.098641] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:32:25 raspberrypi kernel: [1505137.098700] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Apr 28 00:33:51 raspberrypi kernel: [1505222.815097] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
```

Needed to reboot the system now. Would at least be great to know if there´s another workaround besides the reboot - or to find a solution of course.

"sudo ifdown" and "sudo ifup" of the eth0 interfaces made no change by the way - network still dead.

tevdi, 2019-05-02:
Can I suggest to unload & load the module ?

bcutter, 2019-05-02:
@tevdi:
a) which one
b) how
?

tevdi, 2019-05-03:
> @tevdi:
> a) which one
> b) how
> ?
The module lan78xx.
`sudo rmmod lan78xx && sudo modprobe lan78xx`

I use my Raspberry to load roms to an arcade machine (Sega Naomi) through ethernet wire. With the first version of Raspberry Pi it works perfect but with Raspberry Pi 3 B+ sometimes Ethernet simply stops working and after many hours investigating the only (easy) way to restart the communication is "restarting" the module lan78xx. I assume the problem is the module lan78xx.

Hope it helps

bcutter, 2019-06-15:
So after 26 days it happened again a few minutes ago while streaming a file from one of the attached HDDs.

I tried to ```sudo rmmod lan78xx && sudo modprobe lan78xx``` BUT:

```rmmod: ERROR: Module lan78xx is builtin.```

BAM - I‘m in a blind alley, stuck without any light.

It‘s freaking annoying. Any more help on this please?

tevdi, 2019-06-18:
> ```
> Jun 15 21:15:37 raspberrypi kernel: [2323645.260480] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:15:37 raspberrypi kernel: [2323645.260512] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:15:37 raspberrypi kernel: [2323645.260546] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:15:37 raspberrypi kernel: [2323645.260576] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:15:37 raspberrypi kernel: [2323645.260617] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:15:37 raspberrypi kernel: [2323645.260647] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:15:37 raspberrypi kernel: [2323645.260683] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:15:37 raspberrypi kernel: [2323645.260719] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:15:37 raspberrypi kernel: [2323645.260770] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:17:12 raspberrypi kernel: [2323740.967419] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:17:12 raspberrypi kernel: [2323740.967648] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:22 raspberrypi kernel: [2323811.197049] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:22 raspberrypi kernel: [2323811.197168] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:22 raspberrypi kernel: [2323811.197210] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:22 raspberrypi kernel: [2323811.197333] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:22 raspberrypi kernel: [2323811.197433] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.397849] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.397956] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.398058] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.398157] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.398261] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.398394] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.398435] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.398557] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.398657] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.398767] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.599284] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.599387] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.599470] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.599567] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:18:23 raspberrypi kernel: [2323811.599666] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:19:21 raspberrypi kernel: [2323870.160607] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:19:26 raspberrypi kernel: [2323874.422689] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:19:27 raspberrypi kernel: [2323876.172812] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:19:29 raspberrypi kernel: [2323878.166767] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:19:29 raspberrypi kernel: [2323878.238730] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> Jun 15 21:19:48 raspberrypi kernel: [2323897.203027] warning: `/bin/ping' has both setuid-root and effective capabilities. Therefore not raising all capabilities.
> Jun 15 21:19:55 raspberrypi kernel: [2323903.733114] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
> ```
> 
> So after 26 days it happened again a few minutes ago while streaming a file from one of the attached HDDs.
> 
> I tried to `sudo rmmod lan78xx && sudo modprobe lan78xx` BUT:
> 
> `rmmod: ERROR: Module lan78xx is builtin.`
> 
> BAM - I‘m in a blind alley, stuck without any light.
> 
> It‘s freaking annoying. Any more help on this please?

Oh sorry, this error is because the driver lan78xx is kernel builtin. To solve this, I recompilated the Kernel configuring driver lan78xx as a module in the .config file. If you are interested in recompile the Kernel follow this guide: https://www.raspberrypi.org/documentation/linux/kernel/building.md (you need knowledge about Linux).

bcutter, 2019-06-18:
Sorry, I am looking for a solution and as a user with some (!) technical knowledge the compilation of a kernel is definitely out of my comfort zone.

What else can I do to help others helping me? I ran out of ideas and need to fully rely on other users‘ knowledge :-(

I mean... it‘s a bug, right? Some usb timing issue in the lan78xx module I assume. How to fix it?

JamesH65, 2019-06-18:
Who knows where the bug is. It takes so long to replicate it will be impossible to debug without finding a quicker way of reproducing. Something is clogging the system or taking much longer than expected, causing the workqueue system to give this error. But it's so rare its going to be some weird timing issue somewhere, perhaps related to packet content to size, and could be in the lan driver or elsewhere. Very difficult to fix so if anyone can figure out a way of replicating it reliably, that would make things a LOT easier.

dkgof, 2019-06-26:
We are diskless booting an archlinux installation on a PI 3 B+, and with the latest kernel we are ketting the 
lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped as soon as the root file system is mounted on nfs and it starts booting, happens every time in our setup. Only by downgrading, (We used the latest 4.14 kernel, does it boot without the error).

bcutter, 2019-07-08:
After 23 days of uptime it happened again one hour ago. Needed to reboot as there is no other workaround. No load on the USB interfaces by the way, but on the ethernet interface instead.

It´s a neverending story... hell I´m happy to replace this peace of shit by the new Pi 4 which hopefully will have another implementation or at least segregation of ethernet and USB lanes, finger´s crossed.

JamesH65, 2019-07-09:
It practically impossible to debug an issue that open happens every 20 days or so, so really no idea how to progress this. Without some indication of what is causing it, or a simple way of replicate it, it's so so difficult. What is odd is that is it not being reported by many others, which sort of indicates HW either onboard, or in your environment. But again, who knows.

tevdi, 2019-07-09:
If you haven't anything else to do, I recommend you recompile the kernel and enable the driver lan78xx as a module, instead of built-in in the kernel.

Regards

Corvo, 2019-07-18:
Once (sometimes twice) per day may Raspberry Pi 3B+ suffers from this too (`kevent 0 may have been dropped`)
Result is that eth0 shuts down completely. When monitor and keyboard are hooked-up, I can do a reboot from the CLI otherwise unplugging and replugging the powercable is the only solution.

What can I do to help someone debug this particular issue?

The RPI is hooked-up with an official adapter + cable, wifi and bluetooth are disabled (as well as most GPIO related features that can be tweaked using `raspi-confi`). System is otherwise, up to date.

JamesH65, 2019-07-18:
This is much more frequent than any other reports, which can goes weeks without it happening. Not sure how you would go about doing it, but a wireshark trace around the point where the kevent problem fires might be useful. I suspect this is some sort of packet/set of packets on the network causing something to go slowly, causing the handling of the kevent to be delayed, causing the next occurence of the kevent to be dropped. You could probably have a script running `dmesg -w | grep kevent` or similar that causes wireshark to save its trace for the last few minutes. I'm not familiar enough with wireshark to know how to do that bit, if indeed it's possible!

lategoodbye, 2019-07-18:
Is anyone how affected by this problem able to test this patch:
https://marc.info/?l=linux-netdev&m=154939669728320&w=2

Corvo, 2019-07-20:
Well, I'm certainly open to applying a patch and testing it...
Problem is I wouldn't know where to start. Anybody able to build this patched kernel which I then can (somehow) install on my RPI??

All I can tell, for now, is that my RPI 3B+ has a temperature of 53,6°C with a load average of 0.15 / 0.13 / 0.11 when the network decides to stop working.

lategoodbye, 2019-07-20:
I setup a branch with this fix, you can fetch the necessary binaries from here:
https://github.com/lategoodbye/rpi-zero/releases/download/lategoodbye-kernel-lan78xx-v1/rpi-4.19.57-armv7.tar.gz

Corvo, 2019-07-20:
Thanks @lategoodbye

As per your instructions I've replaced the binaries on my RPI with the ones contained in the tarball you supllied. And now... I'll wait. _* fingers crossed *_

Corvo, 2019-07-21:
This time around the entire RPI froze..
I do have a picture of the last _dmesg_ output (there was no "kevent 0 may have been dropped" on those last few lines... Only a few stacktraces).
I will try to type a summary of it later today.

bcutter, 2019-07-30:
Happened again after almost 16 days (upgraded meanwhile to Raspbian Buster, but still on Pi 3 B+ hardware, will switch to Pi 4 in roughly 3 to 4 weeks).

```
Jul 30 12:26:17 raspberrypi kernel: [1347034.830973] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.831008] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.831021] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.831142] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.831155] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.831521] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.831689] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.831804] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.831964] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.832072] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.832145] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.832200] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.832231] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.841104] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.841140] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.841347] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.841419] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.841653] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.841773] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.841864] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.841927] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.842169] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.842309] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.842415] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.842483] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.842735] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.842853] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.842947] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.843009] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.843187] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.843254] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:17 raspberrypi kernel: [1347034.843289] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:41 raspberrypi kernel: [1347058.998148] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:42 raspberrypi kernel: [1347059.398850] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:42 raspberrypi kernel: [1347059.558222] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:26:44 raspberrypi kernel: [1347061.338251] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:11 raspberrypi kernel: [1347149.192638] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:11 raspberrypi kernel: [1347149.192686] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:23 raspberrypi kernel: [1347160.591662] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:23 raspberrypi kernel: [1347160.591696] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:23 raspberrypi kernel: [1347160.591732] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:23 raspberrypi kernel: [1347160.591768] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:23 raspberrypi kernel: [1347160.591820] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:23 raspberrypi kernel: [1347160.591857] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:23 raspberrypi kernel: [1347160.591889] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:23 raspberrypi kernel: [1347160.591964] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:23 raspberrypi kernel: [1347160.592004] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:23 raspberrypi kernel: [1347160.592058] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
[... cut ...]
Jul 30 12:28:39 raspberrypi kernel: [1347177.175493] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:39 raspberrypi kernel: [1347177.175566] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:28:40 raspberrypi kernel: [1347177.375924] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:29:02 raspberrypi kernel: [1347199.720778] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Jul 30 12:29:02 raspberrypi kernel: [1347199.720929] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
```

Corvo, 2019-07-30:
_Let me chime in as well:_
I've had the patched kernel until last weekend. The patched kernel would most of the time end up with the "kevent 0"-problem, but sometimes would freeze entirely. Currently running the latest (unpatched) kernel again. Problem stull occurs roughly once per day, and as a stop-gap I've connected the rPi to a smart socket which allows me to remotely pull-the-plug. Since so few people seem to have this problem, I'm somewhat thinking it is some faulty hardware.
I'm going to try to get mine replaced or fixed... and if that fails (hope not), I'll consider an rPi4 with some actively-cooled case.

lategoodbye, 2019-07-30:
@Corvo Could please describe your setup of the Raspberry Pi 3B+ (connected devices, running services), so we might have a chance to reproduce it?

Corvo, 2019-07-31:
@lategoodbye I'm running an up to date Raspbian Stretch install on the rPi3B+. Besides the default packages of Raspbian, I have installed:
* `openjdk-8-jre-headless` packages to be used with an install of the _Ubiquiti UniFi controller_ (`unifi` package from the official repository)
* `haveged` package
* _Pi-hole_ (by using the `curl -sSL https://install.pi-hole.net | bash` method)

The install is configured to use a static IP.

The PI is connected to a non-PoE port of a UniFi switch. No other peripherals are connected. The rPi is powered by the (starter kit) official plug.

If you need more info, let me know.

bcutter, 2019-08-01:
Happened again after only 2 days:

```
Aug  1 23:11:41 raspberrypi kernel: [211268.026627] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.026665] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.026704] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.026757] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.026777] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.026989] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027009] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027015] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027032] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027038] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027043] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027060] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027247] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027253] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027259] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027264] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027270] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027414] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027420] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027425] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027441] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027446] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027536] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027542] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027630] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027636] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027641] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027722] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027736] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027741] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027862] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027876] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027882] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027887] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027957] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027963] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.027985] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.028064] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.028143] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.028149] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.028275] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.028282] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.028287] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.028293] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.028307] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.028378] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.028384] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.028450] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.028456] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:11:41 raspberrypi kernel: [211268.028470] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
[... cut ...]
Aug  1 23:18:55 raspberrypi kernel: [211702.063730] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  1 23:19:30 raspberrypi kernel: [211737.052874] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
```

It happened after copying/streaming a lot of data from/to the HDD attached to the Pi during the last two days. It seems like it (ethernet port, USB port or lane between them) either is only capable of a certain amount of data (few gigabytes) or it´s a temperature thing. This is the only thing I could imagine, would probably explain the different time spans between the incidents.

bcutter, 2019-08-01:
And again. I think intensive streaming might be a way to debug things, at least it can be provoked this way...

```
Aug  2 01:19:56 raspberrypi kernel: [ 7148.744924] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.744957] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745040] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745206] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745211] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745228] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745234] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745239] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745315] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745321] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745472] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745478] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745484] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745489] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745508] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745599] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745606] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745683] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745689] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745767] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745785] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745791] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745915] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745930] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745935] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.745940] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746010] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746016] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746084] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746090] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746189] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746201] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746207] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746355] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746369] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746374] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746379] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746384] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746454] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746461] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746474] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746542] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746548] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746603] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746617] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746680] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746686] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746767] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:19:56 raspberrypi kernel: [ 7148.746774] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
[... cut ...]
Aug  2 01:20:18 raspberrypi kernel: [ 7171.116679] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:20:18 raspberrypi kernel: [ 7171.317310] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:20:18 raspberrypi kernel: [ 7171.317354] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:20:47 raspberrypi kernel: [ 7200.232261] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Aug  2 01:21:14 raspberrypi kernel: [ 7227.545796] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
```

bcutter, 2019-08-19:
Switched to Pi 4 hardware now. Finger‘s crossed...

bcutter, 2019-10-05:
Update. So Pi 4 hardware is running constantly and stable for more than 32 days now (a time where Pi 3B+ hardware would´ve crashed already at least once, the usually time was about maximum 21 days) I assume it´s a 3B+ specific error - which I really don´t care anymore, I spent so much time on this **** of crappy hardware - I´m very happy with the extremely fast Pi 4 now. As it´s not really fixed I don´t want to close the issue though.

JamesH65, 2019-10-07:
There is en entirely new ethernet controller on the Pi4, so this is unsurprising. The new controller will have its own weird issues, so no need to include this one!

JamesH65, 2020-01-17:
There has been some recent movement/changes to the lan78xx driver that may have some impact on this on the Pi3B+. Once they are merged I'll flag it up here. If you are happy to build you own kernel etc then let me know and I can post links to the patches.

JamesH65, 2020-01-17:
Actually, will probably be in rpi-update in the next few days, as we have merged them to our github tree. So you can wait for that, or just get the latest tree and build it.

puer, 2021-01-28:
Is this issue addressed in 2021 Jan 's apt upgrade kernel?

ernstkl, 2021-02-12:
I am having this issue with 5.4.79-v7+ on a 3B+

labtec901, 2022-08-07:
I am also having this issue on 5.15.32-v8 on a 3B+