raspberrypi/linux #2931: Please stop rebasing branch considered as stable (4.19)

schnitzeltony, 2019-04-10:
Or have you been hacked

JamesH65, 2019-04-11:
4.19 is not considered stable from our point of view.

schnitzeltony, 2019-04-11:
Honestly?
* You stopped maintaining 4.14
* 4.19 was set default branch

JamesH65, 2019-04-11:
So? We are still working on 4.19, and it is not yet our release branch. There's a known memory leak in 4.19 that needs to be fixed before we can release it.

schnitzeltony, 2019-04-11:
Was not aware - thanks.
It is just that 4.19 is marked default. In the past one could rely on default branch not being rebased. So let's hope situation changes soon.

paul-betafive, 2019-04-11:
@JamesH65 : If the rpi-4.19.y branch is unstable, please switch the default branch back to repi-4.14.y.

I'm happy to create a merge request to bring in the upstream 4.14.111 release to the rpi-4.14.y branch if that's likely to be accepted.

paul-betafive, 2019-04-11:
@schnitzeltony Can we re-open this issue?

schnitzeltony, 2019-04-11:
Sure

agherzan, 2019-04-16:
I'm also unsure why don't we have the default branch set to the last stable branch. I was always assuming that is the case.

popcornmix, 2019-04-16:
4.19 is where development is happening and where we would be accepting pull requests.
As such it makes sense to me to be the default branch.
Note: default branch != stable branch

paul-betafive, 2019-04-16:
@popcornmix: What is the current stable branch? The 4.14.y branch doesn't appear to be merging upstream stable kernel releases any longer.

kraj, 2019-04-17:
@popcornmix if 4.19.y is development branch still it would help the community to consume it and pre-test it in distribution if it was using merge update plan. I am sure you have reasons for constant rebasing, but the naming convention for branch puts it in public branch namespace. It would be cleared if you created something like $username/$branchname to indicate that its a private branch which can be rebased without notice.

schnitzeltony, 2019-04-17:
4.19 is not ready for production we now should be aware of now.

Although I don't understand why it's necessary for development to abuse git and destroy history - how about this:
@paul-betafive : sends a pull request to update 4.14 and we continue to use it as long as 4.19 is not ready

agherzan, 2019-04-17:
@schnitzeltony I reckon that's the best way forward for now.

paul-betafive, 2019-04-17:
I'll try to get the kernel.org 4.14.112 release merged into the rpi-4.14.y branch via a pull request today. If that fails to be accepted I'll create a new linux-rpi-lts repository.

kraj, 2019-04-17:
yes lets do it for now and keep the 4.19 recipe around for future when this is declared stable

popcornmix, 2019-05-20:
4.19 is now in apt and is considered stable.
4.19 tree will now have merges when updating rather than rebases.
Newer trees will be rebased.

schnitzeltony, 2019-05-20:
Good news - thanks for heads up.

closed 2019-05-20 by schnitzeltony

pbrkr, 2019-05-20:
This is good news but I still have some open questions @popcornmix :

Is there any official place we can find out which kernel branch is considered stable? For those of us building our own images with Yocto Project or Buildroot we're not likely to see what's going on in the apt repositories in Raspbian.

Why were updates of the rpi-4.14.y branch stopped long before rpi-4.19.y was considered stable? The last stable kernel merge on the rpi-4.14.y branch before this issue was opened was on 12th Feb (https://github.com/raspberrypi/linux/commit/5d63a4595d32a8505590d5fea5c4ec1ca79fd49d) and rpi-4.19.y has only just been marked stable today - that's 3 months where there was no stable branch where security updates were being applied if I understand correctly.

If you need some help in this area in the future I'm happy to get involved but I'd like to do this in a proactive way rather than reacting when we notice that builds are failing and branches are falling behind.

popcornmix, 2019-05-20:
You can see the current apt stable kernel from:
https://github.com/raspberrypi/firmware/blob/stable/extra/uname_string7

Or the source it was build from (latest tag) here:
https://github.com/raspberrypi/linux/tags

Also keep an eye on the Advanced Users forum. Threads like:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=224931

will typically give a few months notice of a planned major bump to the kernel.