raspberrypi/linux #2932: rpi3: ttyAMA0 vs ttyS0 + magic sysrq + pi3-disable-bt = confusion?

m1027, 2019-04-11:
On my rpi3:

I thought, with rpi3 there is ttyS0 to be used instead of ttyAMA0 for standard serial communication.

However, on ttyS0, everything worked besides the magic sysrq. As recommended here #2142, I added dtoverlay=pi3-disable-bt. By this, in turn, I lost any output on ttyS0 after next reboot.

Solution was: With both, dtoverlay=... and using ttyAMA0, now both works: console output
and magic sysrq.

I am confused, though, why to use ttyAMA0 on rpi3? Is that correct?

pelwell, 2019-04-11:
As they say, "This behaviour is by design."

ttyAMA0 is the more capable UART - it has larger FIFOs, functional flow control, and a baud rate that is independent of the VPU's core clock. For this reason it is used for Bluetooth by default, leaving ttyS0 to run the console.

`pi3-disable-bt` releases ttyAMA0 from Bluetooth duty and switches it back to the normal UART pins (14 & 15). `pi3-miniuart-bt` does all of that, but also maps ttyS0 to the Bluetooth modem pins for a lower-bandwidth Bluetooth link (but you need to lock the core frequency to, say, 250MHz - "core_freq=250" - to get a stable baud rate).

To insulate the Linux configuration from these changes there are aliases for the two UARTs - "serial0" and "serial1". The firmware searches `cmdline.txt` and replaces "serial0" with the name of the "primary" 
 (console) UART; "serial1" is replaced with the "secondary" (other) UART (often used for Bluetooth). To complement this, one of the Raspbian system scripts creates symbolic links "/dev/serial0" and "/dev/serial1" to point to the appropriate UART device. The combination of the two should make it possible to remove explicit references to either of the two UARTs and use the "serial<n>" aliases instead.

m1027, 2019-04-11:
Thanks for clarifying that.

On serial0/1: I am not using raspbian, so I'll probably better stick to the real device names.

closed 2019-04-11 by m1027

pelwell, 2019-04-11:
You can still use serial<n> in cmdline.txt - the firmware handles that - but it's completely up to you.