raspberrypi/linux #2933: Please provide Raspbian Docker images

raffaeler, 2019-04-11:
*This is not a bug but a warm suggestion.*

Compiling large libraries on the RPi can be a pain because it requires a huge amount of time due to the lack of RAM and the speed of the I/O to the SD card.
In some cases (OpenCV, boost, etc.) the compiler goes out of memory and either swap indefinitely or crashes.

Cross compiling is a solution, but it isn't exactly a smooth process and may require a very long and complex setup.

The best solution is using Docker. I used it to successfully compile OpenCV starting from the Debian Stretch base image. Anyway, the Debian image brings some minor glitches due to the fact that Raspbian comes with a number of pre-installed libraries and some other have a different version.

*Running on a X64 host a Docker container with a different architecture (ARM) works already*.

Having a Docker image published on the Docker Hub would be a killer feature. You just run `docker -it -v /src:/src raspberrypi/raspbian` (of course the docker image still doesn't exist right now) and run the compilation script. Thanks to the shared volume the compiled libraries is saved on the mapped host disk. Once finished the library can be copied to the SD and installed using `make install`.

Beyond the compilation scenario, it can be extremely useful for CI/CD testing scenarios.

I hope you'll take into consideration this suggestion and thanks for the great ongoing work.

XECDesign, 2019-04-12:
I wouldn't volunteer to do this. There are already plenty of raspbian containers on docker hub. And since I don't use docker myself, I wouldn't know what people want from it exactly. qemu-debootstrap+schroot+sbuild work very well and integrate nicely with the rest of debian's native tools.

raffaeler, 2019-04-12:
The point is having a stable and official line of containers every time a new image is published.
No need of qemu or other. The plain vanilla installation of Docker on Windows over HyperV is ready to run cross-architecture containers.

JamesH65, 2019-09-19:
Closing as a Won't Fix. This means that although the issue is acknowledged, it is felt that no further action is likely. This may be down to a bad benefit vs cost analysis, or it may actually not be possible to fix.

closed 2019-09-19 by JamesH65